introlab/rtabmap_ros #431: compressed rate parameter in rgbd_sync is ineffective

vivekrathi14, 2020-06-13:
In /rtabmap/rgbd_sync node what is the use of the parameter compressed_rate? 
Changing its value has no effect on the bandwidth of the /rtabmap/rgbd_image/compressed topic, i.e.  `rostopic echo bw /rtabmap/rgbd_image/compressed `
The above command gives the same response whatsoever.

matlabbe, 2020-06-23:
This is not a dynamic parameter, so if you tried to modify it after launching rgbd_sync, nothing will change. The bandwidth should decrease when used if the `compressed_rate` is lower than the actual frame rate of the camera:
https://github.com/introlab/rtabmap_ros/blob/83c40d079d20660b67507f6bfc8af929fb537800/src/nodelets/rgbd_sync.cpp#L226-L233

vivekrathi14, 2020-06-23:
@matlabbe 
I had tried values from 5 to 1000 for compressed_rate using <param> in launch file but bw did not change at all.
Your explanation helps though.
However, even if compressed_rate parameter is used or not /rtabmab/rgbd_sync produces both /rgbd_image & /rgbd_image/compressed. Can you explain why?

matlabbe, 2020-07-06:
I just tried this example: https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/test_rgbd_image.launch
It seems to work as expected:
```
$ roslaunch rtabmap_ros test_rgbd_image.launch compressed:=true compressed_rate:=2

[ INFO] [1594063099.019244364]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1032s, Maps update=0.0014s pub=0.0001s (local map=1, WM=1)
[ INFO] [1594063099.026752741]: Odom: quality=288, std dev=0.015383m|0.027549rad, update time=0.042571s
[ INFO] [1594063099.551306742]: Odom: quality=277, std dev=0.017150m|0.027549rad, update time=0.036781s
[ INFO] [1594063099.611636050]: rtabmap (2): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0514s, Maps update=0.0012s pub=0.0000s (local map=1, WM=1)
[ INFO] [1594063100.046151812]: Odom: quality=277, std dev=0.016699m|0.027549rad, update time=0.025242s
[ INFO] [1594063100.578239634]: Odom: quality=279, std dev=0.014116m|0.032761rad, update time=0.036643s
[ INFO] [1594063100.637969112]: rtabmap (3): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0530s, Maps update=0.0015s pub=0.0000s (local map=1, WM=1)
[ INFO] [1594063101.076101274]: Odom: quality=273, std dev=0.016227m|0.027549rad, update time=0.026846s
[ INFO] [1594063101.572083982]: Odom: quality=266, std dev=0.036939m|0.043116rad, update time=0.023323s
[ INFO] [1594063101.629773649]: rtabmap (4): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0512s, Maps update=0.0014s pub=0.0000s (local map=1, WM=1)
[ INFO] [1594063102.129017868]: Odom: quality=289, std dev=0.035869m|0.046331rad, update time=0.042221s
[ INFO] [1594063102.573664703]: Odom: quality=287, std dev=0.041706m|0.027549rad, update time=0.022092s
[ INFO] [1594063102.631935348]: rtabmap (5): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0511s, Maps update=0.0012s pub=0.0000s (local map=1, WM=1)
[ INFO] [1594063103.110458867]: Odom: quality=283, std dev=0.031082m|0.041195rad, update time=0.029284s
[ INFO] [1594063103.577464571]: Odom: quality=261, std dev=0.044586m|0.046331rad, update time=0.022800s
[ INFO] [1594063103.709445144]: rtabmap (6): Rate=1.00s, Limit=0.000s, RTAB-Map=0.1216s, Maps update=0.0054s pub=0.0000s (local map=1, WM=1)
```
We can see the odometry log is at 2 Hz, meaning that `/rgbd_image/compressed` is received at 2 Hz. Other stats:
```
$ rostopic hz /camera/rgbd_image /camera/rgbd_image/compressed
subscribed to [/camera/rgbd_image]
subscribed to [/camera/rgbd_image/compressed]
            topic                rate   min_delta   max_delta   std_dev   window
================================================================================
/camera/rgbd_image              30.02   0.002871    0.06105     0.01068   28    
/camera/rgbd_image/compressed   2.02    0.4951      0.4951      0.0       28
```
Which rtabmap version are you using? on which platform?

vivekrathi14, 2020-07-20:
Yes, compressed_rate of 1-3 works well for me too.
System Specifications -
-Ubuntu 16.04 Xenial
-ROS - Kinetic distro
-rtabmap & rtabmap_ros version 0.19.3
Let me elaborate my question-
If you do `roslaunch rtabmap_ros rtabmap.launch rgbd_sync:=true`, you can notice that /rtabmab/rgbd_sync produces both /rgbd_image & /rgbd_image/compressed even if we use compressed=true or not. Can you explain why? Mostly it should be in source code, but I didn't get it.

matlabbe, 2020-07-31:
Yes, the node provide both. However for efficiency, we only compress the images [if there is a subscriber ](https://github.com/introlab/rtabmap_ros/blob/ae854dfb4cb8b38299e1abc7b632f178e9e67bb4/src/nodelets/rgbd_sync.cpp#L223)on `/rgbd_image/compressed`. Setting `compressed=true` will make `rtabmap` node subscribing to compressed image topic instead of the raw one.

vivekrathi14, 2020-07-31:
Thanks, that helps.

closed 2020-07-31 by vivekrathi14