introlab/rtabmap_ros #432: Low frequency of loop closure detection in rtabmap

vivekrathi14, 2020-06-13:
System Specifications -
-Ubuntu 16.04 Xenial
-ROS - Kinetic distro
-rtabmap & rtabmap_ros version 0.19.3

I am using rtabmap with default configuration with few changes in input topics. 
1. In rtabmapviz, the rate/frequency of the images in loop closure detection is very low, i.e. frames/images does not change smoothly like this following [tutorial](https://github.com/introlab/rtabmap/wiki/Loop-closure-detection). 
2. As of now I am using gazebo and performing the simulation with input RGB & Depth images as the inputs. 
3. I tried changing Rtabmap/DetectionRate to 0, still no improvement in the smoothness or rate of the images in loop closure detection.
3. Does input source frequency play a crucial part here? In Rtabmapviz I can see source input frequency/rate as 0 Hz, however I could not change it. Morever, the input images i.e. rgb/raw & depth/raw are published at approximately 20 Hz which is passed as input to rtabmap. This questions me about the source frequency from rtabmapviz. It seems very confusing to me. 
4. I am not able to change most of rtabmapviz parameters and also don't see same UI as per this [tutorial](https://github.com/introlab/rtabmap/wiki/Change-parameters) 
Please let me know any possible solution to resolve this?

matlabbe, 2020-06-30:
Hi,

Using `Rtabmap/DetectionRate=0`, you should have get the same framerate than the input images. If parameters are disabled in rtabmapviz, it means that `rtabmapviz` cannot call the ros services of `rtabmap` node (because they are not in same namespace?). Odometry parameters are always disabled in rabmapviz, they should be set in launch files when launching the odometry nodes.

Look at the terminal logs, rtabmap node should log at same rate than it publishes and show the actual rate that it is using. If the rate is not 0, then the parameters may not have been set at the correct place.

vivekrathi14, 2020-07-02:
Hi,
1. I am able to change/view all parameters through rtabmapviz gui but not the ones in source tab. I am launching rtabmapviz from rtabmap.launch only, so I believe namespace issue must be resolved.
2. For loop closure, making Rtabmap/DetectionRate=0 doesn't help, in rtabmapviz, the images/frames move very slow. The images from loop closure detection window & 3D map window are not in sync, we can see that images in loop closure detection window lag a lot.
3. Keeping RGBD/Enabled parameter to false helps getting loop closure detection working smooth but it needs map_always_update parameter set to true. Setting may_always_update=false fails to load the map & navigation stack too.
4. Can you suggest a way to have correct loop closure detection & get it working with either map_always_update=true or map_always_update=false?

matlabbe, 2020-07-10:
1. Source and Odometry tabs are always disabled in `rtabmapviz`. `rtabmapviz` can only changes dynamically parameters from `rtabmap` node.
2. Looks like a rendering performance issue. What is the frame rate you are expecting? What kind of computer do you have? Does it have a GPU?
3. `RGBD/Enabled` should stay true when poses are used. If the detection is smoother with `RGBD/Enabled` to false, we would have to check which part is taking the most computation time at each iteration. `map_always_update` is used to update the global occupancy grid on each update, whether a new nodes is added to graph or not. Are you in localization mode (`Mem/IncrementalMemory=false`)? If not, I don't recommend setting `Rtabmap/DetectionRate` to 0. 
4. Loop closure detection is independent of `map_always_update`, do you have database to share? We could spot quite fast which part is using the most computation time.

vivekrathi14, 2020-07-20:
1. Okay. What is the way/procedure to achieve it?
2. I am using my windows laptop with VM. I am not using any GPU.
3.1 As I followed this tutorial, `RGBD/Enabled` is kept False. By default it is set to True. Changing the parameter to False only gave me updated frequency and improved loop closure which was confirmed through rtabmap_viz.
3.2 Yes, but somehow `map_always_update`  & `RGBD/Enabled` do not go well. You can try that.
3.3 No, I am not in localization mode (`Mem/IncrementalMemory=true`) in my case. Why do you recommend so. Making Rtabmap/DetectionRate=0 helped me for 3.1. I won't say that they are dependent but both of the parameters helped me combined to improve my performance.
4. I don't have a camera with me. Does data-base from simulation (gazebo) works for you?

matlabbe, 2020-07-31:
Yes, a database from simulation works too. It is not recommended to use Gazebo in a VM without GPU, it will be very slow and this could impact topic synchronization (causing rtabmap to lag because it doesn't receive all images corrected at the same time).