introlab/rtabmap_ros #433: Limit Global Loop closure distance

eliabntt, 2020-06-14:
Hi, I ran into a problem, as you can see in the picture, is that there are very similar desks/chairs... in my simulation. 
[image]

I started in front of the white table, went right, approached the desk, went back and all the way to the left.

As you can see the global loop closures triggers from one to the other side of the map that are already ruining the map.
is there a way to limit this behavior? For example by limiting the linear distance for a global loop closure? I can't find any working parameter. It's not due to ProximityDetection since I've also tried to disable that. Or limiting the number of considered nodes, even though this may something that also ruins performance.

Currently, I'm using both ICP and Visual loop closure but I've tried this also only with ICP and only with Visual and the result is pretty much the same (even though the system is much more stable by using both).

matlabbe, 2020-06-30:
If I understand, is the problem the global loop closures (red links)? In rtabmap there is a mechanisn to reject loop closures if the graph deformation is over 3x times the covariances in the links. The parameter is `RGBD/OptimizeMaxError` (default 3), you could try decreasing it to 2 or 1. However, this approach will only work well if the covariances set in the links make sense. If the covariance is overestimated, it will still accept the loop closure. If the covariance is underestimated, all loop closures would be rejected.

eliabntt, 2020-07-01:
Yup, the problem is that I've turned on the robust flag. With that off (so with OptimizeMaxError) it seems working ;)

matlabbe, 2020-07-10:
The difference between `RGBD/OptimizeMaxError` and `Optimizer/Robust` is:
 * `Optimizer/Robust`: Accept the loop closure (bad or good), but during optimization it would disable wrong loop closures. Covariances should be carefully set to work well.
 * `RGBD/OptimizeMaxError`: Reject a loop closure if the graph deformation is over a ratio of he covariances. Covariances should also make sense, setting odometry covariance to identity won't work.

There are a couple of examples on this page: https://github.com/introlab/rtabmap/wiki/Robust-Graph-Optimization

closed 2020-07-10 by matlabbe