introlab/rtabmap_ros #435: advantages of using Velodyne

molinadavid, 2020-06-22:
Hi Mathieu,

This is more of clarification question rather than an issue. As you may remember we are working on with Rtabmap under ROS with a Zed camera and assisted by a Velodyne 3D Lidar.

Our question is how does the Velodyne improves the mapping and localization process and how would it compare against using a 2D Lidar or no Lidar at all.

Also if you have some numbers to share about the different cases or further suggestions it would be very much appreciated.

Best regards,
David

matlabbe, 2020-07-06:
We don't have a direct comparison between 2d and 3d lidars for typical 2d operation. My feeling is that with a 3D lidar you could have a more robust ICP registration. Beware that in rtabmap the ICP parameters are tuned by default for 2D lidars, most of them should be tuned for 3D lidars. I also stronly recommend to build rtabmap with libpointmatcher support when using 3D lidars.

molinadavid, 2020-07-07:
Hi Mathieu,

Thank you very much the response, regarding the ICP registration I would like to ask if the full range of the velodyne (360 degrees) is taken into consideration or is only the camera's FOV used by Rtabmap.

matlabbe, 2020-07-10:
ICP registration uses the lidar directly, so all points from the lidar are used.