introlab/rtabmap_ros #437: Some questions about RTABMAP VIO

getupcurry, 2020-06-27:
Hi, matlabbe. 
According to your demo of test_d435i_vio.launch, I wrote a launch file of mynteye, the file as follows:
`<launch>

<arg name="database_path"     default="rtabmap.db"/>
<arg name="rtabmapviz"        default="false"/>
<arg name="localization"      default="false"/>
<arg     if="$(arg localization)" name="args"  default=""/>
<arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>
<arg name="depth_mode" default="true"/>

<arg name="left_image_topic"        value="/mynteye/left_rect/image_rect"/>
<arg name="right_image_topic"       value="/mynteye/right_rect/image_rect"/>
<arg name="left_camera_info_topic"  value="/mynteye/left_rect/camera_info"/>
<arg name="right_camera_info_topic" value="/mynteye/right_rect/camera_info"/>

<arg name="wait_for_transform"  default="0.2"/> 

<node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_node">
   <param name="use_mag"       value="false"/>
   <param name="publish_tf"    value="false"/>
   <param name="world_frame"   value="enu"/>
   <remap from="/imu/data_raw" to="/mynteye/imu/data_raw"/>
   <remap from="/imu/data"     to="/rtabmap/imu"/>
</node>

<node pkg="topic_tools" type="throttle" name="imu_throttle" args="messages /rtabmap/imu 100.0"/>

<group ns="rtabmap">
<node if="$(arg depth_mode)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" args="--Optimizer/GravitySigma 0.3 --Reg/Force3DoF true --Odom/Strategy 9 --OdomVINS/ConfigPath /home/yu/vins-fusion-mynt/src/MYNT-EYE-VINS-FUSION-Samples/config/mynteye-s/mynt_stereo_imu_config.yaml --Rtabmap/ImagesAlreadyRectified false --Mem/UseOdomGravity true" output="screen">
   <remap from="left/image_rect"   to="/mynteye/left_rect/image_rect"/>
   <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
   <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
   <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>
   <remap from="imu"               to="/rtabmap/imu_throttle"/>
   <param name="frame_id"          value="base_link"/>
   <param name="wait_imu_to_init"  value="true"/>
</node>
</group>

<include if="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg name="rtabmap_args"      value="--delete_db_on_start --Optimizer/GravitySigma 0.3"/>
   <arg name="rgb_topic"         value="/mynteye/left_rect/image_rect"/>
   <arg name="depth_topic"       value="/mynteye/depth/image_raw"/>
   <arg name="camera_info_topic" value="/mynteye/left_rect/camera_info"/>
   <arg name="visual_odometry"   value="false"/>
   <arg name="frame_id"          value="base_link"/>
   <arg name="imu_topic"         value="/rtabmap/imu_throttle"/>
   <arg name="rtabmapviz"        value="$(arg rtabmapviz)"/>
   <arg name="rviz"              value="true"/>
</include>

</launch>`

In this launch file, I add "--Reg/Force3DoF true", but the odometry is still 6DoF.  What should I do to set VIO as 3DoF?

Thanks.

getupcurry, 2020-06-27:
<launch>
<arg name="database_path"     default="rtabmap.db"/>
<arg name="rtabmapviz"        default="false"/>
<arg name="localization"      default="false"/>
<arg     if="$(arg localization)" name="args"  default=""/>
<arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>
<arg name="depth_mode" default="true"/>

<arg name="left_image_topic"        value="/mynteye/left_rect/image_rect"/>
<arg name="right_image_topic"       value="/mynteye/right_rect/image_rect"/>
<arg name="left_camera_info_topic"  value="/mynteye/left_rect/camera_info"/>
<arg name="right_camera_info_topic" value="/mynteye/right_rect/camera_info"/>

<arg name="wait_for_transform"  default="0.2"/> 

<node pkg="imu_filter_madgwick" type="imu_filter_node" name="imu_filter_node">
   <param name="use_mag"       value="false"/>
   <param name="publish_tf"    value="false"/>
   <param name="world_frame"   value="enu"/>
   <remap from="/imu/data_raw" to="/mynteye/imu/data_raw"/>
   <remap from="/imu/data"     to="/rtabmap/imu"/>
</node>

<node pkg="topic_tools" type="throttle" name="imu_throttle" args="messages /rtabmap/imu 100.0"/>

<group ns="rtabmap">
<node if="$(arg depth_mode)" pkg="rtabmap_ros" type="stereo_odometry" name="stereo_odometry" args="--Optimizer/GravitySigma 0.3 --Reg/Force3DoF true --Odom/Strategy 9 --OdomVINS/ConfigPath /home/yu/vins-fusion-mynt/src/MYNT-EYE-VINS-FUSION-Samples/config/mynteye-s/mynt_stereo_imu_config.yaml --Rtabmap/ImagesAlreadyRectified false --Mem/UseOdomGravity true" output="screen">
   <remap from="left/image_rect"   to="/mynteye/left_rect/image_rect"/>
   <remap from="right/image_rect"  to="/mynteye/right_rect/image_rect"/>
   <remap from="left/camera_info"  to="/mynteye/left_rect/camera_info"/>
   <remap from="right/camera_info" to="/mynteye/right_rect/camera_info"/>
   <remap from="imu"               to="/rtabmap/imu_throttle"/>
   <param name="frame_id"          value="base_link"/>
   <param name="wait_imu_to_init"  value="true"/>
</node>
</group>

<include if="$(arg depth_mode)" file="$(find rtabmap_ros)/launch/rtabmap.launch">
   <arg name="rtabmap_args"      value="--delete_db_on_start --Optimizer/GravitySigma 0.3"/>
   <arg name="rgb_topic"         value="/mynteye/left_rect/image_rect"/>
   <arg name="depth_topic"       value="/mynteye/depth/image_raw"/>
   <arg name="camera_info_topic" value="/mynteye/left_rect/camera_info"/>
   <arg name="visual_odometry"   value="false"/>
   <arg name="frame_id"          value="base_link"/>
   <arg name="imu_topic"         value="/rtabmap/imu_throttle"/>
   <arg name="rtabmapviz"        value="$(arg rtabmapviz)"/>
   <arg name="rviz"              value="true"/>
</include>
</launch>

matlabbe, 2020-07-08:
VINS doesn't support 3DoF. However, normally would should be able to force it to 3DoF aftwerwards. I fixed it in this commit: https://github.com/introlab/rtabmap/commit/4a09c4bdcf68491cd7941fa28d5c8db0bfcf729d

cheers,
Mathieu