introlab/rtabmap_ros #440: Catkin_make -j1 Error

doomb007, 2020-07-17:
I'm trying to install rtabmap_ros by following this [tutorial](https://github.com/introlab/rtabmap_ros), everything goes smoothly, but when I install RTAB-Map ros-pkg in the src folder of my Catkin workspace and run `catkin_make -j1`, I got an error:

```
Base path: /home/martin/catkin_ws
Source space: /home/martin/catkin_ws/src
Build space: /home/martin/catkin_ws/build
Devel space: /home/martin/catkin_ws/devel
Install space: /home/martin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/martin/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/martin/catkin_ws/build"
####
[  0%] Built target _beginner_tutorials_generate_messages_check_deps_Num
[  0%] Built target std_msgs_generate_messages_eus
[  0%] Built target _beginner_tutorials_generate_messages_check_deps_AddTwoInts
[  1%] Built target beginner_tutorials_generate_messages_eus
[  1%] Built target std_msgs_generate_messages_lisp
[  2%] Built target beginner_tutorials_generate_messages_lisp
[  2%] Built target std_msgs_generate_messages_py
[  4%] Built target beginner_tutorials_generate_messages_py
[  4%] Built target std_msgs_generate_messages_nodejs
[  4%] Built target beginner_tutorials_generate_messages_nodejs
[  4%] Built target std_msgs_generate_messages_cpp
[  4%] Built target beginner_tutorials_generate_messages_cpp
[  4%] Built target beginner_tutorials_generate_messages
[  4%] Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeGraphMsg
[  4%] Built target _lsd_slam_viewer_generate_messages_check_deps_keyframeMsg
[  5%] Built target lsd_slam_viewer_generate_messages_py
[  6%] Built target lsd_slam_viewer_generate_messages_cpp
[  7%] Built target lsd_slam_viewer_generate_messages_eus
[  7%] Built target lsd_slam_viewer_generate_messages_lisp
[  8%] Built target lsd_slam_viewer_generate_messages_nodejs
[  8%] Built target lsd_slam_viewer_generate_messages
[  8%] Built target lsd_slam_viewer_gencfg
[ 10%] Built target viewer
[ 11%] Built target lsd_slam_core_gencfg
[ 18%] Built target lsdslam
[ 19%] Built target live_slam
[ 19%] Built target dataset
[ 19%] Built target bond_generate_messages_lisp
[ 19%] Built target _catkin_empty_exported_target
[ 19%] Built target roscpp_generate_messages_py
[ 19%] Built target rosgraph_msgs_generate_messages_eus
[ 19%] Built target rosgraph_msgs_generate_messages_nodejs
[ 19%] Built target roscpp_generate_messages_cpp
[ 19%] Built target rosgraph_msgs_generate_messages_cpp
[ 19%] Built target roscpp_generate_messages_eus
[ 19%] Built target rosgraph_msgs_generate_messages_py
[ 19%] Built target rosgraph_msgs_generate_messages_lisp
[ 19%] Built target roscpp_generate_messages_nodejs
[ 19%] Built target roscpp_generate_messages_lisp
[ 19%] Built target dynamic_reconfigure_generate_messages_lisp
[ 19%] Built target dynamic_reconfigure_generate_messages_py
[ 19%] Built target dynamic_reconfigure_generate_messages_cpp
[ 19%] Built target dynamic_reconfigure_generate_messages_nodejs
[ 19%] Built target dynamic_reconfigure_generate_messages_eus
[ 19%] Built target sensor_msgs_generate_messages_nodejs
[ 19%] Built target geometry_msgs_generate_messages_cpp
[ 19%] Built target dynamic_reconfigure_gencfg
[ 19%] Built target sensor_msgs_generate_messages_eus
[ 19%] Built target geometry_msgs_generate_messages_py
[ 19%] Built target geometry_msgs_generate_messages_eus
[ 19%] Built target geometry_msgs_generate_messages_lisp
[ 19%] Built target geometry_msgs_generate_messages_nodejs
[ 19%] Built target sensor_msgs_generate_messages_cpp
[ 19%] Built target sensor_msgs_generate_messages_lisp
[ 19%] Built target sensor_msgs_generate_messages_py
[ 19%] Built target tf_generate_messages_cpp
[... 170 lines cut ...]
                 from /opt/ros/kinetic/include/ros/time.h:57,
                 from /opt/ros/kinetic/include/ros/ros.h:38,
                 from /home/martin/catkin_ws/src/rtabmap_ros/src/StereoCameraNode.cpp:28:
/usr/include/stdint.h:55:27: note:   ‘uint64_t’
 typedef unsigned long int uint64_t;
                           ^
/usr/include/stdint.h:55:27: note:   ‘uint64_t’
In file included from /usr/local/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/CameraStereo.h:36:0,
                 from /home/martin/catkin_ws/src/rtabmap_ros/src/StereoCameraNode.cpp:31:
/usr/local/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/camera/CameraMyntEye.h:107:27: error: template argument 1 is invalid
  std::vector<std::uint64_t> acc_;
                           ^
/usr/local/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/camera/CameraMyntEye.h:107:27: error: template argument 2 is invalid
rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/build.make:62: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/src/StereoCameraNode.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/src/StereoCameraNode.cpp.o] Error 1
CMakeFiles/Makefile2:5107: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_stereo_camera.dir/all] Error 2
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
```

the software env is: Ubuntu16.04+Ros Kinetic

what should I do to solve the problem? thank you in advance!

matlabbe, 2020-07-19:
Can you try with `catkin_make VERBOSE=1` to see compilation flags. It seems catkin is not building with c++11 support.

Here is an example of build on xenial+kinetic: https://travis-ci.org/github/introlab/rtabmap_ros/jobs/708450819

doomb007, 2020-07-20:
> Can you try with `catkin_make VERBOSE=1` to see compilation flags. It seems catkin is not building with c++11 support.
> 
> Here is an example of build on xenial+kinetic: https://travis-ci.org/github/introlab/rtabmap_ros/jobs/708450819

Thank you for your answer!

The output of `catkin_make VERBOSE=1` is a very long document, like this:
```
[ 93%] Linking CXX shared library /home/martin/catkin_ws/devel/lib/librtabmap_plugins.so
cd /home/martin/catkin_ws/build/rtabmap_ros && /usr/bin/cmake -E cmake_link_script CMakeFiles/rtabmap_plugins.dir/link.txt --verbose=1
/usr/bin/c++  -fPIC   -O3 -DNDEBUG   -shared -Wl,-soname,librtabmap_plugins.so -o /home/martin/catkin_ws/devel/lib/librtabmap_plugins.so CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_odometry.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/stereo_odometry.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbdicp_odometry.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/icp_odometry.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/data_throttle.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/stereo_throttle.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/data_odom_sync.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/point_cloud_xyzrgb.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/point_cloud_xyz.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/disparity_to_depth.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/pointcloud_to_depthimage.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/obstacles_detection.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/obstacles_detection_old.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/point_cloud_aggregator.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/point_cloud_assembler.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/undistort_depth.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/imu_to_tf.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_sync.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/stereo_sync.cpp.o CMakeFiles/rtabmap_plugins.dir/src/nodelets/rgbd_relay.cpp.o /home/martin/catkin_ws/devel/lib/librtabmap_ros.so /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_stitching3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_superres3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_videostab3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_aruco3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_bgsegm3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_bioinspired3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ccalib3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_cvv3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dpm3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_face3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_photo3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_fuzzy3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_hdf3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_img_hash3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_line_descriptor3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_optflow3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_reg3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_rgbd3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_saliency3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_stereo3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_structured_light3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_viz3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_phase_unwrapping3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_surface_matching3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_tracking3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_datasets3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_plot3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_text3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_dnn3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xfeatures2d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ml3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_shape3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_video3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_ximgproc3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_calib3d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_features2d3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_flann3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_highgui3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_videoio3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xobjdetect3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_objdetect3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_xphoto3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_imgproc3.so.3.3.1 /opt/ros/kinetic/lib/x86_64-linux-gnu/libopencv_core3.so.3.3.1 /usr/lib/x86_64-linux-gnu/libvtkImagingStencil-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkIOParallelNetCDF-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkLocalExample-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkTestingGenericBridge-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkIOMINC-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.2.so.6.2.0 /usr/lib/x86_64-linux-gnu/libvtkRenderingImage-6.2.so.6.2.0 /usr/lib/x86_64-linux- [... cut ...] r/lib/x86_64-linux-gnu/hdf5/openmpi/lib:/usr/lib/openmpi/lib:/usr/lib/x86_64-linux-gnu/hdf5/serial/lib: 
make[2]: Leaving directory '/home/martin/catkin_ws/build'
[ 93%] Built target rtabmap_plugins
make[1]: Leaving directory '/home/martin/catkin_ws/build'
Makefile:127: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make VERBOSE=1 -j8 -l8" failed
```
How can I confirm the problem from this text？

I‘ll try to follow the link you provided to reinstall RTABmap_ros next, thank you!

doomb007, 2020-07-21:
Hi,  matlabbe@matlabbe, the output of following this [example](https://travis-ci.org/github/introlab/rtabmap_ros/jobs/708450819) is as follow:

```
martin@martin-TM1701:~/catkin_ws$ catkin_make
Base path: /home/martin/catkin_ws
Source space: /home/martin/catkin_ws/src
Build space: /home/martin/catkin_ws/build
Devel space: /home/martin/catkin_ws/devel
Install space: /home/martin/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/martin/catkin_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/martin/catkin_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/martin/catkin_ws/devel;/opt/ros/kinetic
-- This workspace overlays: /home/martin/catkin_ws/devel;/opt/ros/kinetic
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: False
-- catkin 0.7.20
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 5 packages in topological order:
-- ~~  - realsense2_description
-- ~~  - beginner_tutorials
-- ~~  - rtabmap (plain cmake)
-- ~~  - realsense2_camera
-- ~~  - usb_cam
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
CMake Error at /opt/ros/kinetic/share/catkin/cmake/catkin_workspace.cmake:95 (message):
  This workspace contains non-catkin packages in it, and catkin cannot build
  a non-homogeneous workspace without isolation.  Try the
  'catkin_make_isolated' command instead.
Call Stack (most recent call first):
  CMakeLists.txt:67 (catkin_workspace)

-- Configuring incomplete, errors occurred!
See also "/home/martin/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/martin/catkin_ws/build/CMakeFiles/CMakeError.log".
Makefile:3274: recipe for target 'cmake_check_build_system' failed
make: *** [cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed
```

Is it because the source file contains the ROS2 package?

matlabbe, 2020-07-31:
Build/install rtabmap package outside the catkin workspace. See step 2 [here](https://github.com/introlab/rtabmap_ros#build-from-source), where `rtabmap` (not `rtabmap_ros`) is cloned outside the catkin workspace like a standard library.

doomb007, 2020-08-11:
Yes，rtabmap have been built outside the catkin workspace, and It can be run normally, but I always get errors when run `catkin_make -j1` 

the command all I used is as follow:
```
source /opt/ros/noetic/setup.bash
source ~/catkin_ws/devel/setup.bash
sudo apt install ros-noetic-rtabmap ros-noetic-rtabmap-ros
sudo apt remove ros-noetic-rtabmap ros-noetic-rtabmap-ros
(install g2o, GTSAM, libpointmatcher)
cd ~
git clone https://github.com/introlab/rtabmap.git rtabmap
cd rtabmap/build
cmake .. 
make
sudo make install
cd ~/catkin_ws
git clone https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
catkin_make -j1 -DRTABMAP_SYNC_MULTI_RGBD=ON -DRTABMAP_SYNC_USER_DATA=ON
```

Is the command I used wrong? @matlabbe