introlab/rtabmap_ros #442: Object Marker Map Location Wrong

thohemp, 2020-07-23:
I want add Objects Markers to the map similar to your thoughts here: http://official-rtab-map-forum.67519.x6.nabble.com/Detected-Object-Location-Mark-on-RTABMAP-tp4571p4594.html

I broadcast tf and ids like you do here: https://github.com/introlab/find-object/blob/master/src/ros/FindObjectROS.cpp#L58

I use UserData to save the objects in the nodes and to visualize them in rviz, following here: https://github.com/introlab/rtabmap_ros/blob/master/src/SaveObjectsExample.cpp

The objects are perfectly localized in camera frame, but when transforming them into odometry reference the visualized markers are wrong. Here, the keyboard marker should be on the keyboard.

[image]

My callback for adding markers (from https://github.com/introlab/rtabmap_ros/blob/master/src/SaveObjectsExample.cpp).
```
	// from rtabmap to rviz visualization
	void mapDataCallback(const rtabmap_ros::MapDataConstPtr &msg)
	{
		std::map<int, rtabmap::Signature> signatures;
		std::map<int, rtabmap::Transform> poses;
		std::multimap<int, rtabmap::Link> links;
		rtabmap::Transform mapToOdom;
		rtabmap_ros::mapDataFromROS(*msg, poses, links, signatures, mapToOdom);

		// handle the case where we can receive only latest data, or if all data are published
		for (std::map<int, rtabmap::Signature>::iterator iter = signatures.begin(); iter != signatures.end(); ++iter)
		{
			int id = iter->first;
			rtabmap::Signature &node = iter->second;

			nodeStamps_.insert(std::make_pair(node.getStamp(), node.id()));

			if (!node.sensorData().userDataCompressed().empty() && nodeToObjects_.find(id) == nodeToObjects_.end())
			{
				cv::Mat data = rtabmap::uncompressData(node.sensorData().userDataCompressed());
				ROS_ASSERT(data.cols == 9 && data.type() == CV_64FC1);
				ROS_INFO("Node %d has %d object(s)", id, data.rows);
				nodeToObjects_.insert(std::make_pair(id, data));
			}
		}

		// for the logic below, we should keep only stamps for
		// nodes still in the graph (in case nodes are ignored when not moving)
		std::map<double, int> nodeStamps;
		for (std::map<double, int>::iterator iter = nodeStamps_.begin(); iter != nodeStamps_.end(); ++iter)
		{
			std::map<int, rtabmap::Transform>::const_iterator jter = poses.find(iter->second);
			if (jter != poses.end())
			{
				nodeStamps.insert(*iter);
			}
		}

		// Publish markers accordingly to current optimized graph
		std::map<int, float> objectsAdded;
		visualization_msgs::MarkerArray markers;

		if (nodeToObjects_.size())
		{
			for (std::map<int, rtabmap::Transform>::iterator iter = poses.lower_bound(1); iter != poses.end(); ++iter)
			{
				if (nodeToObjects_.find(iter->first) != nodeToObjects_.end())
				{
					cv::Mat data = nodeToObjects_.at(iter->first);
					for (int i = 0; i < data.rows; ++i)
					{
						int objId = int(data.at<double>(i, 0));
						double stamp = data.at<double>(i, 1);

						// The object detection may have been taken between two nodes, interpolate its position.
						std::map<double, int>::iterator previousNode = nodeStamps.lower_bound(stamp); // lower bound of the stamp
						if (previousNode != nodeStamps.end() && previousNode->first > stamp && previousNode != nodeStamps.begin())
						{
							--previousNode;
						}
						std::map<double, int>::iterator nextNode = nodeStamps.upper_bound(stamp); // upper bound of the stamp

						if (previousNode != nodeStamps.end() &&
							nextNode != nodeStamps.end() &&
							previousNode->second != nextNode->second &&
							poses.find(previousNode->second) != poses.end() && poses.find(nextNode->second) != poses.end())
						{
							rtabmap::Transform poseA = poses.at(previousNode->second);
							rtabmap::Transform poseB = poses.at(nextNode->second);
							double stampA = previousNode->first;
							double stampB = nextNode->first;
							UASSERT(stamp >= stampA && stamp <= stampB);

							double ratio = (stamp - stampA) / (stampB - stampA);
							rtabmap::Transform robotPose = poseA.interpolate(ratio, poseB);

							// transform object pose in map frame
							rtabmap::Transform objPose(
								data.at<double>(i, 2), data.at<double>(i, 3), data.at<double>(i, 4),
								data.at<double>(i, 5), data.at<double>(i, 6), data.at<double>(i, 7), data.at<double>(i, 8));

							float distanceFromNode = objPose.getNorm();
							objPose = robotPose * objPose;
	
							if (objectsAdded.find(objId) == objectsAdded.end())
							{
								visualization_msgs::Marker marker;
								marker.header.frame_id = msg->header.frame_id;
								marker.header.stamp = msg->header.stamp;
								marker.ns = "objects";
								marker.id = objId;
								marker.action = visualization_msgs::Marker::ADD;
								marker.pose.position.x = objPose.x();
								marker.pose.position.y = objPose.y();
								marker.pose.position.z = objPose.z();
								Eigen::Quaterniond q = objPose.getQuaterniond();
								marker.pose.orientation.x = q.x();
								marker.pose.orientation.y = q.y();
								marker.pose.orientation.z = q.z();
								marker.pose.orientation.w = q.w();
								marker.scale.x = 0.1;
								marker.scale.y = 0.1;
								marker.scale.z = 0.1;
								marker.color.a = 0.8;
								marker.color.r = 0.0;
								marker.color.g = 1.0;
								marker.color.b = 0.0;

								ROS_INFO_STREAM("OBJ: " << objPose.x() << " "<< objPose.y() <<" "<< objPose.z());

								// cube
								marker.type = visualization_msgs::Marker::CUBE;
								markers.markers.push_back(marker);

								/*std::string name;
								if (objId == 1)
									name = "tvmonitor";
								else if (objId == 2)
									name = "cup";
								else
									name = "object";
								// text
								marker.type = visualization_msgs::Marker::TEXT_VIEW_FACING;
								marker.text = name.c_str();
								marker.id = -objId;
								marker.color.a = 1.0;
								marker.pose.position.z -= 0.3; // text over the cube*/
								//markers.markers.push_back(marker);

								objectsAdded.insert(std::make_pair(objId, distanceFromNode));
							}
							else
							{
								// update pose of the marker only if current observation is closer to robot
								if (distanceFromNode < objectsAdded.at(objId))
								{
									for (unsigned int j = 0; j < markers.markers.size(); ++j)
									{
										if (markers.markers[j].id == objId || markers.markers[j].id == -objId)
										{
											markers.markers[j].pose.position.x = objPose.x();
											markers.markers[j].pose.position.y = objPose.y();
											markers.markers[j].pose.position.z = objPose.z() + (markers.markers[j].id < 0 ? 0.3 : 0);
											Eigen::Quaterniond q = objPose.getQuaterniond();
											markers.markers[j].pose.orientation.x = q.x();
											markers.markers[j].pose.orientation.y = q.y();
											markers.markers[j].pose.orientation.z = q.z();
											markers.markers[j].pose.orientation.w = q.w();
										}
									}
									objectsAdded.at(objId) = distanceFromNode;
								}
							}
						}
					}
				}
			}
			if (!markers.markers.empty())
			{
				markersPub_.publish(markers);
			}
		}
```

matlabbe, 2020-07-23:
Hi, it is maybe the [frame_id](https://github.com/introlab/rtabmap_ros/blob/f318436647175866b8e51afe5365b25e66c96f36/src/SaveObjectsExample.cpp#L47) parameter of `save_objects_example` node that is not set correctly. It should be the base frame of the robot.

Here is a small example based on some general launch files:
```bash
$ roslaunch freenect_launch freenect.launch depth_registration:=true
$ roslaunch find_object_2d find_object_3d.launch # make sure odom and map static publishers are removed
$ rosrun rtabmap_ros save_objects_example _frame_id:=camera_link mapData:=/rtabmap/mapData
$ roslaunch rtabmap_ros rtabmap.launch args:="-d" user_data_async_topic:=/objectsData rviz:=true
```

thohemp, 2020-07-24:
This was exactly the problem. Thanks for your time and this great project.

closed 2020-07-24 by thohemp