introlab/rtabmap_ros #445: Multiple RGBD cameras SLAM with only one master camera for VO

szx0112, 2020-07-28:
I'd like to use multiple RGBD cameras for extended FOV. I noticed that when multiple cameras are used, RTABMAP only supports the 3D-3D registration for VO. Thus, I'd like to use only one camera for VO but all the cameras for mapping.

So in rgbd_odometry:
I remap the "rgbd_image" to the rgbd image from only one master camera, and set the number of rgbd_cameras as 1.
I also set to enable BA and 3D-2D pnp for single camera VO.

In rtabmap node:
The "rgbd_cameras" are set as 4 (number of cameras) for mapping using extended FOV
And remap the 'rgbd_image' from each camera as 'rgbd_imageindex"

When I execute the launch file. I got the error:
[ERROR] (2020-07-28 10:16:48.241) RegistrationVis.cpp:1302::computeTransformationImpl() Calibrated camera required (multi-cameras not supported). Id=-1 Models=0 StereoModel=0 weight=0

As far as I know, the error is related to that the 3D-2D pnp is not supported in the multi-camera mode. However, only one camera is used for the VO in my case. How should I modify the launch file in order to make it works?

Any comment is appreciated.

matlabbe, 2020-08-03:
It seems correct. Is the error coming from rgbd_odometry node or rtabmap node? For rtabmap node, make sure to add parameter `Vis/EstimationType` to 0.

closed 2021-11-12 by szx0112

hagen6835, 2023-07-29:
If im only using ICP Registration (```<param name="Reg/Strategy"   value="1"/>``` ), why should this value need to be set?

matlabbe, 2023-07-30:
What "this value" are you referring? If you are using ICP, `Vis/EstimationType` is used for visual loop closure detection even if you set `Reg/Strategy` to 1. If you don't have a multi-camera setup, you can ignore that parameter.

Note for the original question, `Vis/EstimationType=1` (default) can now work if rtabmap is built with [OpenGV](https://github.com/laurentkneip/opengv).

hagen6835, 2023-07-31:
The value i was referring to was the Vis/EstimationType. Knowing it is used for loop closure is quite insightful! We do have a multicamera setup, I will be attaching the launch file below

3 follow up questions then:

We followed the OpenGV installation [here](https://laurentkneip.github.io/opengv/page_installation.html) without any errors, however when building rtabmap from source, we still get "With OpenGV = NO (OpenGV not found)". How do we point rtabmap to opengv?

Is using Vis/EstimationType=1 (3D->2D (PnP)) a large improvement over 3D->3D? I only ask because we are having quite poor mapping performance that results in an incorrect map->odom transform more often than not. We also think that this may be due to poor loop closure but we are quite unsure.

Here is the launch file (we are using stereo rtabmap_odom fed into a ukf filter. I can include that launch file two if necessary). Does anything stick out as obviously causing problems? do you have any general suggestions or things to try?:

<launch>

    
    
    <group ns="rtabmap_global">
        <node pkg="rtabmap_slam" type="rtabmap" name="rtabmap" output="screen" args="--delete_db_on_start">
            
            <param name="subscribe_depth"                      value="false"/>
            <param name="subscribe_rgbd"                       value="true"/>
            <param name="rgbd_cameras"                         value="2"/>
            <param name="frame_id"                             value="base_link"/>

            
            <remap from="odom"          to="/mavros/odometry/out"/>
            <remap from="rgbd_image0"   to="/camera_front/rgbd_image"/>
            <remap from="rgbd_image1"   to="/camera_back/rgbd_image"/>
            
            
            
            
            
            
            <param name="gen_scan"                             value="true"/>
            
            
            
            <param name="map_always_update"                    value="true"/>        

            
            
            <param name="map_empty_ray_tracing"                value="true"/> 
            
            
            
            <param name="gen_depth_fill_holes_size"                    value="0"/>   
            
            
            
            <param name="gen_depth_fill_iterations"                    value="0.1"/>   
            
            
            
            <param name="gen_depth_fill_holes_error"                    value="1"/>   
            
            
            
            <param name="odom_sensor_sync"                    value="false"/>   
            
            
            
            <param name="map_filter_radius"                    value="0.0"/>   
            
            
            
            <param name="map_filter_angle"                    value="15.0"/>   

            

            
            
            
            
            <param name="Vis/EstimationType"            value="0"/>
            
            
            
            
            
            <param name="RGBD/NeighborLinkRefining"     value="true"/>
            
            
            
            <param name="RGBD/ProximityBySpace"         value="true"/>
            
            
            
            <param name="RGBD/AngularUpdate"            value="0.01"/>
            
            
            
            <param name="RGBD/LinearUpdate"             value="0.01"/>
            
            
            
            
            <param name="RGBD/OptimizeFromGraphEnd"     value="false"/>

            
            
            
            <param name="Grid/Sensor"                value="1"/>
            
            
            
            
            
            
            <param name="Grid/CellSize"                 value="0.025"/>
            
            
            <param name="Grid/RangeMax"                 value="3.0"/>
            
            
            <param name="Grid/NoiseFilteringRadius"                 value="1.0"/>
            
            
            <param name="Grid/RayTracing"                 value="True"/>

            
            
            
            <param name="Reg/Force3DoF"                 value="false"/>
            
            
            
            
            <param name="Reg/Strategy"                  value="1"/>

            
            
            <param name="Icp/Strategy"                 value="1"/>

            
            
            <param name="Icp/MaxTranslation"                 value="0.2"/>

            
            
            <param name="Icp/MaxRotation"                 value="0.78"/>

            
            
            <param name="Icp/VoxelSize"                 value="0.05"/>
            
            
            <param name="Icp/MaxCorrespondenceDistance" value="0.1"/>
        
        </node>
    </group>

</launch>

matlabbe, 2023-08-05:
For opengv, did you do `sudo make install`? You may also have to remove `-march=native` https://gist.githubusercontent.com/matlabbe/a412cf7c4627253874f81a00745a7fbb/raw/accc3acf465d1ffd0304a46b17741f62d4d354ef/opengv_disable_march_native.patch

3D->3D is very dependent to depth accuracy, while 2D->3D is less (usually better for cameras as far features can still be used for rotation estimation).

If you don't have a real lidar, I would set `gen_scan` to false. Set back `Reg/Strategy` to 0.

hagen6835, 2023-08-08:
Thank you, i was able to fix the opengv issue by just uninstalling and then following the steps outlined [here](https://github.com/introlab/rtabmap/issues/1084#issuecomment-1624727064).

after some initial tests, 2D->3D does generally seem to perform better with the multicamera setup as well as switching back to visual registration