introlab/rtabmap_ros #450: Rtabmap is not compatible with Ubuntu 18.04 PCL 1.11 in ROS melodic

Joechencc, 2020-08-06:
My environment is Ubuntu 18.04, PCL 1.11. The code works with Ubuntu 16.04, PCL 1.8. Can you please confirm this bug?

The error pops up like following: 

/home/chaonew/catkin_ws/src/rtabmap_ros/src/OdometryROS.cpp: In member function ‘void rtabmap_ros::OdometryROS::processData(const rtabmap::SensorData&, const ros::Time&)’:
/home/chaonew/catkin_ws/src/rtabmap_ros/src/OdometryROS.cpp:735:45: error: call of overloaded ‘PointXYZRGB(int, int, int)’ is ambiguous
     pcl::PointXYZRGB pt(inlier?0:255, 255, 0);
                                             ^
In file included from /usr/local/include/pcl-1.11/pcl/point_types.h:349:0,
                 from /usr/local/include/pcl-1.11/pcl/common/impl/copy_point.hpp:40,
                 from /usr/local/include/pcl-1.11/pcl/common/copy_point.h:58,
                 from /usr/local/include/pcl-1.11/pcl/common/impl/io.hpp:44,
                 from /usr/local/include/pcl-1.11/pcl/common/io.h:587,
                 from /usr/local/include/pcl-1.11/pcl/io/file_io.h:41,
                 from /usr/local/include/pcl-1.11/pcl/io/pcd_io.h:45,
                 from /home/chaonew/catkin_ws/src/perception_pcl/pcl_conversions/include/pcl_conversions/pcl_conversions.h:70,
                 from /home/chaonew/catkin_ws/src/rtabmap_ros/src/OdometryROS.cpp:35:
/usr/local/include/pcl-1.11/pcl/impl/point_types.hpp:641:12: note: candidate: pcl::PointXYZRGB::PointXYZRGB(float, float, float)
     inline PointXYZRGB (float _x, float _y, float _z):
            ^~~~~~~~~~~
/usr/local/include/pcl-1.11/pcl/impl/point_types.hpp:638:12: note: candidate: pcl::PointXYZRGB::PointXYZRGB(uint8_t, uint8_t, uint8_t)
     inline PointXYZRGB (std::uint8_t _r, std::uint8_t _g, std::uint8_t _b):
            ^~~~~~~~~~~
rtabmap_ros/CMakeFiles/rtabmap_ros.dir/build.make:110: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/src/OdometryROS.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:8378: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_ros.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

                                             ^

Joechencc, 2020-08-06:
I fixed the issue by looking at #498 in Rtabmap. But can you please push the code that I do not need to change the source code

matlabbe, 2020-08-06:
Should be ok with the commit above. Can you confirm it works? I cannot test on a PCL 1.11 build right now.

closed 2020-08-06 by matlabbe

Joechencc, 2020-08-06:
I pulled the code to the newest version, there is still some bug like following:

error: no matching function for call to ‘rtabmap::Rtabmap::getSignatureCopy(int&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_images_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_scan_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_user_data_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_grid_type&)’
   Signature s = rtabmap_.getSignatureCopy(id, req.images, req.scan, req.user_data, req.grid);

In file included from /home/chaonew/catkin_ws/src/rtabmap_ros/include/rtabmap_ros/CoreWrapper.h:49:0,
                 from /home/chaonew/catkin_ws/src/rtabmap_ros/src/CoreWrapper.cpp:28:
/usr/local/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Rtabmap.h:183:12: note: candidate: rtabmap::Signature rtabmap::Rtabmap::getSignatureCopy(int, bool, bool, bool, bool, bool, bool) const
  Signature getSignatureCopy(int id, bool images, bool scan, bool userData, bool occupancyGrid, bool withWords, bool withGlobalDescriptors) const;
            ^~~~~~~~~~~~~~~~
/usr/local/lib/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Rtabmap.h:183:12: note:   candidate expects 7 arguments, 5 provided
rtabmap_ros/CMakeFiles/rtabmap_sync.dir/build.make:230: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
CMakeFiles/Makefile2:6867: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_sync.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2020-08-06:
Make sure to update rtabmap library and reinstall it.

commit e96f72d4 (2020-08-06): fixed #450
PR #471 (2020-10-07): updating files to build ros package with rtab-map lib v0.20.3
Hi. While following the README in the "ros2" branch I faced two issues. The first one regards the PCL version. For PCL version equals or greater than 10.1, the change made in 'src/OdometryROS.cpp' is necessary, which is the same modification made to fix #450 but in the "melodic" branch. Second, the latest version of RTAB-LIB (v0.20.3) has changed the variable type of the method 'graph::findNearestNodes', which justifies the modification made in 'src/CoreWrapper.cpp' (please refer to #470).
After these changes, I could successfully b [... cut]