introlab/rtabmap_ros #452: Is there a way for saving the octomap as bt file

NicoNorena, 2020-08-08:
question:
- I saw the Octomap Class can save the octree to a given path, Is there a way call it from rtabmap_ros?  
for example if I want to have the map_assembler_node on the  client device and at some point save the OctoMap as a .bt file to a given path.

NicoNorena, 2020-08-25:
@matlabbe sure I have used the  http://wiki.ros.org/octomap_server#octomap_saver
but I would have to subscribe an octomap_server running subscribed to the rtabmap_map_assembler.
I thought there was a way of calling 
https://github.com/introlab/rtabmap/blob/775c9318d333aa09a00ee1ae06c6664eb1b83265/corelib/src/OctoMap.cpp#L1155
from rtabmap_map_assembler, I am assuming this function is equivalent to octomap_saver

matlabbe, 2020-08-25:
You don't need `octomap_server` and `rtabmap_map_assembler` nodes running. `rtabmap` node itself works like an `octomap_server`. Just do:
```bash
$ rosrun octomap_server octomap_saver mapfile.bt octomap_binary:=/rtabmap/octomap_binary
```
To debug, you can convert the octomap file to a point cloud with this octomap tool:
```
$ octree2pointcloud ~/catkin_ws/mapfile.bt cloud.pcd
$ pcl_viewer cloud.pcd
```

NicoNorena, 2020-09-04:
thanks, I tested and it worked.
it saved the section of the map that was in the Octomap from Rtabmap.
 how do I pull the local map and have it update the Octomap  managed by Rtabmap ? so it can save the local or global map.

closed 2020-10-12 by NicoNorena

valmil2000, 2021-07-21:
Hey, I wondered whether one of you could help me how I can save a octomap included in a python script?

matlabbe, 2021-08-18:
octomap_saver node doesn't work for you? http://wiki.ros.org/octomap_server#octomap_saver

NicoNorena, 2021-08-18:
octomap can be save using the octomap_saver. Essentially, the ROS node takes the point cloud from the Rtabmap topic and builds a secondary map registration, the package octomap_saver has a service call or rosrun  ( i cant remember ) to save it. I don't think it is a python script.

arjunskumar, 2022-03-17:
@matlabbe How to install octree2pointcloud?

matlabbe, 2022-03-17:
It seems here: https://github.com/OctoMap/octomap/blob/7a9ab4a357860f7263769cce9dd45107ec8b2c4a/octomap/src/CMakeLists.txt#L67-L68

arjunskumar, 2022-03-17:
@matlabbe Thanks