introlab/rtabmap_ros #456: Catkin build rtabmap_ros fails with varrying errors.

seanroelofs, 2020-08-27:
Hello, I am trying to get rtabmap working with ros on a jetson xavier nx. I compiled gtsam and rtabmap from source. I followed the steps in this issue (https://github.com/introlab/rtabmap/issues/427#issuecomment-608052821). However, when I run catkin build, the rtabmap_ros package fails. I have tried it multiple times and seen it fail for two different reasons. It failed for each reason twice. The first error I get is as follows: 

```
.......................................................................................................................
_______________________________________________________________________________________________________________________
Errors     << rtabmap_ros:make /home/asl/catkin_ws/logs/rtabmap_ros/build.make.002.log                                 
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Marginals::Marginals(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::Marginals::Factorization, boost::optional<gtsam::Ordering const&>)'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/asl/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/rgbdicp_odometry] Error 1
make[1]: *** [CMakeFiles/rtabmap_rgbdicp_odometry.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Marginals::Marginals(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::Marginals::Factorization, boost::optional<gtsam::Ordering const&>)'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/asl/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/stereo_odometry] Error 1
make[1]: *** [CMakeFiles/rtabmap_stereo_odometry.dir/all] Error 2
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Marginals::Marginals(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::Marginals::Factorization, boost::optional<gtsam::Ordering const&>)'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot3::operator*(gtsam::Point3 const&) const'
collect2: error: ld returned 1 exit status
make[2]: *** [/home/asl/catkin_ws/devel/.private/rtabmap_ros/lib/rtabmap_ros/rgbd_odometry] Error 1
make[1]: *** [CMakeFiles/rtabmap_rgbd_odometry.dir/all] Error 2
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::equals(gtsam::Point2 const&, double) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::range(gtsam::Point3 const&, gtsam::OptionalJacobian<1, 6>, gtsam::OptionalJacobian<1, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point2::print(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Marginals::Marginals(gtsam::NonlinearFactorGraph const&, gtsam::Values const&, gtsam::Marginals::Factorization, boost::optional<gtsam::Ordering const&>)'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Rot2::rotate(gtsam::Point2 const&, gtsam::OptionalJacobian<2, 1>, gtsam::OptionalJacobian<2, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::bearing(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose3::bearing(gtsam::Point3 const&, gtsam::OptionalJacobian<2, 6>, gtsam::OptionalJacobian<2, 3>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Pose2::range(gtsam::Point2 const&, gtsam::OptionalJacobian<1, 3>, gtsam::OptionalJacobian<1, 2>) const'
/usr/local/lib/librtabmap_core.so: undefined reference to `gtsam::Point3::equals(gtsam::Point3 const&, double) const'
[... cut ...]
make[2]: *** [CMakeFiles/rtabmap_sync.dir/src/CoreWrapper.cpp.o] Error 1
make[1]: *** [CMakeFiles/rtabmap_sync.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
cd /home/asl/catkin_ws/build/rtabmap_ros; catkin build --get-env rtabmap_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................
Failed     << rtabmap_ros:make                                [ Exited with code 2 ]
```

seanroelofs, 2020-08-27:
Update: Okay so I solved my own problem. I had built rtabmap on my system multiple times, however make install did not overwrite old binary files with the new ones. So i did make uninstall for rtabmap, then make install, then tried to build rtabmap_ros and it all worked.

closed 2020-08-27 by seanroelofs

flabrosse, 2020-12-08:
Same problem here with the getSignatureCopy() function:
```
/var/tmp/portage/ros-noetic/rtabmap_ros-0.20.0-r3/work/rtabmap_ros-0.20.0/src/CoreWrapper.cpp: In member function ‘bool rtabmap_ros::CoreWrapper::getNodeDataCallback(rtabmap_ros::GetNodeData::Request&, rtabmap_ros::GetNodeData::Respons
e&)’:
/var/tmp/portage/ros-noetic/rtabmap_ros-0.20.0-r3/work/rtabmap_ros-0.20.0/src/CoreWrapper.cpp:2675:92: error: no matching function for call to ‘rtabmap::Rtabmap::getSignatureCopy(int&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_images_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_scan_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_user_data_type&, rtabmap_ros::GetNodeDataRequest_<std::allocator<void> >::_grid_type&)’
 2675 |   Signature s = rtabmap_.getSignatureCopy(id, req.images, req.scan, req.user_data, req.grid);
      |                                                                                            ^
In file included from /var/tmp/portage/ros-noetic/rtabmap_ros-0.20.0-r3/work/rtabmap_ros-0.20.0/include/rtabmap_ros/CoreWrapper.h:49,
                 from /var/tmp/portage/ros-noetic/rtabmap_ros-0.20.0-r3/work/rtabmap_ros-0.20.0/src/CoreWrapper.cpp:28:
/usr/lib64/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Rtabmap.h:183:12: note: candidate: ‘rtabmap::Signature rtabmap::Rtabmap::getSignatureCopy(int, bool, bool, bool, bool, bool, bool) const’
  183 |  Signature getSignatureCopy(int id, bool images, bool scan, bool userData, bool occupancyGrid, bool withWords, bool withGlobalDescriptors) const;
      |            ^~~~~~~~~~~~~~~~
/usr/lib64/rtabmap-0.20/../../include/rtabmap-0.20/rtabmap/core/Rtabmap.h:183:12: note:   candidate expects 7 arguments, 5 provided
```
This is after a fresh install of rtabmap-0.20.3 and trying to install rtabmap_ros-0.20.0 on gentoo.

matlabbe, 2020-12-10:
If you use rtabmap-0.20.3, you should use matching rtabmap_ros version (or master).

flabrosse, 2020-12-11:
Thanks for this.  However, according to github, the latest release of rtabmap-ros is 0.20.0.  I'll try to blindly generate a package and see what happens.

matlabbe, 2020-12-11:
0.20.3 for rtabmap_ros was not tagged. You would have to use rtabmap 0.20.0 to make sure it is compatible with rtabmap_ros 0.20.0, or use both master branches.

flabrosse, 2020-12-16:
Well, rtabmap-0.20.0 is not available any more (if it ever has). so the only release of rtabmap_ros cannot be used.  The only way forward at the moment is to go for potentially unstable code from the master branches.  Could a newer version of rtabmap_ros be made available?

matlabbe, 2020-12-16:
For 0.20.0, tags are here:
https://github.com/introlab/rtabmap/tree/0.20.0-noetic
https://github.com/introlab/rtabmap_ros/tree/0.20.0-noetic

Since this weekend, I released new kinetic/melodic/noetic versions (0.20.7). The ros binaries should then upgrade to 0.20.7 in a couple of days/weeks. But you can build them from source.
https://github.com/introlab/rtabmap/tree/0.20.7-noetic
https://github.com/introlab/rtabmap_ros/tree/0.20.7-noetic

flabrosse, 2020-12-16:
Thank you.  I could not find the 0.20.0 tags on github on the release page.

And thank you for the updates.  I'll produce packages for thiem.

matlabbe, 2020-12-25:
Well, you can click on Tags a the top of the Releases page:
https://github.com/introlab/rtabmap/tags

umbnich, 2023-04-02:
How I can be sure to have uninstalled rtabmap/rtabmap-ros to have  a fresh install?

matlabbe, 2023-04-02:
If you built from source, `rtabmap` installs by default in `/usr/local/[include, lib, bin]`. For `rtabmap_ros`, it would be in your catkin workspace `devel` folder. 

for the binaries, which distro you are using? `sudo apt remove ros-$ROS_DISTRO-rtabmap`

umbnich, 2023-04-04:
I'm using ROS Noetic on Ubuntu 20.04. 
So, you suggest to uninstall everything and try to have a fresh install as in the guide? I want to build from source. 
For clearance, I also have ORB-SLAM3 and the ZED SDK. Do you believe that could be compatibility issues?

matlabbe, 2023-04-16:
It seems there is an [issue](https://github.com/introlab/rtabmap/issues/1013) with latest ZED SDK 4 right now, so if you build from source, do `cmake -DWITH_ZED=OFF ..`

For ORB_SLAM3, by default rtabmap doesn't build with it, so that's fine to keep it.

umbnich, 2023-04-17:
Is possibile to install the binary and use ORB_SLAM3?

matlabbe, 2023-04-20:
No, not if you want to use ORB_SLAM3 inside rtabmap.