introlab/rtabmap_ros #457: Map updates causing jitter of robot

AWilson10, 2020-08-29:
Is it possible to pause / switch off localisation without terminating rtab. Or if we could increase a parameter like MaxInliers to prevent the map getting updated dynamically? We are running in localisation mode, but find that once localised the robot jumps around / jitters by a few centimeters each time there is an update or loop closure. We hoped to freeze this and unfreeze. Is this possible?

matlabbe, 2020-08-29:
You can call `/rtabmap/pause` and `/rtabmap/resume` services. The jumps in `/map` frame follows REP105: https://www.ros.org/reps/rep-0105.html#map
If you need to do local planning, use `/odom` frame instead, which should be continuous.

AWilson10, 2020-08-30:
Thank-you, that is most helpful