introlab/rtabmap_ros #458: Remotee RVIZ RAM overflowing easily while subscribed to /rtabmap/mapData

tianshiz, 2020-08-31:
I am running rviz on a remote computer and subscribed only to mapData. I am using 3 VGA cameras at 15fps for rtabmap.

On startup, with the robot stationary, my rviz computer's ram jumps from 6gb to 12gb in a few minute time. Eventually it overflows my 16gb memory and that causes all sort of issues. 
However, I noticed, that if I unsubscribe from the mapData and then resubscribe to it again, the ram usage plateaus.

Is something wrong here? I set the following:
Mem:
    BinDataKept: true
    ImagePostDecimation: 0 #save memory

why is it causing RAM issue on a remote machine?

matlabbe, 2020-09-01:
15 FPS is quite high for mapping, memory with grow very fast. For RVIZ MapCloud plugin, you can increase the decimation parameter to show less points. With decimation at 1, you are creating 45 640x480 points clouds each second, which is approximately (XYZ+RGB, 4 float per point or 4 bytes per point) 45x640x480x4 = 55 MB per second, like 3 GB per minute! 

In general, we set a frame rate of 1 Hz for rtabmap, and we use a decimation of 4 per default on RVIZ side. For your setup, it would then be 3x160x120x4 = 230KB per second or 13 MB per minute for RVIZ rendering.

tianshiz, 2020-09-01:
rgbd_sync is subscribed to cameras that publish at 15hz, it looks like /rtabmap/mapData is published at 1Hz. So I don't think I'm messing things up there. Unless you mean something else?

For better visualization, I'm trying to remote teleop by relying partly on mapData. Decimation really messes up the view. Maybe a little is ok.

I think I'm not quite understanding what happens with mapData here. I can see a constant size stream of data being published by the robot regardless of the RAM issue. Bmon tells me about 16Mbps at decimation=1. It is my understanding that mapData being published does not change in size.

What I don't understand is why on the RVIZ side, mapData is accumulating in size? In my stationary example, I don't really care about the mapData from a few seconds ago. Can that data be trashed when a more update mapData for that spot is received?

maybe the answer to this is just to lower the decimation, and just have the decimated view update become more dense as the robot moves? Regardless, it sounds like eventually mapData will reach memory cap. In those cases, what can be done?

matlabbe, 2020-09-01:
Sorry, I've read too fast, 15 fps was for the input, it is okay then.

`/rtabmap/mapData` is almost always constant in size. It will increase over time based on the graph (each node has a couple of bytes), not the data. `/rtabmap/mapData` contains only data from the latest node added to map. MapCloud RVIZ plugin buffers the data to be shown. Currently, there is no way to tell MapCloud plugin to drop data, otherwise than unchecking the plugin then rechecking it. On rtabmap side, by default if the robot is not moving, new data are discarded (so RAM would be stable). However on MapCloud side, it cannot know that the new node in `/rtabmap/mapData` has been discarded. I implemented a solution in this commit: https://github.com/introlab/rtabmap_ros/commit/cdc38640ec035e13f319a962ac974c406364f076. Note that if decimation is decimating too much for long range sensors, try increasing voxel parameter instead.

If you are interested in only the latest cloud to show, you could create a `rtabmap_ros/point_cloud_xyzrgb` nodelet on remote side, subscribed to `rgbd_image/compressed` output of rgbd_sync (rgbd_sync has also a `compressed_rate` parameter to limit the output frame rate of the compressed topic for convenience). You may show this high resolution point cloud while showing MapCloud above in low resolution.

tianshiz, 2020-09-02:
Hi,
Really appreciate your efforts to address these issues for me.
I'll give that a try.

Are there any similar RAM limitations on the robot side? I would assume an infinity large map would overflow?
 How do I calculate the RAM needed for my 3 cameras given a traversal around an area size W*L?

Thank you

matlabbe, 2020-09-03:
From a RAM point of view, the RAM is limited by the size of the working memory. If `RTabmap/TimeThr` or `Rtabmap/MemoryThr` are set, the RAM will eventually remains stable. If they are not set, the RAM will increase with the number of nodes in working memory. It would increase less faster than what you saw on rviz side because we keep only visual words and local occupancy grids in RAM, the compressed RGB and depth images are saved to database (hard-drive). The local occupancy grids (see `Grid/` parameters) have a voxel size of 5 cm by default, so they don't take as much space than your clouds in rviz. The RAM usage could be estimated by Bytes/node instead of Bytes/m^2. The RAM usage per node depends on the type of descriptors used, the number per frame and local occupancy grid parameters. You can do `rtabmap-info map.db` to see how memory is split in rtabmap.

Note that if you switch in localization mode, the RAM will stay stable too.

closed 2020-09-03 by tianshiz