introlab/rtabmap_ros #459: Dual Camera ROS RTAB-MAP not working

Sd1010, 2020-09-03:
Hi,

I am trying to use 2 Realsense D435I cameras to generate point cloud map with ROS RTAB-MAP. When i run the launch file I can see the RVIZ opening but it does not show any point cloud or anything else. I am not sure what is wrong here. 

Any help would greatly appreciated. 

**Launch file**
```
`<launch>

  <!-- Multi-cameras demo with 2 D400 cameras -->

  <!-- Choose visualization -->
  <arg name="rviz"       default="true" />
  <arg name="rtabmapviz" default="false" />

  <!-- Cameras -->
  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="camera1" />
    <arg name="serial_no" value="222222222222" />
    <arg name="align_depth" value="true" />
  </include>

  <include file="$(find realsense2_camera)/launch/rs_camera.launch">
    <arg name="camera" value="camera2" />
    <arg name="serial_no" value="11111111111111" />
    <arg name="align_depth" value="true" />
  </include>

  <!-- Frames: Cameras are placed at 90 degrees, clockwise x, y, z , Rz, Ry, Rx -->
  <node pkg="tf" type="static_transform_publisher" name="base_to_camera1_tf"
      args=" 0.0 0.0 0.7096 0.0 0.6466 1.5708 /base_link /camera1_link 100" />
  <node pkg="tf" type="static_transform_publisher" name="base_to_camera2_tf"
      args="0.0 0.0 0.0 0.0 0.6004 1.5708 /base_link /camera2_link 100" />

   <!-- sync rgb/depth images per camera -->
   <group ns="camera1">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager1" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync1" args="load rtabmap_ros/rgbd_sync nodelet_manager1">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>
   <group ns="camera2">
    <node pkg="nodelet" type="nodelet" name="nodelet_manager2" args="manager"/>
    <node pkg="nodelet" type="nodelet" name="rgbd_sync2" args="load rtabmap_ros/rgbd_sync nodelet_manager2">
      <remap from="rgb/image"       to="color/image_raw"/>
      <remap from="depth/image"     to="aligned_depth_to_color/image_raw"/>
      <remap from="rgb/camera_info" to="color/camera_info"/>
      <param name="approx"          value="false"/>
    </node>
   </group>

  <group ns="rtabmap">

    <!-- Odometry -->
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <remap from="rgbd_image0"       to="/camera1/rgbd_image"/>
      <remap from="rgbd_image1"       to="/camera2/rgbd_image"/>

      <param name="subscribe_rgbd"           type="bool"   value="true"/>
      <param name="frame_id"                 type="string" value="base_link"/>
      <param name="rgbd_cameras"             type="int"    value="2"/>
      <param name="Vis/EstimationType"       type="string" value="0"/> <!-- should be 0 for multi-cameras -->
    </node>

    <!-- Visual SLAM (robot side) -->
    <!-- args: "delete_db_on_start" and "udebug" -->
    <node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" output="screen" args="--delete_db_on_start">
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="rgbd_cameras"     type="int"    value="2"/>
      <param name="frame_id"         type="string" value="base_link"/>
      <param name="gen_scan"         type="bool"   value="true"/>
      <param name="map_negative_poses_ignored" type="bool"   value="false"/>        <!-- refresh grid map even if we are not moving-->
      <param name="map_negative_scan_empty_ray_tracing" type="bool" value="false"/> <!-- don't fill empty space between the generated scans-->

      <remap from="rgbd_image0"       to="/camera1/rgbd_image"/>
      <remap from="rgbd_image1"       to="/camera2/rgbd_image"/>

      <param name="Grid/FromDepth"     type="string" value="true"/>
      <param name="Vis/EstimationType" type="string" value="0"/> <!-- should be 0 for multi-cameras -->
    </node>

    <!-- Visualisation RTAB-Map -->
    <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
      <param name="subscribe_depth"  type="bool"   value="false"/>
      <param name="subscribe_rgbd"   type="bool"   value="true"/>
      <param name="subscribe_odom_info" type="bool"   value="true"/>
      <param name="frame_id"            type="string" value="base_link"/>
      <param name="rgbd_cameras"       type="int"    value="2"/>

      <remap from="rgbd_image0"       to="/camera1/rgbd_image"/>
      <remap from="rgbd_image1"       to="/camera2/rgbd_image"/>
    </node>

  </group>

  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(find rtabmap_ros)/launch/config/rgbd.rviz"/>

</launch>
```
`
**Terminal Output** - This is a section of terminal output. It keeps giving error " Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)"

```
[ INFO] [1599153972.781399295]: rtabmap: Database version = "0.20.0".
[ WARN] [1599153972.793585586]: rtabmap: Parameters subscribe_rgb and subscribe_rgbd cannot be true at the same time. Parameter subscribe_rgb is set to false.
[ INFO] [1599153972.794933127]: /rtabmap/rtabmap: subscribe_depth = false
[ INFO] [1599153972.794959036]: /rtabmap/rtabmap: subscribe_rgb = false
[ INFO] [1599153972.794969271]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1599153972.794978883]: /rtabmap/rtabmap: subscribe_rgbd = true (rgbd_cameras=2)
[ INFO] [1599153972.794989159]: /rtabmap/rtabmap: subscribe_odom_info = false
[ INFO] [1599153972.794998292]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1599153972.795016017]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1599153972.795027657]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1599153972.795039757]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1599153972.795051776]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1599153972.795060784]: /rtabmap/rtabmap: approx_sync   = true
**[FATAL] [1599153972.795070017]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)
[ INFO] [1599153972.795132364]: 
[ERROR] (2020-09-03 13:26:12.813) OdometryF2M.cpp:243::computeTransform() Odometry bundle adjustment doesn't work with multi-cameras. It is disabled.**
[ INFO] [1599153972.831279307]: Odom: quality=0, std dev=99.995000m|99.995000rad, update time=0.017631s
[ INFO] [1599153972.883611635]: rtabmap 0.20.0 started...
[ WARN] (2020-09-03 13:26:12.916) RegistrationVis.cpp:1126::computeTransformationImpl() Finding correspondences with the guess cannot be done with multiple cameras, global matching is done instead. Please set "Vis/CorGuessWinSize" to 0 to avoid this warning.
[ INFO] [1599153972.919051409]: /rtabmap/rtabmapviz: subscribe_depth = false
[ INFO] [1599153972.919076153]: /rtabmap/rtabmapviz: subscribe_rgb = false
[ INFO] [1599153972.919082755]: /rtabmap/rtabmapviz: subscribe_stereo = false
[ INFO] [1599153972.919089182]: /rtabmap/rtabmapviz: subscribe_rgbd = true (rgbd_cameras=2)
[ INFO] [1599153972.919094879]: /rtabmap/rtabmapviz: subscribe_odom_info = true
[ INFO] [1599153972.919102553]: /rtabmap/rtabmapviz: subscribe_user_data = false
[ INFO] [1599153972.919108665]: /rtabmap/rtabmapviz: subscribe_scan = false
[ INFO] [1599153972.919114727]: /rtabmap/rtabmapviz: subscribe_scan_cloud = false
[ INFO] [1599153972.919120299]: /rtabmap/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1599153972.919125596]: /rtabmap/rtabmapviz: queue_size    = 10
[ INFO] [1599153972.919130804]: /rtabmap/rtabmapviz: approx_sync   = true
[FATAL] [1599153972.919139188]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)
[ INFO] [1599153972.919200801]: 
[ INFO] [1599153972.919249582]: rtabmapviz started.
[ INFO] [1599153972.943313008]: Odom: quality=416, std dev=0.004184m|0.004184rad, update time=0.069693s
[ WARN] (2020-09-03 13:26:12.988) RegistrationVis.cpp:1126::computeTransformationImpl() Finding correspondences with the guess cannot be done with multiple cameras, global matching is done instead. Please set "Vis/CorGuessWinSize" to 0 to avoid this warning.
```

matlabbe, 2020-09-03:
The binary version doesn't have multi camera support (for some long compilation issues on ros buildfarm). Uninstall `ros-[distro]-rtabmap-ros` and clone [rtabmap_ros](https://github.com/introlab/rtabmap_ros) repo in your catkin_ws with the same branch than your installed rtabmap binary version.
```
$ sudo apt install ros-$ROS_DISTRO-rtabmap
$ cd ~/catkin_ws/src
$ git clone -b $ROS_DISTRO-devel https://github.com/introlab/rtabmap_ros.git
$ cd ~/catkin_ws
$ catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON 
```

cheers,
Mathieu

Sd1010, 2020-09-09:
@matlabbe Thank you very much. It solved my problem.

closed 2020-09-09 by matlabbe

ju-mingyue, 2021-07-02:
(https://github.com/introlab/rtabmap_ros)库。
> 
> ```shell
> $ sudo apt remove ros- $ROS_DISTRO -rtabmap-ros
> $ cd  ~ /catkin_ws/src
> $ git clone -b $ROS_DISTRO -devel https://github.com/introlab/rtabmap_ros.git
> $ cd  ~ /catkin_ws
> $ catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON 
> ```

Hello, I am very happy to see your guidance. After I completed the operation according to your method, the following problems occurred. Can you give me some pointers?
[image]

snakehaihai, 2022-09-20:
looks like memory overflow at upstairs. try to use a better PC

matlabbe, 2022-10-24:
For errors like: `c++: internal compiler error: Killed (program cc1plus)`, it usually means that your RAM is full or other compilation problem. Try rebuild with only one process:
```
catkin_make -j1
```

alvaropascuaI, 2022-11-23:
How i can use this but for ros2 ?

matlabbe, 2022-11-27:
IF you see on installation instructions on ros2 branch: https://github.com/introlab/rtabmap_ros/tree/ros2#installation:
```
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
```
set this to `-j1` before doing `colcon`.