introlab/rtabmap_ros #460: rtabmap node crashes when using existing database

sKlairRad, 2020-09-10:
Hi 

I've built rtabmap and rtabmap_ros from source. When I try to launch rtabmap in localization mode using the database created during mapping, it crashes with the following error:
`rtabmap: malloc.c:2401: sysmalloc: Assertion `(old_top == initial_top (av) && old_size == 0) || ((unsigned long) (old_size) >= MINSIZE && prev_inuse (old_top) && ((unsigned long) old_end & (pagesize - 1)) == 0)' failed.`

I can visualize the database in rtabmap-databaseViewer without any issues. Also, it launches fine with `--delete_db_on_start` flag.

matlabbe, 2020-09-10:
Try section B of this post: https://github.com/introlab/rtabmap_ros/issues/28, it will show more info about which functions were called before malloc.

sKlairRad, 2020-09-10:
I tried your suggestion and got the following: 

[image]

Not sure how to fix it.

matlabbe, 2020-09-10:
Can you show output of cmake of rtabmap? Which OS/computer are you using?
```
$ cd rtabmap/build
$ cmake ..
```

sKlairRad, 2020-09-11:
I'm using Ubuntu 18.04.
Here is the output of cmake:
`CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- PCL definitions don't contain "-march=native", make sure all libraries using Eigen are also compiled without that flag to avoid some segmentation faults (with gdb referring to some Eigen functions).
-- Found Sqlite3: /usr/include /usr/lib/x86_64-linux-gnu/libsqlite3.so
-- Found OpenMP_C: -fopenmp  
-- Found OpenMP_CXX: -fopenmp  
-- Found OpenMP: TRUE   
-- Found OpenMP
-- Found OpenCV: /usr/local/include/opencv4
-- Found PCL: /usr/include/pcl-1.8;/usr/include/eigen3;/usr/include;/usr/include/ni;/usr/include/openni2;/usr/include/vtk-6.3;/usr/include/freetype2;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent;/usr/lib/x86_64-linux-gnu/openmpi/include/openmpi/opal/mca/event/libevent2022/libevent/include;/usr/lib/x86_64-linux-gnu/openmpi/include;/usr/include/python2.7;/usr/include/x86_64-linux-gnu;/usr/include/hdf5/openmpi;/usr/include/libxml2;/usr/include/jsoncpp;/usr/include/tcl
-- Found ZLIB: /usr/include
-- Could NOT find Qt5Svg (missing: Qt5Svg_DIR)
CMake Warning at /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:39 (message):
  Failed to find Qt5 component "Svg" config file at
  "/usr/lib/x86_64-linux-gnu/cmake/Qt5Svg/Qt5SvgConfig.cmake"
Call Stack (most recent call first):
  CMakeLists.txt:283 (FIND_PACKAGE)

-- VTK_RENDERING_BACKEND=OpenGL
-- Found Freenect: /usr/include
-- Found OpenNI2: /usr/include/openni2
-- Found DC1394: /usr/include/dc1394
CMake Warning at cmake_modules/FindG2O.cmake:122 (MESSAGE):
  Latest g2o version detected with c++11 interface (config file:
  /opt/ros/melodic/include/g2o/config.h).  Make sure g2o is built with
  "-DBUILD_WITH_MARCH_NATIVE=OFF" to avoid segmentation faults caused by
  Eigen.
Call Stack (most recent call first):
  CMakeLists.txt:402 (FIND_PACKAGE)

-- Found g2o: /opt/ros/melodic/include;/usr/include/suitesparse;/usr/include/suitesparse
-- FlyCapture2_INCLUDE_DIR=FlyCapture2_INCLUDE_DIR-NOTFOUND
-- FlyCapture2_LIBRARY=FlyCapture2_LIBRARY-NOTFOUND
-- Triclops_INCLUDE_DIR=Triclops_INCLUDE_DIR-NOTFOUND
-- Triclops_LIBRARY=Triclops_LIBRARY-NOTFOUND
-- FlyCaptureBridge_LIBRARY=FlyCaptureBridge_LIBRARY-NOTFOUND
-- A library with BLAS API found.
-- Found CUDA: /usr/local/cuda (found suitable version "11.0", minimum required is "11.0") 
-- Found TensorRT (found version "7.1")
-- Found ZED sdk: /usr/local/zed/include
-- Found CUDA: /usr/local/cuda (found version "11.0") 
-- Found CUDA: /usr/local/cuda/include
-- Found octomap 1.9.0: /opt/ros/melodic/include
[... 154 lines cut ...]
--   With Freenect2            = NO (libfreenect2 not found)
--   With Kinect for Windows 2 = NO (Kinect for Windows 2 SDK not found)
--   With Kinect for Azure     = NO (Kinect for Azure SDK not found)
--   With dc1394               = YES (License: LGPL)
--   With FlyCapture2/Triclops = NO (Point Grey SDK not found)
--   With ZED                  = YES
--   With RealSense            = NO (librealsense not found)
--   With RealSense2           = NO (librealsense2 not found)
--   With MyntEyeS             = NO (mynteye s sdk not found)
-- 
--  Odometry Approaches:
--   With loam_velodyne        = NO (loam_velodyne not found)
--   With libfovis             = NO (libfovis not found)
--   With libviso2             = NO (libviso2 not found)
--   With dvo_core             = NO (dvo_core not found)
--   With okvis                = NO (okvis not found)
--   With msckf_vio            = NO (WITH_MSCKF_VIO=OFF)
--   With VINS-Fusion          = NO (VINS-Fusion not found)
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sukhraj/rtabmap/build
`

matlabbe, 2020-09-11:
The problem is that rtabmap is linking against your local copy of OpenCV4 while rtabmap_ros is linking against both OpenCV versions (source v4 and binaries v3.4 on melodic). If you want to use your OpenCV version built from source, uninstall ros-melodic-cv-bridge and rebuild vision_opencv package from source to link on your opencv version.

sKlairRad, 2020-09-11:
I uninstalled ros-melodic-cv-bridge, rebuilt rtabmap and included and built vision_opencv package in my workspace. I'm still getting the same error. Do I have to create a new map database?

matlabbe, 2020-09-13:
Can you do:
`$ ldd ~/catkin_ws/devel/lib/rtabmap_ros/rtabmap`
and
`$ ldd /usr/local/bin/rtabmap`?

sKlairRad, 2020-09-13:
`ldd ~/catkin_ws/devel/lib/rtabmap_ros/rtabmap` generated following:

        linux-vdso.so.1 (0x00007ffc3397e000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fd9d5c27000)
	libnodeletlib.so => /opt/ros/melodic/lib/libnodeletlib.so (0x00007fd9d59bf000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007fd9d562b000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007fd9d53f5000)
	librtabmap_core.so.0.20 => /usr/local/lib/librtabmap_core.so.0.20 (0x00007fd9d4aad000)
	librtabmap_utilite.so.0.20 => /usr/local/lib/librtabmap_utilite.so.0.20 (0x00007fd9d488c000)
	libopencv_calib3d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_calib3d.so.3.2 (0x00007fd9d453c000)
	libopencv_features2d.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_features2d.so.3.2 (0x00007fd9d4287000)
	libopencv_imgcodecs.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_imgcodecs.so.3.2 (0x00007fd9d404e000)
	libopencv_photo.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_photo.so.3.2 (0x00007fd9d3d8d000)
	libopencv_stitching.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_stitching.so.3.2 (0x00007fd9d3b0b000)
	libopencv_video.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_video.so.3.2 (0x00007fd9d38a8000)
	libopencv_videoio.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_videoio.so.3.2 (0x00007fd9d3671000)
	libopencv_aruco.so.3.2 => /usr/lib/x86_64-linux-gnu/libopencv_aruco.so.3.2 (0x00007fd9d3429000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fd9d30a0000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fd9d2e88000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fd9d2a97000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fd9d2878000)
	libbondcpp.so => /opt/ros/melodic/lib/libbondcpp.so (0x00007fd9d2647000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007fd9d2420000)
	libroslib.so => /opt/ros/melodic/lib/libroslib.so (0x00007fd9d220d000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007fd9d1ff9000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007fd9d1ddf000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007fd9d1bdc000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007fd9d19b7000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007fd9d17b2000)
	libxmlrpcpp.so => /opt/ros/melodic/lib/libxmlrpcpp.so (0x00007fd9d1593000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007fd9d1373000)
	libcpp_common.so => /opt/ros/melodic/lib/libcpp_common.so (0x00007fd9d1168000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007fd9d0f63000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fd9d0bc5000)
	librosconsole_log4cxx.so => /opt/ros/melodic/lib/librosconsole_log4cxx.so (0x00007fd9d09aa000)
	librosconsole_backend_interface.so => /opt/ros/melodic/lib/librosconsole_backend_interface.so (0x00007fd9d07a8000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007fd9d03df000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007fd9d00d7000)
	libopencv_stitching.so.4.4 => /usr/local/lib/libopencv_stitching.so.4.4 (0x00007fd9cfe22000)
	libopencv_aruco.so.4.4 => /usr/local/lib/libopencv_aruco.so.4.4 (0x00007fd9cfbb5000)
	libopencv_xfeatures2d.so.4.4 => /usr/local/lib/libopencv_xfeatures2d.so.4.4 (0x00007fd9cf6f4000)
	libpcl_common.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.8 (0x00007fd9cf438000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007fd9cf1d9000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fd9cefbc000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007fd9cecb3000)
	libpcl_io.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.8 (0x00007fd9ce7d8000)
	libpcl_kdtree.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.8 (0x00007fd9ce49d000)
	libpcl_search.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.8 (0x00007fd9ce096000)
	libpcl_surface.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8 (0x00007fd9cd921000)
	libpcl_filters.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.8 (0x00007fd9ccf07000)
	libpcl_features.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.8 (0x00007fd9caa5c000)
	libpcl_sample_consensus.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.8 (0x00007fd9c9bb5000)
	libpcl_segmentation.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.8 (0x00007fd9c8c92000)
	libfreenect.so.0.5 => /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007fd9c8a82000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007fd9c880c000)
	libg2o_core.so => /opt/ros/melodic/lib/libg2o_core.so (0x00007fd9c85b4000)
	libg2o_types_slam2d.so => /opt/ros/melodic/lib/libg2o_types_slam2d.so (0x00007fd9c835d000)
	libg2o_types_slam3d.so => /opt/ros/melodic/lib/libg2o_types_slam3d.so (0x00007fd9c80fb000)
	libg2o_types_sba.so => /opt/ros/melodic/lib/libg2o_types_sba.so (0x00007fd9c7ea7000)
	libg2o_stuff.so => /opt/ros/melodic/lib/libg2o_stuff.so (0x00007fd9c7c8a000)
	libg2o_csparse_extension.so => /opt/ros/melodic/lib/libg2o_csparse_extension.so (0x00007fd9c7a84000)
	libcxsparse.so.3 => /usr/lib/x86_64-linux-gnu/libcxsparse.so.3 (0x00007fd9c7859000)
[... cut ...]
	libvtkCommonColor-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonColor-6.3.so.6.3 (0x00007fd9885c1000)
	libvtkImagingFourier-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkImagingFourier-6.3.so.6.3 (0x00007fd988392000)
	libvtkalglib-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkalglib-6.3.so.6.3 (0x00007fd98816f000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007fd987f46000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007fd987d37000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007fd987aed000)

sKlairRad, 2020-09-13:
`ldd /usr/local/bin/rtabmap` generated following:
        linux-vdso.so.1 (0x00007ffd62ab7000)
	librtabmap_gui.so.0.20 => /usr/local/lib/librtabmap_gui.so.0.20 (0x00007fc184310000)
	librtabmap_core.so.0.20 => /usr/local/lib/librtabmap_core.so.0.20 (0x00007fc1839c8000)
	librtabmap_utilite.so.0.20 => /usr/local/lib/librtabmap_utilite.so.0.20 (0x00007fc1837a7000)
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007fc182f60000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007fc182815000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fc18248c000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fc182274000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fc181e83000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fc181c7e000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fc181a5f000)
	libpcl_common.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.8 (0x00007fc1817a3000)
	libpcl_io.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.8 (0x00007fc1812c8000)
	libpcl_search.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.8 (0x00007fc180ec1000)
	libpcl_surface.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8 (0x00007fc18074c000)
	libpcl_visualization.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_visualization.so.1.8 (0x00007fc1803ce000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fc180030000)
	libvtkIOMPIImage-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkIOMPIImage-6.3.so.6.3 (0x00007fc17fe08000)
	libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeTypeFontConfig-6.3.so.6.3 (0x00007fc17fbfe000)
	libvtkRenderingLOD-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingLOD-6.3.so.6.3 (0x00007fc17f9e5000)
	libvtkRenderingMatplotlib-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingMatplotlib-6.3.so.6.3 (0x00007fc17f7d4000)
	libvtkRenderingVolumeOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingVolumeOpenGL-6.3.so.6.3 (0x00007fc17f52d000)
	libopencv_imgcodecs.so.4.4 => /usr/local/lib/libopencv_imgcodecs.so.4.4 (0x00007fc17f2c6000)
	libopencv_calib3d.so.4.4 => /usr/local/lib/libopencv_calib3d.so.4.4 (0x00007fc17ee37000)
	libopencv_imgproc.so.4.4 => /usr/local/lib/libopencv_imgproc.so.4.4 (0x00007fc17d074000)
	libopencv_core.so.4.4 => /usr/local/lib/libopencv_core.so.4.4 (0x00007fc17be6b000)
	libQt5PrintSupport.so.5 => /usr/lib/x86_64-linux-gnu/libQt5PrintSupport.so.5 (0x00007fc17bbfc000)
	libvtkGUISupportQt-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkGUISupportQt-6.3.so.6.3 (0x00007fc17b9b2000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007fc17b249000)
	libvtkRenderingOpenGL-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingOpenGL-6.3.so.6.3 (0x00007fc17ada9000)
	libvtkInteractionStyle-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkInteractionStyle-6.3.so.6.3 (0x00007fc17ab23000)
	libvtkRenderingAnnotation-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingAnnotation-6.3.so.6.3 (0x00007fc17a7d9000)
	libvtkRenderingFreeType-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingFreeType-6.3.so.6.3 (0x00007fc17a516000)
	libvtkRenderingCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkRenderingCore-6.3.so.6.3 (0x00007fc17a096000)
	libvtkFiltersSources-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersSources-6.3.so.6.3 (0x00007fc179dd5000)
	libvtkFiltersGeneral-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkFiltersGeneral-6.3.so.6.3 (0x00007fc1798a6000)
	libvtkCommonExecutionModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonExecutionModel-6.3.so.6.3 (0x00007fc1795ff000)
	libvtkCommonDataModel-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonDataModel-6.3.so.6.3 (0x00007fc1790a6000)
	libvtkCommonMath-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonMath-6.3.so.6.3 (0x00007fc178e86000)
	libvtkCommonCore-6.3.so.6.3 => /usr/lib/x86_64-linux-gnu/libvtkCommonCore-6.3.so.6.3 (0x00007fc178966000)
	libopencv_stitching.so.4.4 => /usr/local/lib/libopencv_stitching.so.4.4 (0x00007fc1786b1000)
	libopencv_aruco.so.4.4 => /usr/local/lib/libopencv_aruco.so.4.4 (0x00007fc178444000)
	libopencv_xfeatures2d.so.4.4 => /usr/local/lib/libopencv_xfeatures2d.so.4.4 (0x00007fc177f83000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007fc177d24000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fc177b07000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007fc1777fe000)
	libpcl_kdtree.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.8 (0x00007fc1774c3000)
	libpcl_filters.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.8 (0x00007fc176aa9000)
	libpcl_features.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.8 (0x00007fc1745fe000)
	libpcl_sample_consensus.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.8 (0x00007fc173757000)
	libpcl_segmentation.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.8 (0x00007fc172834000)
	libfreenect.so.0.5 => /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007fc172624000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007fc1723ae000)
	libg2o_core.so => /opt/ros/melodic/lib/libg2o_core.so (0x00007fc172156000)
	libg2o_types_slam2d.so => /opt/ros/melodic/lib/libg2o_types_slam2d.so (0x00007fc171eff000)
	libg2o_types_slam3d.so => /opt/ros/melodic/lib/libg2o_types_slam3d.so (0x00007fc171c9d000)
[... cut ...]
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007fc13f663000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007fc13f42d000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007fc13f217000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007fc13efee000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007fc13eddf000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007fc13eb95000)
	libcrypt.so.1 => /lib/x86_64-linux-gnu/libcrypt.so.1 (0x00007fc13e95d000)

matlabbe, 2020-09-13:
If you search for "opencv" in those logs, you can see that rtabmap library (second output) is linking only on OpenCV4, which is okay. However for rtabmap_ros, it is linking to OpenCV 3.2 and OpenCV 4.4. Make sure vision_opencv is built against the right OpenCV. You can also do `ldd ~/catkin_ws/devel/lib/libcv_bridge.so` to see if it is linking to OpenCV 4.4 (normally cv_bridge should find OpenCV 4 before OpenCV 3, see [here](https://github.com/ros-perception/vision_opencv/blob/noetic/cv_bridge/CMakeLists.txt#L21)).

sKlairRad, 2020-09-13:
I am using ROS Melodic and the source files for cv_bridge (melodic) are using OpenCV 3 [here](https://github.com/ros-perception/vision_opencv/blob/578af4d6c7846876b3fc64512b1cc92a54894483/cv_bridge/CMakeLists.txt#L16). When I change it to OpenCV 4, I get build errors.

sKlairRad, 2020-09-14:
I uninstalled OpenCV4 and built OpenCV 3.4 from source with xfeatures2d module.
Now, when I launch rtab_ros with SURF, i still get a warning saying OpenCV wasn't built with XFeatures2d module. 

[image]

I get following when I run `ldd ~/catkin_ws/devel/lib/rtabmap_ros/rtabmap`:
`	linux-vdso.so.1 (0x00007ffda5237000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007fa86c032000)
	libnodeletlib.so => /opt/ros/melodic/lib/libnodeletlib.so (0x00007fa86bdca000)
	libroscpp.so => /opt/ros/melodic/lib/libroscpp.so (0x00007fa86ba36000)
	librosconsole.so => /opt/ros/melodic/lib/librosconsole.so (0x00007fa86b800000)
	librtabmap_core.so.0.20 => /usr/local/lib/librtabmap_core.so.0.20 (0x00007fa86aec2000)
	librtabmap_utilite.so.0.20 => /usr/local/lib/librtabmap_utilite.so.0.20 (0x00007fa86aca1000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa86a918000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa86a700000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa86a30f000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa86a0f0000)
	libbondcpp.so => /opt/ros/melodic/lib/libbondcpp.so (0x00007fa869ebf000)
	libclass_loader.so => /opt/ros/melodic/lib/libclass_loader.so (0x00007fa869c98000)
	libroslib.so => /opt/ros/melodic/lib/libroslib.so (0x00007fa869a85000)
	libtinyxml2.so.6 => /usr/lib/x86_64-linux-gnu/libtinyxml2.so.6 (0x00007fa869871000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007fa869657000)
	libroscpp_serialization.so => /opt/ros/melodic/lib/libroscpp_serialization.so (0x00007fa869454000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007fa86922f000)
	libconsole_bridge.so.0.4 => /usr/lib/x86_64-linux-gnu/libconsole_bridge.so.0.4 (0x00007fa86902a000)
	libxmlrpcpp.so => /opt/ros/melodic/lib/libxmlrpcpp.so (0x00007fa868e0b000)
	librostime.so => /opt/ros/melodic/lib/librostime.so (0x00007fa868beb000)
	libcpp_common.so => /opt/ros/melodic/lib/libcpp_common.so (0x00007fa8689e0000)
	libboost_chrono.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_chrono.so.1.65.1 (0x00007fa8687db000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa86843d000)
	librosconsole_log4cxx.so => /opt/ros/melodic/lib/librosconsole_log4cxx.so (0x00007fa868222000)
	librosconsole_backend_interface.so => /opt/ros/melodic/lib/librosconsole_backend_interface.so (0x00007fa868020000)
	liblog4cxx.so.10 => /usr/lib/x86_64-linux-gnu/liblog4cxx.so.10 (0x00007fa867c57000)
	libboost_regex.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_regex.so.1.65.1 (0x00007fa86794f000)
	libopencv_stitching.so.3.4 => /usr/local/lib/libopencv_stitching.so.3.4 (0x00007fa8676bb000)
	libopencv_aruco.so.3.4 => /usr/local/lib/libopencv_aruco.so.3.4 (0x00007fa867452000)
	libopencv_xfeatures2d.so.3.4 => /usr/local/lib/libopencv_xfeatures2d.so.3.4 (0x00007fa866fb1000)
	libpcl_common.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_common.so.1.8 (0x00007fa866cf5000)
	libOpenNI2.so.0 => /usr/lib/libOpenNI2.so.0 (0x00007fa866a96000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fa866879000)
	libsqlite3.so.0 => /usr/lib/x86_64-linux-gnu/libsqlite3.so.0 (0x00007fa866570000)
	libpcl_io.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_io.so.1.8 (0x00007fa866095000)
	libpcl_kdtree.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_kdtree.so.1.8 (0x00007fa865d5a000)
	libpcl_search.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_search.so.1.8 (0x00007fa865953000)
	libpcl_surface.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_surface.so.1.8 (0x00007fa8651de000)
	libpcl_filters.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_filters.so.1.8 (0x00007fa8647c4000)
	libpcl_features.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_features.so.1.8 (0x00007fa862319000)
	libpcl_sample_consensus.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_sample_consensus.so.1.8 (0x00007fa861472000)
	libpcl_segmentation.so.1.8 => /usr/lib/x86_64-linux-gnu/libpcl_segmentation.so.1.8 (0x00007fa86054f000)
	libfreenect.so.0.5 => /usr/lib/x86_64-linux-gnu/libfreenect.so.0.5 (0x00007fa86033f000)
	libdc1394.so.22 => /usr/lib/x86_64-linux-gnu/libdc1394.so.22 (0x00007fa8600c9000)
	libg2o_core.so => /opt/ros/melodic/lib/libg2o_core.so (0x00007fa85fe71000)
	libg2o_types_slam2d.so => /opt/ros/melodic/lib/libg2o_types_slam2d.so (0x00007fa85fc1a000)
	libg2o_types_slam3d.so => /opt/ros/melodic/lib/libg2o_types_slam3d.so (0x00007fa85f9b8000)
	libg2o_types_sba.so => /opt/ros/melodic/lib/libg2o_types_sba.so (0x00007fa85f764000)
	libg2o_stuff.so => /opt/ros/melodic/lib/libg2o_stuff.so (0x00007fa85f547000)
	libg2o_csparse_extension.so => /opt/ros/melodic/lib/libg2o_csparse_extension.so (0x00007fa85f341000)
	libcxsparse.so.3 => /usr/lib/x86_64-linux-gnu/libcxsparse.so.3 (0x00007fa85f116000)
	libcholmod.so.3 => /usr/lib/x86_64-linux-gnu/libcholmod.so.3 (0x00007fa85ee40000)
	libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007fa857030000)
	libopenblas.so.0 => /usr/lib/x86_64-linux-gnu/libopenblas.so.0 (0x00007fa854d8a000)
	liboctomap.so.1.9 => /opt/ros/melodic/lib/liboctomap.so.1.9 (0x00007fa854b44000)
	liboctomath.so.1.9 => /opt/ros/melodic/lib/liboctomath.so.1.9 (0x00007fa85493d000)
	libopencv_videoio.so.3.4 => /usr/local/lib/libopencv_videoio.so.3.4 (0x00007fa8546e0000)
	libopencv_video.so.3.4 => /usr/local/lib/libopencv_video.so.3.4 (0x00007fa854475000)
	libopencv_imgcodecs.so.3.4 => /usr/local/lib/libopencv_imgcodecs.so.3.4 (0x00007fa8541c1000)
[... cut ...]
	libkrb5.so.26 => /usr/lib/x86_64-linux-gnu/libkrb5.so.26 (0x00007fa82af6a000)
	libasn1.so.8 => /usr/lib/x86_64-linux-gnu/libasn1.so.8 (0x00007fa82acc8000)
	libhcrypto.so.4 => /usr/lib/x86_64-linux-gnu/libhcrypto.so.4 (0x00007fa82aa92000)
	libroken.so.18 => /usr/lib/x86_64-linux-gnu/libroken.so.18 (0x00007fa82a87c000)
	libwind.so.0 => /usr/lib/x86_64-linux-gnu/libwind.so.0 (0x00007fa82a653000)
	libheimbase.so.1 => /usr/lib/x86_64-linux-gnu/libheimbase.so.1 (0x00007fa82a444000)
	libhx509.so.5 => /usr/lib/x86_64-linux-gnu/libhx509.so.5 (0x00007fa82a1fa000)
`

matlabbe, 2020-09-15:
To have SURF, we should also enable OPENCV_ENABLE_NONFREE option when building opencv. For cv_bridge and opencv 4, you could have tried the noetic version instead (even in melodic).

If OpencV is built with nonfree, it should be shown in the About dialog of RTAB-Map and the license of OpenCV would be set to Non commercial:
[image]