introlab/rtabmap_ros #463: cannot find -lBoost::timer

NamDinhRobotics, 2020-09-16:
Hello,
I tried to build rtabmap from source in Ubuntu 18.04, but it has the problem with cannot find -lBoost::timer,
/usr/bin/ld: C
collect2: error: ld returned 1 exit status
can you guess what problem?

matlabbe, 2020-09-16:
Can you show output of cmake of rtabmap build?

NamDinhRobotics, 2020-09-17:
Hello, I solved the problem, 
We simply need to add find_package(boost ....) into cmake, then it worked fine.
Thanks,

matlabbe, 2020-09-17:
In which CMakeLists.txt did you added the line? (rtabmap or rtabmap_ros?)

NamDinhRobotics, 2020-09-17:
I added in these files
https://github.com/introlab/rtabmap/blob/master/corelib/src/CMakeLists.txt
https://github.com/introlab/rtabmap_ros/blob/master/CMakeLists.txt
FIND_PACKAGE( Boost 1.61 COMPONENTS filesystem program_options iostreams timer REQUIRED )

matlabbe, 2020-09-17:
Note that I explicitly do it already if libpointmatcher is detected: https://github.com/introlab/rtabmap/blob/3d33370e4c5530c964ba810ca4639dbaa520f342/CMakeLists.txt#L432

Maybe there is another dependency that requires boost and doesn't search for it. For rtabmap, can you show output of cmake so I can see which dependency may be causing the missing boost? For rtabmap_ros, I commented a while ago this line:
https://github.com/introlab/rtabmap_ros/blob/58dd3134be6f2ccc81f998df4e258354f0071e44/CMakeLists.txt#L33
Normally, dependencies using boost would have already looked for those boost targets. You may not need to modify rtabmap_ros if you already did in rtabmap.