introlab/rtabmap_ros #465: Scene doubled when playing RGB-TUM rosbag

thohemp, 2020-09-17:
Hello,

I want to evaluate the rmse of different settings using the rgb-d tum dataset.
When I play rosbag:

`roslaunch rtabmap_ros rgbdslam_datasets.launch rtabmapviz:=true rviz:=false `
`rosbag play --clock rgbd_dataset_freiburg2_desk_with_person.bag`

It displays the scene two times:
[image]

The odometry is jumping between the two trajectories, also seen in the resulting pose.txt:

> 
> 1311870427,644555 0,000602 0,017790 -0,087685 0,050533 0,867158 -0,494885 -0,023942
> 1311870429,031368 -0,151585 -0,351755 1,530364 -0,494079 0,659023 -0,450356 0,344606
> 1311870430,231223 -0,007119 0,021980 -0,083133 0,039424 0,866977 -0,496059 -0,026867
> 1311870431,231528 -0,078153 0,034339 -0,077807 0,061110 0,864864 -0,496194 -0,045466
> 1311870432,231754 -0,192167 0,071352 -0,051138 0,070960 0,855047 -0,510257 -0,059142
> 1311870433,431583 -0,341955 0,087052 -0,064164 0,058933 0,862040 -0,501307 -0,045882
> 1311870434,431695 -0,297104 -0,771203 1,529780 -0,463999 0,688166 -0,463722 0,309990
> 1311870435,499650 -0,574490 0,049792 -0,162125 0,037686 0,919218 -0,387120 -0,061287
> 1311870436,831852 -0,259911 -1,006039 1,491239 -0,533761 0,697186 -0,392709 0,273515
> 1311870437,831896 -0,916322 -0,048408 -0,100996 -0,009943 0,896357 -0,442841 0,018383
> 1311870438,832114 -0,299063 -1,380470 1,565531 -0,580434 0,620857 -0,381949 0,362971
> 1311870439,867821 -1,247698 -0,200203 -0,135647 -0,096139 0,885756 -0,447814 0,075213
> 1311870441,232068 -1,477027 -0,282972 -0,204907 -0,221278 0,888787 -0,383258 0,119198
> 1311870442,300086 -1,602140 -0,341088 -0,218909 -0,211090 0,899655 -0,363055 0,119389
> 1311870443,433543 0,004680 -1,855151 1,544865 0,703840 -0,517049 0,281851 -0,397277
> 1311870444,436312 -1,772271 -0,551095 -0,212067 -0,308140 0,844264 -0,383990 0,211705
> 1311870445,436350 -1,965098 -0,678574 -0,323275 -0,353382 0,850016 -0,307579 0,240809
> 1311870446,436540 0,231225 -2,293618 1,425785 0,830718 -0,376795 0,146453 -0,382733
> 1311870447,636328 -2,136514 -1,017041 -0,386747 -0,518404 0,751463 -0,271250 0,304932
> 1311870448,736438 -1,924944 -1,090658 -0,356730 -0,523170 0,771092 -0,247800 0,265151
> 1311870449,736551 0,639439 -2,188838 1,458826 0,911129 -0,112805 0,040775 -0,394279
> 1311870450,840324 0,687150 -2,330219 1,399000 0,920906 -0,049951 -0,026592 -0,385656
> 1311870451,938315 0,785945 -2,583629 1,388727 0,919922 0,011722 -0,048898 -0,388863
> 1311870452,968792 0,946598 -2,716748 1,379881 0,912460 0,107496 -0,093664 -0,383521
> 1311870454,137000 1,105674 -2,670134 1,360409 0,895950 0,171509 -0,148145 -0,381984
> 1311870455,169130 1,228143 -2,691117 1,349357 0,876700 0,250123 -0,173098 -0,372656
> 1311870457,369144 1,313389 -2,699138 1,330467 0,867992 0,274771 -0,194799 -0,364890
> 1311870458,569254 -2,088936 -2,075005 -0,735018 -0,820696 0,258872 -0,054019 0,506483
> 1311870459,569420 1,316417 -2,649319 1,352374 0,876253 0,223815 -0,179464 -0,387144
> 1311870460,637439 1,548272 -2,734644 1,345232 0,885183 0,226892 -0,145437 -0,379235
> 1311870462,005294 1,576611 -2,947017 1,412022 0,899049 0,098905 -0,083909 -0,418196

matlabbe, 2020-09-17:
The camera is jumping between the ground truth and odometry trajectories. You need two extra steps to make the rosbag compatible with that launch file:
https://github.com/introlab/rtabmap_ros/blob/58dd3134be6f2ccc81f998df4e258354f0071e44/launch/tests/rgbdslam_datasets.launch#L7-L8
```
wget https://gist.githubusercontent.com/matlabbe/897b775c38836ed8069a1397485ab024/raw/6287ce3def8231945326efead0c8a7730bf6a3d5/tum_rename_world_kinect_frame.py
python tum_rename_world_kinect_frame.py rgbd_dataset_freiburg2_desk_with_person.bag
```

Then with
```
roslaunch rtabmap_ros rgbdslam_datasets.launch rtabmapviz:=true rviz:=false
rosbag play --clock rgbd_dataset_freiburg2_desk_with_person.bag
```
you should get something similar to this:
[image]

Also beware that because of your system's language, your exported poses contain commas `,` instead of dots `.` for floating point numbers. This could affect a script using this file (a fast fix is to do a search and replace in a text editor).

cheers,
Mathieu

matlabbe, 2020-09-17:
Update: Note also that it seems to have a slithly scale issue between the depth/calibration and the ground truth. Doing this recompute the RMSE without that scale difference to get a 5.6 cm error:
```
rtabmap-report --scale ~/.ros/rtabmap.db
Database: /home/mathieu/.ros/rtabmap.db
   rtabmap.db (1333, s=0.941):	error lin=0.056m (max=0.110m, odom=0.056m) ang=3.2deg, slam: avg=7ms (max=336ms) loops=20, odom: avg=0ms (max=0ms), camera: avg=0ms, map=-1MB
```
Without scaling we get the 11 cm error that we saw in the screenshot:
```
rtabmap-report ~/.ros/rtabmap.db
Database: /home/mathieu/.ros/rtabmap.db
   rtabmap.db (1333, s=1.000):	error lin=0.110m (max=0.110m, odom=0.120m) ang=3.2deg, slam: avg=7ms (max=336ms) loops=20, odom: avg=0ms (max=0ms), camera: avg=0ms, map=-1MB
```

thohemp, 2020-09-18:
Thank you very much! Everything works great now!

Can you elaborate what the scale issue is about? What causes it and does it only refer to F2M and F2F, or also when the ORBSLAM2 is used for odomtry?

Can you tell me what the difference is between the odometry poses and slam poses?

matlabbe, 2020-09-20:
The scale can come from the calibration (we assume that the focal distance in camera_info included in the bag is correct, but it may be slightly off) or it could come from the depth image itself, having depth values with some scale error. If the scale is off and you record a ground truth that is perfectly calibrated, the trajectory of the camera will seem smaller or larger than the ground truth. The scale issue would affect all odometry approaches (the same way... so F2M and ORB_SLAM2 odometries should give a trajectory similar with same scale when comparing each other). If I remember well, the results in ORB_SLAM2 paper are also scaled.

With slam poses, it contains loop closure detections, as for odometry it is the frame-to-keyframe trajectory estimation. Normally with slam, the RMSE would be better as it will correct odometry drift.

thohemp, 2020-09-21:
Understood, thank you.

One last question: I would like to use landmarks for graph optimization to analyze its influence on the RMSE. 

I added tag_detection to rgbd_dataset.launch to trigger the apriltag subscriber and it receives my published landmarks successfully:
[image]

But it appears that landmarks disappear after some time and I don't think these landmarks are used to optimize the graph.

Usually I used a modified version of rtabmap.launch for testing landmarks on my own bag files. This works great, but I don't have corresponding groundtruth data.
```
<launch>
  <!-- Convenience launch file to launch odometry, rtabmap and rtabmapviz nodes at once -->

  <!-- For stereo:=false
        Your RGB-D sensor should be already started with "depth_registration:=true".
        Examples:
           $ roslaunch freenect_launch freenect.launch depth_registration:=true 
           $ roslaunch openni2_launch openni2.launch depth_registration:=true -->
           
  <!-- For stereo:=true
        Your camera should be calibrated and publishing rectified left and right 
        images + corresponding camera_info msgs. You can use stereo_image_proc for image rectification.
        Example:
           $ roslaunch rtabmap_ros bumblebee.launch -->
     
  <!-- Choose between depth and stereo, set both to false to do only scan -->      
  <arg name="stereo"                    default="false"/>
  <arg     if="$(arg stereo)" name="depth"  default="false"/>
  <arg unless="$(arg stereo)" name="depth"  default="true"/>
   <arg name="save_objects" default="false"/>
  <!-- Choose visualization -->
  <arg name="rtabmapviz"              default="true" /> 
  <arg name="rviz"                    default="true" />
  
  <!-- Localization-only mode -->
  <arg name="localization"            default="false"/>
  
  <!-- sim time for convenience, if playing a rosbag -->
  <arg name="use_sim_time"            default="false"/>
  <param if="$(arg use_sim_time)" name="use_sim_time" value="true"/>
  
  <!-- Corresponding config files -->
  <arg name="cfg"                     default="" /> <!-- To change RTAB-Map's parameters, set the path of config file (*.ini) generated by the standalone app -->
  <arg name="gui_cfg"                 default="~/.ros/rtabmap_gui.ini" />
  <arg name="rviz_cfg"                default="$(find rtabmap_ros)/launch/config/demo_find_object.rviz"/> <!--default="$(find rtabmap_ros)/launch/config/rgbd.rviz" />-->
  
  <arg name="frame_id"                default="camera_link"/>     <!-- Fixed frame id, you may set "base_link" or "base_footprint" if they are published -->
  <arg name="odom_frame_id"           default=""/>                <!-- If set, TF is used to get odometry instead of the topic -->
  <arg name="map_frame_id"            default="map"/>
  <arg name="ground_truth_frame_id"   default=""/>     <!-- e.g., "world" -->
  <arg name="ground_truth_base_frame_id" default=""/>  <!-- e.g., "tracker", a fake frame matching the frame "frame_id" (but on different TF tree) -->
  <arg name="namespace"               default="rtabmap"/>
  <arg name="database_path"           default="~/.ros/rtabmap.db"/>
  <arg name="queue_size"              default="10"/>
  <arg name="wait_for_transform"      default="0.2"/>
  <arg name="args"                    default=""/>              <!-- delete_db_on_start, udebug -->
  <arg name="rtabmap_args"            default="$(arg args)"/>   <!-- deprecated, use "args" argument -->
  <arg name="launch_prefix"           default=""/>              <!-- for debugging purpose, it fills launch-prefix tag of the nodes -->
  <arg name="output"                  default="screen"/>        <!-- Control node output (screen or log) -->
  <arg name="publish_tf_map"          default="true"/>

  <!-- if timestamps of the input topics are synchronized using approximate or exact time policy-->
  <arg     if="$(arg stereo)" name="approx_sync"  default="false"/>
  <arg unless="$(arg stereo)" name="approx_sync"  default="$(arg depth)"/>         
   
  <!-- RGB-D related topics -->
  <arg name="rgb_topic"               default="/camera/rgb/image_rect_color" />
  <arg name="depth_topic"             default="/camera/depth_registered/image_raw" />
  <arg name="camera_info_topic"       default="/camera/rgb/camera_info" />
[... 331 lines cut ...]
      <remap from="odom"                   to="$(arg odom_topic)"/>
    </node>
  
  </group>
  
  <!-- Visualization RVIZ -->
  <node if="$(arg rviz)" pkg="rviz" type="rviz" name="rviz" args="-d $(arg rviz_cfg)"/>
  <node if="$(arg rviz)" pkg="nodelet" type="nodelet" name="points_xyzrgb" args="standalone rtabmap_ros/point_cloud_xyzrgb" output="$(arg output)">
    <remap if="$(arg stereo)" from="left/image"        to="$(arg left_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="right/image"       to="$(arg right_image_topic_relay)"/>
    <remap if="$(arg stereo)" from="left/camera_info"  to="$(arg left_camera_info_topic)"/>
    <remap if="$(arg stereo)" from="right/camera_info" to="$(arg right_camera_info_topic)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/image"         to="$(arg rgb_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="depth/image"       to="$(arg depth_topic_relay)"/>
    <remap unless="$(arg subscribe_rgbd)" from="rgb/camera_info"   to="$(arg camera_info_topic)"/>
    <remap from="rgbd_image"        to="$(arg rgbd_topic_relay)"/>
    <remap from="cloud"             to="voxel_cloud" />

    <param name="decimation"  type="double" value="4"/>
    <param name="voxel_size"  type="double" value="0.0"/>
    <param name="approx_sync" type="bool"   value="$(arg approx_sync)"/>
  </node>
 <include file="$(find darknet_ros)/launch/yolo_v3.launch"> </include>
</launch>
```

matlabbe, 2020-09-27:
Is /rtabmap/landmarks published? Can you share a database with the landmarks?

thohemp, 2020-09-28:
I use /tag_detections for publishing landmarks. This is rqt_graph:

[image]

i just tested it again and it know also shows the edges for the landmarks:

[image]

But given the log of my object_mapper node, there are a lot more observations published.  Is there any setting for the /tag_detections subscription to increase the callback rate?

thohemp, 2020-09-28:
There might be an issue with the stamps causing my published landmarks to be rejected. I receive a lot of warnings:

> [ WARN] [1601295376.834261642, 1311870567.448799192]: Updating existing nodes: Lookup would require extrapolation into the past.  Requested time 1311870490.443150935 but the earliest data is at time 1311870557.294714652, when looking up transform from frame [mouse_9700] to frame [kinect]
> [ WARN] [1601295376.834278332, 1311870567.448799192]: Updating existing nodes: Lookup would require extrapolation into the past.  Requested time 1311870511.680527965 but the earliest data is at time 1311870557.294714652, when looking up transform from frame [cup_13559] to frame [kinect]
> [ WARN] [1601295376.834295312, 1311870567.448799192]: Updating existing nodes: Lookup would require extrapolation into the past.  Requested time 1311870548.090696881 but the earliest data is at time 1311870557.294714652, when looking up transform from frame [person_19706] to frame [kinect]
> [ WARN] [1601295376.834311622, 1311870567.448799192]: Updating existing nodes: Lookup would require extrapolation into the past.  Requested time 1311870554.823586166 but the earliest data is at time 1311870557.294714652, when looking up transform from frame [person_21093] to frame [kinect]

I don't know why this happens as I use the latest stamp in the buffer:
`tfListener_.lookupTransform(frameId_, objectNodes[i].child_frameID, stamp_ = ros::Time(0), pose);`

matlabbe, 2020-09-28:
`Rtabmap/DetectionRate` can be increased to add more nodes, thus more landmark observations.

Request time is 1311870490.443150935, and earliest time in tf buffer is 1311870557.294714652, the delta is about 67 seconds. 

Is it because the TF between [cup_13559] and [kinect] did not yet published since 67 seconds? Well, if the camera didn't see the cup_13559 since 67 seconds, it is okay, so your code should not try to get it.

thohemp, 2020-09-29:
It should be there. And as rtabmap maps the landmark correctly, why isn't it drawing the edges from the nodes?

The structure is the following.

darknet_ros node is broadcasting object coordinates with objectids and the frame_id=/openni_rgb_optical_frame.

-> the object mapper node takes the objectsids and looks the tf up with the frame_id=kinect. Is this correct?
[image]

Here is shot of the broadcasted tfs, it want to lookup the tfs in camera reference to derive the world position in the next step. So frame_id=kinect should be correct for this dataset?

thohemp, 2020-09-29:
For example, if I want to map the cup and mouse on the desk:
rtabmap receives the objects from /tag_detection and maps them:
[image]
They are also in the graph:
[image]

But there are no green egdes and they are never use for graph optimization, even tthough the object are getting republished with the same id for almost every node.

The system works on my own bags perfectly.

thohemp, 2020-10-20:
@matlabbe I investigated this issue and found out, that the landmarks are not published when playing the database. Reason is that in this line: https://github.com/introlab/rtabmap_ros/blob/master/src/CoreWrapper.cpp#L3535 the last condition `stats.poses().begin()->first < 0` is always false. 
It also seems odd to me, that the nodes of the position graph are mostly not connected. You can see this also in the image from my last comment.

matlabbe, 2020-10-20:
When you say "when playing the database", is it by using data_player in ros or by replaying it as source in standalone? Can you share that database?

thohemp, 2020-10-20:
I use ros and followed your instructions:
```

roslaunch rtabmap_ros rgbdslam_datasets.launch rtabmapviz:=true rviz:=true  // in rviz I subscribed the rtabmap/landmarks topic
rosbag play --clock rgbd_dataset_freiburg2_desk_with_person.bag
```

So, the database is the rgbd tum database with bags like rgbd_dataset_freiburg2_desk_with_person.bag.
I then use tag_detection topic to send landmarks (e.g the position of the can on the desk).

matlabbe, 2020-10-20:
> The system works on my own bags perfectly.
Note sure why it works on some bags and not on others. The loop closures may be rejected, so landmarks would be rejected. For the missing blue links, maybe the odometry gets lost and is automatically reset. Set [this](https://github.com/introlab/rtabmap_ros/blob/2ae24855af927a9cdcd8805d7edfff124dcd28b5/launch/tests/rgbdslam_datasets.launch#L34) to 0. With multiple sessions not linked together, the landmarks link may not appear. Having the database corresponding your graph above would help to detect other problems.

thohemp, 2020-10-20:
I will try that.

> 
> 
> > The system works on my own bags perfectly.
> 
> Note sure why it works on some bags and not on others. 

It's not only the different bag. I use also the  "rtabmap.launch" with my bags. But now I want to test my system on a database with ground truth.

> The loop closures may be rejected, so landmarks would be rejected. For the missing blue links, maybe the odometry gets lost >and is automatically reset. Set [...]

I will try that and I am gonna record a bag with my additional `tag_detection topic`.

thohemp, 2020-10-21:
@matlabbe 

I record a bag, so you can reproduce my experience: https://drive.google.com/file/d/1tAuzDNI29FMi8BO0aGf-OZrO6G8YRyIi/view?usp=sharing (be aware it's compressed)
This is essentially the rgbd_dataset_freiburg2_desk_with_person.bag, but I additionally recorded my /tag_detection topic.

I basically used your launch file "rgbdslam_datasets.launch". My only changes are to activate the /tag_detection subscription in rtabmab and the /rtabmap/landmark subscription in rviz to visualize the landmarks.

```
roslaunch rtabmap_ros rgbdslam_datasets.launch rtabmapviz:=true rviz:=true 
rosbag play --clock rgbd_dataset_freiburg2_desk_with_person_with_landmarks.bag
```

thohemp, 2020-12-14:
Hello @matlabbe,

I am not sure what a I messed up with the rgbd_dataset launch file. So I included the ground truth into my own launch file. Now the tag detections are working flawless.

Thanks

closed 2020-12-14 by thohemp