introlab/rtabmap_ros #468: Issue building from source: "Linking CXX executable ../../../bin/rtabmap"

ArvidPeterson, 2020-09-24:
Hi!

I am running into problems when building rtabmap from source following [this](https://github.com/introlab/rtabmap_ros#build-from-source) guide. On Ubuntu 18.04 having built ibpointmatcher from source. At step 2 when executing `make` command i get the following error

```
[ 80%] Built target rtabmap_gui
[ 80%] Generating moc_ObjDeletionHandler.cpp
Scanning dependencies of target rtabmap
[ 80%] Building CXX object app/src/CMakeFiles/rtabmap.dir/main.cpp.o
[ 81%] Building CXX object app/src/CMakeFiles/rtabmap.dir/moc_ObjDeletionHandler.cpp.o
[ 81%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::getFeatureViewByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::setDefault()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ErrorMinimizer::getWeightedPointUsedRatio() const'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::getDescriptorRowViewByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::featureExists(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::~ICPChainBase()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::loadAdditionalYAMLContent(YAML_PM::Node&)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::Label::Label(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::DataPoints()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::~ICPChainBase()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcherSupport::InvalidElement::InvalidElement(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::~ICPChainBase()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::Labels::Labels()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::get()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::conservativeResize(long)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::getDescriptorRowViewByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned int) const'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::DataPoints(PointMatcher<float>::DataPoints::Labels const&, PointMatcher<float>::DataPoints::Labels const&, unsigned long)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::ICPChainBase()'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::DataPoints::getFeatureViewByName(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICP::operator()(PointMatcher<float>::DataPoints const&, PointMatcher<float>::DataPoints const&)'
../../../bin/librtabmap_core.so.0.20.3: undefined reference to `PointMatcher<float>::ICPChainBase::loadFromYaml(std::istream&)'
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:498: recipe for target '../bin/rtabmap' failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target 'app/src/CMakeFiles/rtabmap.dir/all' failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:151: recipe for target 'all' failed
make: *** [all] Error 2

```

I have tried the LD_LIBRARY_PATH fix suggested [here](https://github.com/introlab/rtabmap_ros/issues/217#issuecomment-364485707), it made some progress (built to 81% instead of 80%) but still does not work. 
Any suggestions are highly appreciated!

ArvidPeterson, 2020-09-25:
Solved by update in libpointmatcher repo.

closed 2020-09-25 by ArvidPeterson

matlabbe, 2020-09-27:
crossref: http://official-rtab-map-forum.67519.x6.nabble.com/Building-from-source-linking-issues-td6951.html#a6958