introlab/rtabmap_ros #469: [bug?] Service octomap binary/full

eliabntt, 2020-09-28:
Hi, I was trying to use the service `rtabmap/octomap_{binary,full}`. If I try to call it from shell it gives me `ERROR: transport error completing service call: receive_once[/rtabmap/octomap_binary]: DeserializationError cannot deserialize: unknown error handler name 'rosmsg'` and in c++ when I try to access the octomap (after doing MapFromMsg) I receive segfault errors.

Is it the same for you or it's my problem?

matlabbe, 2020-09-28:
Try casting to `rtabmap::RtabmapColorOcTree*` instead of `octomap::OcTree*`, with `#include <rtabmap/core/OctoMap.h>`.

cheers,
Mathieu

eliabntt, 2020-09-29:
it's the dynamic_cast that doesn't work. with static everything goes on as expected. I'll try your solution and come back here ;)

SuYouge, 2020-11-09:
Hi, I also meet this problem. I find `/rtabmap/octomap_full` (and binary) response a octomap msg with `id = "ColorOcTree"` not "OcTree". 
So basically I can not use `octomap::OcTree*` as my dynamic_cast target pointer type, and `octomap::ColorOcTree*` wont cause crash.
I am tring to porting a [grid_map demo](https://github.com/ANYbotics/grid_map/blob/04c40c5fb920ff46368b2b680df0e1309159bea8/grid_map_demos/src/OctomapToGridmapDemo.cpp#L81) to this rtabmap octomap service and failed to implement this pointer conversion.
So, is there a good way to convert the `ColorOcTree` type tree to `OcTree` type tree?

eliabntt, 2020-11-12:
IDK, I had it working with the static cast and never looked into that again. Mostly, it's just overloading of the class and c++ is complaining because we're not probably giving the exact same type or it has difficulties understanding the type because of that overload.

closed 2021-03-24 by eliabntt