introlab/rtabmap_ros #472: Can't launch the example file - ROS2 (eloquent)

mirellameelo, 2020-10-06:
Hi again. :)

After building Rtabmap ROS packages, I tried to launch the example provided [here](https://github.com/introlab/rtabmap_ros/blob/ros2/README.md). As follows, my step by step:

**Terminal 1:**
```
source /opt/ros/eloquent/setup.bash
export TURTLEBOT3_MODEL=waffle
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/opt/ros/<ros2-distro>/share/turtlebot3_gazebo/models
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py #(it works fine!)
```

**Terminal 2:**
```
source /opt/ros/eloquent/setup.bash
source $HOME/ros2_ws/install/setub.bash #related to rtab-map environment
ros2 launch rtabmap_ros turtlebot3_scan.launch.py
```

Then I have the following error:
```
[rtabmap-1] double free or corruption (out)
[rtabmapviz-2] double free or corruption (out)
[ERROR] [rtabmap-1]: process has died [pid 11249, exit code -6, cmd '/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap -d --ros-args --params-file /tmp/launch_params_1c_5ilq_ -r scan:=/scan'].
[ERROR] [rtabmapviz-2]: process has died [pid 11250, exit code -6, cmd '/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmapviz --ros-args --params-file /tmp/launch_params_3akqiq_a -r scan:=/scan'].
```

Please, did I miss anything? Really appreciate any help.

mirellameelo, 2020-10-06:
Update: once Rtab-map newest release ([v0.20.3](https://github.com/introlab/rtabmap/releases/tag/0.20.3)) considers PCL versions >= 10.1, I did modifications in the [Rtab-map_ros project](https://github.com/introlab/rtabmap_ros/tree/ros2) ("ros2" branch - can be found in #471), which allow me to build the package, but when I run it, I get ```double free or corruption (out)```. I tried a couple of things but no success... 
So, there is any updated guidance to build and run Rtab-map ROS2 package? Any expectation to align it with the new release of Rtab-map lib? For a stable version, it is recommended for now to use the packages for ROS1 then? 
Thanks in advance!

matlabbe, 2020-10-07:
Currently it is recommended to use ROS1 package, which is the one most up to date and always tested with latest rtabmap library. Just tried again the turtlebot3 example with the pull request above and it works on my computer (Ubuntu 18.04/eloquent):
```bash
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/ros2_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models
$ export TURTLEBOT3_MODEL=waffle
$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py

$ ros2 launch rtabmap_ros turtlebot3_scan.launch.py

[INFO] [launch]: All log files can be found below /home/mathieu/.ros/log/2020-10-07-17-29-05-655196-mathieu-XPS-15-7590-30442
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [rtabmap-1]: process started with pid [30474]
[INFO] [rtabmapviz-2]: process started with pid [30475]
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_depth = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_rgb = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_stereo = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_odom_info = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_user_data = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_scan = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap: subscribe_scan_cloud = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: queue_size    = 10
[rtabmap-1] [INFO] [rtabmap]: rtabmap: approx_sync   = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_cleanup                = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_always_update          = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[rtabmap-1] [INFO] [rtabmap]: rtabmap: frame_id      = base_footprint
[rtabmap-1] [INFO] [rtabmap]: rtabmap: map_frame_id  = map
[rtabmap-1] [INFO] [rtabmap]: rtabmap: use_action_for_goal  = false
[rtabmap-1] [INFO] [rtabmap]: rtabmap: tf_delay      = 0.050000
[rtabmap-1] [INFO] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[rtabmap-1] [INFO] [rtabmap]: rtabmap: odom_sensor_sync   = false
[rtabmap-1] [INFO] [rtabmap]: Setting RTAB-Map parameter "RGBD/NeighborLinkRefining"="True"
[rtabmap-1] [INFO] [rtabmap]: Setting RTAB-Map parameter "Reg/Strategy"="1"
[rtabmap-1] [WARN] [rtabmap]: Setting "Grid/FromDepth" parameter to false (default true) as "subscribe_scan" or "subscribe_scan_cloud" is true. The occupancy grid map will be constructed from laser scans. To get occupancy grid map from cloud projection, set "Grid/FromDepth" to true. To suppress this warning, add <param name="Grid/FromDepth" type="string" value="false"/>
[rtabmap-1] [WARN] [rtabmap]: Setting "RGBD/ProximityPathMaxNeighbors" parameter to 10 (default 0) as "subscribe_scan" is true and "Reg/Strategy" uses ICP. Proximity detection by space will be also done by merging close scans. To disable, set "RGBD/ProximityPathMaxNeighbors" to 0. To suppress this warning, add <param name="RGBD/ProximityPathMaxNeighbors" type="string" value="10"/>
[rtabmap-1] [INFO] [rtabmap]: Setting "Grid/RangeMax" parameter to 0 (default 5.000000) as "subscribe_scan" or "subscribe_scan_cloud" is true.
[rtabmap-1] [INFO] [rtabmap]: RTAB-Map detection rate = 1.000000 Hz
[rtabmap-1] [INFO] [rtabmap]: rtabmap: Deleted database "/home/mathieu/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[rtabmap-1] [INFO] [rtabmap]: rtabmap: Using database from "/home/mathieu/.ros/rtabmap.db" (0 MB).
[rtabmap-1] [INFO] [rtabmap]: rtabmap: Database version = "0.20.4".
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_depth = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_rgb = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_stereo = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_rgbd = false (rgbd_cameras=1)
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_odom_info = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_user_data = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_scan = true
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: subscribe_scan_cloud = false
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: queue_size    = 10
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: approx_sync   = true
[rtabmap-1] [INFO] [rtabmap]: Setup callbacks
[rtabmap-1] [INFO] [rtabmap]: Setup scan callback
[rtabmap-1] [INFO] [rtabmap]: 
[rtabmap-1] rtabmap subscribed to (approx sync):
[rtabmap-1]    odom,
[rtabmap-1]    scan
[rtabmapviz-2] [INFO] [rtabmapviz]: rtabmapviz: Using configuration from "/home/mathieu/.ros/rtabmapGUI.ini"
[rtabmap-1] [WARN] [rtabmap]: There is no image subscription, bag-of-words loop closure detection will be disabled...
[rtabmap-1] [WARN] [rtabmap]: Setting Kp/MaxFeatures=-1 (bag-of-words disabled)
[rtabmap-1] [INFO] [rtabmap]: rtabmap (1): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0044s, Maps update=0.0003s pub=0.0002s (local map=1, WM=1)
[rtabmapviz-2] libpng warning: iCCP: known incorrect sRGB profile
[rtabmapviz-2] libpng warning: iCCP: known incorrect sRGB profile
[... cut ...]
[rtabmap-1] [INFO] [rtabmap]: rtabmap (7): Rate=1.00s, Limit=0.000s, RTAB-Map=0.0027s, Maps update=0.0005s pub=0.0000s (local map=1, WM=1)
...

When there is a seg fault on start, it is often because wrong libraries are loaded on runtime (that are different from the libraries rtabmap has built with). You could post the cmake log for rtabmap library, and ldd for rtabmap_ros:
```
$ cd rtabmap/build
$ cmake ..

$ ldd ~/ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap
```
Do you have also ros1 installed on the same computer?

mirellameelo, 2020-10-08:
Sure! Here's the output of my "cmake .." command (rtabmap lib):

```
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Checking for module 'eigen3'
--   Found eigen3, version 3.3.4
-- Found Eigen: /usr/local/include/eigen3 (Required is at least version "3.1") 
-- Eigen found (include: /usr/local/include/eigen3, version: 3.3.4)
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.21
-- Found USB_10: /usr/lib/x86_64-linux-gnu/libusb-1.0.so  
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- The imported target "vtkRenderingPythonTkWidgets" references the file
   "/usr/lib/x86_64-linux-gnu/libvtkRenderingPythonTkWidgets.so"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- Found libusb-1.0: /usr/include  
-- Checking for module 'flann'
--   Found flann, version 1.9.1
-- Found FLANN: /usr/lib/x86_64-linux-gnu/libflann_cpp.so  
-- Found Qhull: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so  
-- QHULL found (include: /usr/include, lib: optimized;/usr/lib/x86_64-linux-gnu/libqhull.so;debug;/usr/lib/x86_64-linux-gnu/libqhull.so)
-- OpenNI found (include: /usr/include/ni, lib: /usr/lib/libOpenNI.so)
-- Found PCL_COMMON: /usr/local/lib/libpcl_common.so  
-- Found PCL_OCTREE: /usr/local/lib/libpcl_octree.so  
-- Found PCL_IO: /usr/local/lib/libpcl_io.so  
-- Found PCL_KDTREE: /usr/local/lib/libpcl_kdtree.so  
-- Found PCL_SEARCH: /usr/local/lib/libpcl_search.so  
-- Found PCL_SURFACE: /usr/local/lib/libpcl_surface.so  
-- Found PCL_FILTERS: /usr/local/lib/libpcl_filters.so  
-- Found PCL_2D: /usr/local/include/pcl-1.11  
-- Found PCL_FEATURES: /usr/local/lib/libpcl_features.so  
-- Found PCL_REGISTRATION: /usr/local/lib/libpcl_registration.so  
-- Found PCL_SAMPLE_CONSENSUS: /usr/local/lib/libpcl_sample_consensus.so  
[... 218 lines cut ...]
--   With ORB_SLAM2            = NO (WITH_G2O should be OFF as ORB_SLAM2 uses its own g2o version)
-- Show all options with: cmake -LA | grep WITH_
-- --------------------------------------------
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mirellameelo/rtabmap/build

```
After installing the lib, I use ```colcon build``` for rtablib_ros. There is an observation: every time I built this package I have an "**Internal compiler error**":

```
c++: internal compiler error: Killed (program cc1plus)
Please submit a full bug report,
with preprocessed source if appropriate.
See <file:///usr/share/doc/gcc-7/README.Bugs> for instructions.
make[2]: *** [CMakeFiles/rtabmap_sync.dir/src/impl/CommonDataSubscriberRGBD2.cpp.o] Error 4
```

But then I use ```colcon build``` again and it builds... So maybe this is causing the corruption? 

PS: I don't know what do you mean with "**ldd** for rtabmap_ros", could you please clarify it? Then I can add this info here. 

PS_2: I have the melodic version as well, but since I have the SLAM system in ROS2, would be great to have this Rtabmap package in ROS2 too, but in the last case, I'll use a bridge for ros2 and ros1 communication. =/

Thanks for your help!

matlabbe, 2020-10-08:
**ldd** tool shows the runtime libraries that a node is using. This helps to detect if different versions of OpenCV or PCL libraries are loaded. 

PCL and OpenCV are built from source, this can be a problem if [cv_bridge](https://github.com/ros-perception/vision_opencv/tree/eloquent/cv_bridge) and [pcl_ros](https://github.com/ros-perception/perception_pcl/tree/eloquent-devel) are not rebuilt also from source to link on those versions. The problem is that rtabmap library is linking on PCL and OpenCV built from source, but rtabmap_ros is also linking on `cv_bridge` and `pcl_ros`, which, if not rebuilt, would link on system's PCL and OpenCV versions. When launching rtabmap node, it will load two versions of PCL and OpenCV libraries, causing the crash. I suggest to uninstall (`sudo apt remove libopencv-dev... libpcl-dev...`) system's OpenCV and PCL to make sure that dependencies will rebuild with the right version (built from source).

mirellameelo, 2020-10-08:
Thanks for your patience!
1) I notice that I had two PCL lib versions in /usr/local/lib, and I could see that Rtabmap lib was mixing them, for example, getting some components of PCL 8.0 and other from PCL 11.1. I discovered it when I uninstall PCL 8.0 and I could not build anymore the Rtabmap lib since some components were not found anymore. In the end, I uninstalled  PCL 11.1 as well and installed 10.0.0 (also I had to downgrade my Eigen version to 3.4). The Rtabmap lib was built, hopefully, in the proper way.

2) I uninstalled OpenCV and PCL as you suggested (```sudo apt remove libopencv-dev... libpcl-dev```). Then, I created the folder: ros2_ws/src, and inside that I put [vision_opencv]( https://github.com/ros-perception/vision_opencv/tree/eloquent) and [perception_pcl](https://github.com/ros-perception/perception_pcl/tree/eloquent-devel), both from the eloquent branch. Since ```colcon build``` deals with dependencies, I could use it inside the ros2_ws folder. It crashes at least 2 times, but in the end, it builds.

3) Time to run it, and I have a new error:
```
ros2_ws/install/rtabmap_ros/lib/rtabmap_ros/rtabmap: error while loading shared libraries: libpcl_surface.so.1.10: cannot open shared object file: No such file or directory
```
but the file exists, so I use: ```LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib```

And then, ```double free or corruption (out)``` =/

I`ll try ROS1 version and check how viable to use the bridge for communication. Feel free for new suggestions.

Thanks once more!

matlabbe, 2020-10-09:
RTAB-Map [should build with PCL 1.11](https://github.com/introlab/rtabmap/commit/24052a6ebe65fb260ee1d838ca5305b12b15d928). When changing OpenCV and pcl versions, cleanup the build directory to make sure cmake cache is cleared and no remaining paths of old libraries are still there.

After doing `make` in the build directory, you can use colcon afterwards. 

This is the output of the rtabmap library (the same library is used with ros1 and ros2):
```bash
~/workspace/rtabmap/build$ cmake ..
CMake Deprecation Warning at CMakeLists.txt:64 (cmake_policy):
  The OLD behavior for policy CMP0020 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- VTK_MODULES_REQUESTED=vtkChartsCore;vtkCommonColor;vtkCommonComputationalGeometry;vtkCommonCore;vtkCommonDataModel;vtkCommonExecutionModel;vtkCommonMath;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkDICOMParser;vtkDomainsChemistry;vtkFiltersAMR;vtkFiltersCore;vtkFiltersExtraction;vtkFiltersFlowPaths;vtkFiltersGeneral;vtkFiltersGeneric;vtkFiltersGeometry;vtkFiltersHybrid;vtkFiltersHyperTree;vtkFiltersImaging;vtkFiltersModeling;vtkFiltersParallel;vtkFiltersParallelFlowPaths;vtkFiltersParallelGeometry;vtkFiltersParallelImaging;vtkFiltersParallelMPI;vtkFiltersParallelStatistics;vtkFiltersProgrammable;vtkFiltersReebGraph;vtkFiltersSMP;vtkFiltersSelection;vtkFiltersSources;vtkFiltersStatistics;vtkFiltersTexture;vtkFiltersVerdict;vtkGUISupportQt;vtkGUISupportQtOpenGL;vtkGUISupportQtSQL;vtkGUISupportQtWebkit;vtkGeovisCore;vtkIOAMR;vtkIOCore;vtkIOEnSight;vtkIOExodus;vtkIOExport;vtkIOFFMPEG;vtkIOGDAL;vtkIOGeoJSON;vtkIOGeometry;vtkIOImage;vtkIOImport;vtkIOInfovis;vtkIOLSDyna;vtkIOLegacy;vtkIOMINC;vtkIOMPIImage;vtkIOMPIParallel;vtkIOMovie;vtkIOMySQL;vtkIONetCDF;vtkIOODBC;vtkIOPLY;vtkIOParallel;vtkIOParallelExodus;vtkIOParallelLSDyna;vtkIOParallelNetCDF;vtkIOParallelXML;vtkIOPostgreSQL;vtkIOSQL;vtkIOVPIC;vtkIOVideo;vtkIOXML;vtkIOXMLParser;vtkIOXdmf2;vtkImagingColor;vtkImagingCore;vtkImagingFourier;vtkImagingGeneral;vtkImagingHybrid;vtkImagingMath;vtkImagingMorphological;vtkImagingSources;vtkImagingStatistics;vtkImagingStencil;vtkInfovisBoost;vtkInfovisBoostGraphAlgorithms;vtkInfovisCore;vtkInfovisLayout;vtkInteractionImage;vtkInteractionStyle;vtkInteractionWidgets;vtkLocalExample;vtkMetaIO;vtkParallelCore;vtkParallelMPI;vtkParseOGLExt;vtkRenderingAnnotation;vtkRenderingContext2D;vtkRenderingContextOpenGL;vtkRenderingCore;vtkRenderingExternal;vtkRenderingFreeType;vtkRenderingFreeTypeFontConfig;vtkRenderingGL2PS;vtkRenderingImage;vtkRenderingLIC;vtkRenderingLOD;vtkRenderingLabel;vtkRenderingOpenGL;vtkRenderingParallel;vtkRenderingParallelLIC;vtkRenderingQt;vtkRenderingTk;vtkRenderingVolume;vtkRenderingVolumeAMR;vtkRenderingVolumeOpenGL;vtkTclTk;vtkTestingCore;vtkTestingGenericBridge;vtkTestingIOSQL;vtkTestingRendering;vtkUtilitiesEncodeString;vtkUtilitiesHashSource;vtkVPIC;vtkViewsContext2D;vtkViewsCore;vtkViewsGeovis;vtkViewsInfovis;vtkViewsQt;vtkWrappingJava;vtkWrappingTcl;vtkWrappingTools;vtkalglib;vtkexodusII;vtkexpat;vtkfreetype;vtkftgl;vtkgl2ps;vtkglew;vtkhdf5;vtkjpeg;vtkjsoncpp;vtklibproj4;vtklibxml2;vtknetcdf;vtkoggtheora;vtkpng;vtksqlite;vtksys;vtktiff;vtkverdict;vtkxdmf2;vtkzlib
-- VTK_MODULES_REQUESTED=vtkChartsCore;vtkCommonColor;vtkCommonComputationalGeometry;vtkCommonCore;vtkCommonDataModel;vtkCommonExecutionModel;vtkCommonMath;vtkCommonMisc;vtkCommonSystem;vtkCommonTransforms;vtkDICOMParser;vtkDomainsChemistry;vtkFiltersAMR;vtkFiltersCore;vtkFiltersExtraction;vtkFiltersFlowPaths;vtkFiltersGeneral;vtkFiltersGeneric;vtkFiltersGeometry;vtkFiltersHybrid;vtkFiltersHyperTree;vtkFiltersImaging;vtkFiltersModeling;vtkFiltersParallel;vtkFiltersParallelFlowPaths;vtkFiltersParallelGeometry;vtkFiltersParallelImaging;vtkFiltersParallelMPI;vtkFiltersParallelStatistics;vtkFiltersProgrammable;vtkFiltersReebGraph;vtkFiltersSMP;vtkFiltersSelection;vtkFiltersSources;vtkFiltersStatistics;vtkFiltersTexture;vtkFiltersVerdict;vtkGUISupportQt;vtkGUISupportQtOpenGL;vtkGUISupportQtSQL;vtkGUISupportQtWebkit;vtkGeovisCore;vtkIOAMR;vtkIOCore;vtkIOEnSight;vtkIOExodus;vtkIOExport;vtkIOFFMPEG;vtkIOGDAL;vtkIOGeoJSON;vtkIOGeometry;vtkIOImage;vtkIOImport;vtkIOInfovis;vtkIOLSDyna;vtkIOLegacy;vtkIOMINC;vtkIOMPIImage;vtkIOMPIParallel;vtkIOMovie;vtkIOMySQL;vtkIONetCDF;vtkIOODBC;vtkIOPLY;vtkIOParallel;vtkIOParallelExodus;vtkIOParallelLSDyna;vtkIOParallelNetCDF;vtkIOParallelXML;vtkIOPostgreSQL;vtkIOSQL;vtkIOVPIC;vtkIOVideo;vtkIOXML;vtkIOXMLParser;vtkIOXdmf2;vtkImagingColor;vtkImagingCore;vtkImagingFourier;vtkImagingGeneral;vtkImagingHybrid;vtkImagingMath;vtkImagingMorphological;vtkImagingSources;vtkImagingStatistics;vtkImagingStencil;vtkInfovisBoost;vtkInfovisBoostGraphAlgorithms;vtkInfovisCore;vtkInfovisLayout;vtkInteractionImage;vtkInteractionStyle;vtkInteractionWidgets;vtkLocalExample;vtkMetaIO;vtkParallelCore;vtkParallelMPI;vtkParseOGLExt;vt [... cut ...]
I use opencv and pcl system's libraries (from `apt get`). I am using g2o and octomap built from source (removed ros1 g2o and octomap binary libraries), so I can use the rtabmap library in ros1 and ros2. As a rule of thumb, if the dependencies are system or built from source, they should work in both ros1 and ros2.

For the FastRTPS error, not sure why on my system I don't have the error.

When building PCL >=1.8 from source, it uses `-march=native` by default (see cmake log above). It means that GTSAM and g2o should be built (if used) with that compilation flag too (Eigen issue).

mirellameelo, 2020-10-12:
Hi. It is working!
Since I had some libs built from source and other from the system's libraries, it has been hard to me to find which one was corrupting the Rtabmap package. I opted for the more radical option by reinstalling Ubuntu. Then:
1) I installed [ROS2 eloquent](https://index.ros.org/doc/ros2/Installation/Eloquent/Linux-Install-Debians/) from debian packages
2) I installed the dependencies from the system's libraries (eigen, qt5, opencv, pcl....) 
3) I followed the [readme](https://github.com/introlab/rtabmap_ros/blob/ros2/README.md) file

OBS: I believe it is only possible to run the example by building the Turtlebot3 package from source in case using eloquent version, which I did, but not sure the proper Gazebo version to run ```ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py``` without errors. But anyways, the Rtabmap is working for me now. :)
  
Thanks for your time!!

closed 2020-10-12 by mirellameelo