introlab/rtabmap_ros #473: Odometry node floods terminal with info

ArvidPeterson, 2020-10-08:
Odom node floods the terminal with info messages like this:

`
[ INFO] [1602166630.272139336]: Odom: quality=173, std dev=0.003864m|0.053164rad, update time=0.040281s
`

Since these are printed multiple times every second it makes it harder to look for other warnings etc. Is there a quick fix for this?

//Arvid P

matlabbe, 2020-10-08:
I don't think a change in the code is required. There are some other options (more standard to ROS). First you can remove `output="screen"`from the launch files. If you are using `rtabmap.launch`, you can launch it with `output:="log"` to only output in log file. The info msg may still be output to log file, increasing its size for nothing. If you want to set the minimum logging level to warning, it is possible by creating a [config file](http://wiki.ros.org/rosconsole#Configuration) or calling service to adjust the logger level: https://answers.ros.org/question/197964/set-log-level-in-launch-file/

Example with service:
```bash
rosservice call /rtabmap/rgbd_odometry/set_logger_level "logger: 'ros' level: 'warn'"
# for rtabmap node:
rosservice call /rtabmap/rtabmap/set_logger_level "logger: 'ros' level: 'warn'"
```
Well, the most cleanest way is the [global config file](http://wiki.ros.org/rosconsole#Configuration):
```
echo -e "log4j.logger.ros=WARN\nlog4j.logger.ros.roscpp.superdebug=WARN" >> ~/.ros/rosconsole.config
export ROSCONSOLE_CONFIG_FILE=~/.ros/rosconsole.config
```
Afterwards, you can add the export above in your `~/.bashrc` to avoid exporting the environment variable on each new terminal.

cheers,
Mathieu

ArvidPeterson, 2020-10-09:
Wow! Thank you for taking the time to write a thorough explanation.

//Arvid P

closed 2020-10-09 by ArvidPeterson