introlab/rtabmap_ros #475: rgbd_sync error: Input stamps changed between the beginning and the end of the callback, but I'm using

tianshiz, 2020-10-08:
I'm getting errors like this from rgbd_sync, strange cause it doesn't complain when using simulated cameras.

`"Input stamps changed between the beginning and the end of the callback! Make ""sure the node publishing the topics doesn't override the same data after publishing them. A ""solution is to use this node within another nodelet manager. Stamps`

Here is my launch file. I have 3 cameras, each under its own nodelet manager camera_managerX. So it's strange that I would get this error.

`<launch> 
<group ns="camera_manager0">
<node name="rgbd_sync" pkg="nodelet" type="nodelet" args="load rtabmap_ros/rgbd_sync camera_manager0"  >
<remap from="rgb/image" to="zed_node/rgb/image_rect_color" />
<remap from="depth/image" to="zed_node/depth/depth_registered" />
<remap from="rgb/camera_info" to="zed_node/rgb/camera_info" />
<param name="approx" value="true" />
<param name="compressed_rate" value="5" />
</node>
<node name="rgbd_odometry" pkg="nodelet" type="nodelet" args="load rtabmap_ros/rgbd_odometry camera_manager0"  >
<param name="subscribe_rgbd" value="true" />
<param name="rgbd_cameras" value="1" />
<param name="frame_id" value="base_link" />
<param name="publish_tf" value="false" />
<param name="initial_pose" value="0 0 0.59 0 0 0" />
<remap from="odom" to="zed_node/odom" />
</node>
</group>
<group ns="camera_manager1">
<node name="rgbd_sync" pkg="nodelet" type="nodelet" args="load rtabmap_ros/rgbd_sync camera_manager1"  >
<remap from="rgb/image" to="zed_node/rgb/image_rect_color" />
<remap from="depth/image" to="zed_node/depth/depth_registered" />
<remap from="rgb/camera_info" to="zed_node/rgb/camera_info" />
<param name="approx" value="true" />
<param name="compressed_rate" value="5" />
</node>
<node name="points_xyz_planner" pkg="nodelet" type="nodelet" args="load rtabmap_ros/point_cloud_xyz camera_manager1"  >
<remap from="depth/image" to="zed_node/depth/depth_registered" />
<remap from="depth/camera_info" to="zed_node/rgb/camera_info" />
<param name="voxel_size" value="0.05" />
<param name="decimation" value="4" />
<param name="max_depth" value="20" />
</node>
<node name="obstacles_detection" pkg="nodelet" type="nodelet" args="load rtabmap_ros/obstacles_detection camera_manager1"  >
<param name="frame_id" value="base_footprint" />
<param name="map_frame_id" value="map" />
<param name="Grid/MaxObstacleHeight" value="1.7" />
<param name="Grid/MaxGroundAngle" value="50" />
</node>
</group>
<group ns="camera_manager2">
<node name="rgbd_sync" pkg="nodelet" type="nodelet" args="load rtabmap_ros/rgbd_sync camera_manager2"  >
<remap from="rgb/image" to="zed_node/rgb/image_rect_color" />
<remap from="depth/image" to="zed_node/depth/depth_registered" />
<remap from="rgb/camera_info" to="zed_node/rgb/camera_info" />
<param name="approx" value="true" />
<param name="compressed_rate" value="5" />
</node>
</group>
<group ns="rtabmap">
<node name="rtabmap" pkg="rtabmap_ros" type="rtabmap" args="--delete_db_on_start"  >
<rosparam file="$(find mantis_nav)/config/rtabmap.yaml" />
<remap from="grid_map" to="/map" />
<remap from="rgbd_image0" to="/camera_manager0/rgbd_image/compressed" />
<remap from="rgbd_image1" to="/camera_manager1/rgbd_image/compressed" />
<remap from="rgbd_image2" to="/camera_manager2/rgbd_image/compressed" />
<param name="rgbd_cameras" value="3" />
</node>
</group>
</launch>
`

matlabbe, 2020-10-17:
Try with standalone nodelets, instead of:
```xml
args="load rtabmap_ros/rgbd_sync camera_manager0"
```
try:
```xml
args="standalone rtabmap_ros/rgbd_sync"
```

If the camera driver nodelet for the camera0 is also in camera_manager0, this could be the problem. If the camera driver and rgbd_sync are in the same manager, only a pointer of the image is passed between them. If the camera driver reupdate the image in the same pointer while rgbd_sync is accessing it, the warning above will appear.

tianshiz, 2020-10-17:
Yea my camera nodelet is under the same camera manager as rgbd sync. Was
hoping to save a little bandwidth from having to pass my images to that
nodelet.

Everything still works despite the warning. Is there any issue with this?

matlabbe, 2020-10-18:
Which camera driver are you using? A solution could be to make sure in the camera node that the data in the pointer published doesn't get modified.

tianshiz, 2020-10-20:
I'm using zed nodelets. That sounds like it would take some changes to the
zed wrapper source?

matlabbe, 2020-10-20:
I was looking at the zed wrapper: https://github.com/stereolabs/zed-ros-wrapper/blob/bb3a5e53d8d1921e99ea3db31e4ca49584434bb2/zed_nodelets/src/zed_nodelet/src/zed_wrapper_nodelet.cpp#L1652-L1658
It seems the image data is copied to a new Image pointer, but camera_info pointer is reused between each call. However, I don't use the camera_info stamps in this comparison: https://github.com/introlab/rtabmap_ros/blob/c23dbbc24969b7db440eb75c2745978895b4b73c/src/nodelets/rgbd_sync.cpp#L276-L277

This could be a ros::Time to double conversion problem, which makes the comparison failing afterwards. In your original post above, you didn't show the stamps in the error, what are they? Are they very close?

tianshiz, 2020-10-21:
` Stamps: rgb=1603234530.062429->1603234530.127306 depth=1603234530.062429->1603234530.062429`

they are very close. I think you got to the root of the issue.

matlabbe, 2020-10-28:
```
1603234530.062429
1603234530.127306
```
To be a double precision conversion error, I would have expected an error in μs, not ms. Here the difference is 64.877 ms, which seems matching a camera framerate of 15 Hz. The publisher node seems indeed modifying the topic while the subscriber is processing it.

tianshiz, 2020-11-02:
hmm, so this may be a zed publisher issue. I'll check with them for any inputs.
thanks for looking into this

closed 2020-11-02 by tianshiz