introlab/rtabmap_ros #476: does the occupancy grid /octomap gets updated in localization mode [question]

NicoNorena, 2020-10-12:
@matlabbe 
Does the occupancy grid and octomap in RTABmap gets updated by appending the newest node when running in localization mode? meaning whether the topic "octomap_binary" would show the new objects or moving objects without actually adding them  to the map. I am using  it in RGBD mode. 

I read that when it has a laser and RGBD, it uses the laser for updating with newest node when setting the parameter "raytracing".  Is this correct?

matlabbe, 2020-10-12:
By default no. If you set `map_always_update` to `true` for rtabmap node, the occupancy grids/octomap will get updated without actually changing the original map (i.e., the new obstacles added won't be there if you restart rtabmap). If you enable `Grid/RayTracing` (default true for scans), it will be able to clear obstacles not here anymore.

NicoNorena, 2020-10-13:
cool thanks, I will test it out.

can  `Grid/RayTracing` be used when only using RGBD?  basically is a scan input required when enabling `Grid/RayTracing`

matlabbe, 2020-10-17:
`Grid/RayTracing` works with stereo and RGB-D data. You can see the Figure 7 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf) for all possible combinations:
[image]

Note that if Grid/3D is enabled, OctoMap's 3D ray tracing is done, which can be computationally expensive.

closed 2020-10-21 by NicoNorena