introlab/rtabmap_ros #481: Building RTABMAP from source gives undefined reference error.

s3668134, 2020-10-21:
Hello everyone,

I haven't been using ros for very long so I'm quite new to this. I've been trying to follow the following link to install rtabmap_ros from source [https://github.com/introlab/rtabmap_ros#build-from-source ](url), when reaching the make command in step two, I've been given this error, shown below. I know that this error is similar to some posted before. 

Could this be because in our CMakeLists.txt file, it was said that it's using pcl 1.8 rather than the default pcl 1.7?

`[  4%] Built target rtabmap_utilite
[  5%] Built target res_tool
[ 42%] Built target rtabmap_core
[ 80%] Built target rtabmap_gui
[ 80%] Linking CXX executable ../../../bin/rtabmap
../../../bin/librtabmap_core.so.0.20.5: undefined reference to pcl::search::Search<pcl::PointXYZ>::getName() const
../../../bin/librtabmap_gui.so.0.20.5: undefined reference to pcl::search::Search<pcl::PointXYZRGBNormal>::getName() const
../../../bin/librtabmap_core.so.0.20.5: undefined reference to pcl::search::Search<pcl::PointXYZI>::getName() const
../../../bin/librtabmap_core.so.0.20.5: undefined reference to pcl::search::Search<pcl::PointXYZRGB>::getName() const
../../../bin/librtabmap_core.so.0.20.5: undefined reference to pcl::search::Search<pcl::PointNormal>::getName() const
collect2: error: ld returned 1 exit status
app/src/CMakeFiles/rtabmap.dir/build.make:498: recipe for target ../bin/rtabmap failed
make[2]: *** [../bin/rtabmap] Error 1
CMakeFiles/Makefile2:416: recipe for target app/src/CMakeFiles/rtabmap.dir/all failed
make[1]: *** [app/src/CMakeFiles/rtabmap.dir/all] Error 2
Makefile:151: recipe for target all  failed
make: *** [all] Error 2`

I also know in this forum, it was said that uninstalling pcl and reinstalling pcl would fix it but I would not know how to go about this either [https://github.com/PointCloudLibrary/pcl/issues/2406#issuecomment-417831432](url)

How can I go about fixing this error?

matlabbe, 2020-10-28:
What is your system as the pcl binaries should be ok? RTAB-Map should build with PCL from version 1.7 to latest one.

s3668134, 2020-11-03:
@matlabbe thank you for your help, I found another post in which re-installing the rtabmap libraries and installing a newer version of the pcl libraries was able to solve my issue [http://official-rtab-map-forum.67519.x6.nabble.com/Compiling-in-ROS-Indigo-with-PCL-1-7-2-Undefined-reference-to-td386.html](url)

closed 2020-11-03 by s3668134