introlab/rtabmap_ros #482: ROS2 status

Marsfan, 2020-10-21:
I am currently in the planning stages of a project. We intend to use ROS2 as this is a clean sheet design. What is the status/functionality of the ROS2 build of RTABmap and what versions of ROS2 is it known to work with?

matlabbe, 2020-10-28:
I tested ros2 branch primarily on Crystal. I expect that it would work on Foxy. While the ros2 interface is lacking latest updates from ros1 interface, it should build with latest rtabmap library version.

Marsfan, 2020-10-28:
@matlabbe Awesome. Thanks. Looking forward to actually using RTABMap for its intended use. If I stumble across any ROS2 bugs I plant to try to patch them and push back to you, or at least report them.

blacksoul000, 2020-11-02:
@matlabbe 
I don't know where to post it but i think you will be interested it and may be comment.
https://github.com/ros2/geometry2/issues/338.
Got this "feature" in retabmap