introlab/rtabmap_ros #483: Does rtabmap accept transparency?

Andyshen555, 2020-10-22:
I would like to encode some data in the 'a' channel while constructing the point cloud, but not sure if rgb/image accepts that. Does it work?

Andyshen555, 2020-10-22:
It seems not accepting the 'a' channel, but if I want to modify the source code to enable that, which files should I get my hands on? I fear changing some lines will make the entire rtabmap fail.

matlabbe, 2020-10-28:
I would not see a problem to use alpha channel... for the logic. The thing is that all code based on the number of channels (1 or 3) like [this](https://github.com/introlab/rtabmap/blob/72e1649cdde0533250108a8c52b3d4f66eb10d6a/corelib/src/Memory.cpp#L4089-L4091) or [this](https://github.com/introlab/rtabmap/blob/72e1649cdde0533250108a8c52b3d4f66eb10d6a/corelib/src/util3d.cpp#L516-L524) will break. Accepting 4 channels could be done (having `if(imageRgb.channels() == 3 || imageRgb.channels() == 4)` for example), but many places would have to be updated. To render the point clouds with an alpha channel, it seems possible to have a type pcl::PointXYZRGBA, but a lot of place in the gui library pcl::PointXYZRGB is hard-coded. They would have all to be changed to pcl::PointXYZRGBA to avoid duplicating code. I don't remember if pcl::PointXYZRGBA exists since PCL 1.7.2. When images are [compressed](https://github.com/introlab/rtabmap/blob/72e1649cdde0533250108a8c52b3d4f66eb10d6a/corelib/src/Memory.cpp#L5038), by default they are compressed in JPEG, they would have to be compressed in PNG when 4 channels is detected. 

On the ros side, beware that there are some places like this:
https://github.com/introlab/rtabmap_ros/blob/c23dbbc24969b7db440eb75c2745978895b4b73c/src/MsgConversion.cpp#L1710
which automatically converts to `bgr8` from any color format (e.g., bayer, rgb, rgba, bgr, bgra...). You would have to add a special case when a format with alpha channel is detected, to convert to `bgra8`.

In summary there are asserts at many places in the code, so you could change some code, check until it asserts and modify the next one.