introlab/rtabmap_ros #484: Initialization in Localization mode and localization pose update [question]

NicoNorena, 2020-10-23:
@matlabbe 
It looks like from rviz, the initial pose can be set over the topic `initialpose`. I don't see  that it is updating the starting frame "map". Does it needs to be reloaded after setting a new initialization pose? or what is this callback `initialPoseCallback` suppose to do? Is the tf 'map' based on the initialization pose ? Ideally I would like to see the odometry pose reference different from the global pose in the map.

from my understanding `localization_pose` gets updated if a loop  closure is found, but it does not come up until later when moving around and it does not update continuously.
- Is there a way for the pose to be updated more often?
-  Is there a way to request the pose until it converges  at initialization?

NicoNorena, 2020-10-28:
@matlabbe 
When the function `setInitialPose` gets called:
https://github.com/introlab/rtabmap/blob/master/corelib/src/Rtabmap.cpp#L709

what happens if the the function `optimizeCurrentMap` can't optimized the poses based on the new initial position?
https://github.com/introlab/rtabmap/blob/72e1649cdde0533250108a8c52b3d4f66eb10d6a/corelib/src/Rtabmap.cpp#L4159-L4167

would it required for 2d pose estimates from RVIZ to be called again until it resolves the new initial pose? there is an issue on this https://github.com/introlab/rtabmap_ros/issues/220
Would it make sense to continue calling `optimizeCurrentMap` with a new captured `getLastWorkingSignature` for couple of times if it fails?
how many Neighbor ids (`memory->getNeighborsId`) get retrieved when called?

matlabbe, 2020-11-07:
`setInitialPose` can be used only in localization mode. It will force rtabmap to update `map->odom` so that the robot is located at the requested position in the map.

`optimizeCurrentMap` won't be called in localization mode. `optimizeCurrentMap` is called only when a loop closure happens (in slam mode).

For `getNeighborsId`, there is a parameter to limit the number of neighbors, otherwise the whole graph would be returned:
https://github.com/introlab/rtabmap/blob/7be22d1b67c6a1f95e61cff0b66b9ef178d2c7ff/corelib/src/Memory.cpp#L1341-L1354

NicoNorena, 2020-11-10:
thanks, I noticed that in RVIZ i was not setting the prefix to be rtabmap, after I corrected the topic to  /rtabmap/initialpose, Rtabmap was able to received  the initial position and force it to start there as you described.

thanks.

closed 2020-11-10 by NicoNorena