introlab/rtabmap_ros #485: catkin_make error

zxsawqedcrfv, 2020-10-23:
Hi, I'm new for ros and github. 

I did successfully built "rtabmap_ros" package into my environment and was able to run mapping. Until few days ago, I want to add another package to my workspace and I ran "**camkin_make**", but it seems to **have some errors while building** "rtabmap_ros". 

Even if I create a new workspace and **reinstall and rebuild "rtabmap_ros" separately** in the new workspace, it will encounter the same errors:

```
Base path: /home/t3204/catkin_ws
Source space: /home/t3204/catkin_ws/src
Build space: /home/t3204/catkin_ws/build
Devel space: /home/t3204/catkin_ws/devel
Install space: /home/t3204/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/t3204/catkin_ws/build"
####
####
#### Running command: "make -j1" in "/home/t3204/catkin_ws/build"
####
[  1%] Built target rtabmap_costmap_plugins
[  2%] Built target rtabmap_pointcloud_to_depthimage
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_ScanDescriptor
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_SetLabel
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Point3f
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Info
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GetPlan
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_MapData
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_UserData
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_ResetPose
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_RGBDImage
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GetNodesInRadius
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GlobalDescriptor
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap2
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_PublishMap
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Point2f
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Link
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_ListLabels
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_SetGoal
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_AddLink
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GetNodeData
[  2%] Built target sensor_msgs_generate_messages_eus
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_OdomInfo
[  2%] Built target geometry_msgs_generate_messages_eus
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_KeyPoint
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Path
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_EnvSensor
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GetMap
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_GPS
[  2%] Built target std_msgs_generate_messages_eus
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_Goal
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_MapGraph
[  2%] Built target _rtabmap_ros_generate_messages_check_deps_NodeData
[ 13%] Built target rtabmap_ros_generate_messages_eus
[ 14%] Built target rtabmap_ros_gencfg
[ 14%] Built target std_msgs_generate_messages_py
[ 14%] Built target geometry_msgs_generate_messages_py
[ 14%] Built target sensor_msgs_generate_messages_py
[ 27%] Built target rtabmap_ros_generate_messages_py
[ 27%] Built target geometry_msgs_generate_messages_lisp
[ 27%] Built target std_msgs_generate_messages_lisp
[ 27%] Built target sensor_msgs_generate_messages_lisp
[ 38%] Built target rtabmap_ros_generate_messages_lisp
[ 38%] Built target sensor_msgs_generate_messages_cpp
[ 38%] Built target geometry_msgs_generate_messages_cpp
[ 38%] Built target std_msgs_generate_messages_cpp
[ 50%] Built target rtabmap_ros_generate_messages_cpp
[ 50%] Built target geometry_msgs_generate_messages_nodejs
[ 50%] Built target std_msgs_generate_messages_nodejs
[ 50%] Built target sensor_msgs_generate_messages_nodejs
[ 61%] Built target rtabmap_ros_generate_messages_nodejs
[ 63%] Built target rtabmap_ros
[... 80 lines cut ...]
/home/t3204/catkin_ws/src/rtabmap_ros/src/costmap_2d/voxel_layer.cpp: In member function ‘virtual void rtabmap_ros::VoxelLayer::updateOrigin(double, double)’:
/home/t3204/catkin_ws/src/rtabmap_ros/src/costmap_2d/voxel_layer.cpp:409:93: error: no matching function for call to ‘tf::TransformListener::lookupTransform(std::__cxx11::string&, std::__cxx11::string&, ros::Time)’
 Stamped = tf_->lookupTransform(global_frame_, robot_base_frame_, ros::Time(0));
                                                                              ^
In file included from /opt/ros/kinetic/include/costmap_2d/layer.h:43:0,
                 from /home/t3204/catkin_ws/src/rtabmap_ros/src/costmap_2d/voxel_layer.h:41,
                 from /home/t3204/catkin_ws/src/rtabmap_ros/src/costmap_2d/voxel_layer.cpp:38:
/opt/ros/kinetic/include/tf/tf.h:129:8: note: candidate: void tf::Transformer::lookupTransform(const string&, const string&, const ros::Time&, tf::StampedTransform&) const
   void lookupTransform(const std::string& target_frame, const std::string& sour
        ^
/opt/ros/kinetic/include/tf/tf.h:129:8: note:   candidate expects 4 arguments, 3 provided
/opt/ros/kinetic/include/tf/tf.h:142:8: note: candidate: void tf::Transformer::lookupTransform(const string&, const ros::Time&, const string&, const ros::Time&, const string&, tf::StampedTransform&) const
   void lookupTransform(const std::string& target_frame, const ros::Time& target
        ^
/opt/ros/kinetic/include/tf/tf.h:142:8: note:   candidate expects 6 arguments, 3 provided
rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins2.dir/build.make:62: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins2.dir/src/costmap_2d/voxel_layer.cpp.o' failed
make[2]: *** [rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins2.dir/src/costmap_2d/voxel_layer.cpp.o] Error 1
CMakeFiles/Makefile2:6357: recipe for target 'rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins2.dir/all' failed
make[1]: *** [rtabmap_ros/CMakeFiles/rtabmap_costmap_plugins2.dir/all] Error 2
Makefile:138: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j1" failed
```

My ROS distribution is Kinetic, does anyone tell me how to fix it?  Thanks.

matlabbe, 2020-10-28:
The` costmap_2d` package may have been installed by installing your new packages. When rtabmap detects you have that package, it builds some `costmap_2d` plugins. Depending on the costmap2d version, they used a `pcl::PointCloud<>` format or `PointCloud2` format. `rtabmap_ros` normally detects which `costmap_2d` version we are using:
https://github.com/introlab/rtabmap_ros/blob/c23dbbc24969b7db440eb75c2745978895b4b73c/CMakeLists.txt#L488-L490
 to use the right type:
https://github.com/introlab/rtabmap_ros/blob/c23dbbc24969b7db440eb75c2745978895b4b73c/src/costmap_2d/voxel_layer.cpp#L157-L175

In your case, it seems detecting `costmap_2d` version is over 1.15, but uses the old `pcl::PointCloud<>` interface. Which costmap_2d version is installed? The condition could fail if `costmap_2d` version is `0.16`. Well, the use of the `costmap_2d` can be disabled by commenting this line if it causes too much problems and you don't need rtabmap's costmap_2d plugins: 
https://github.com/introlab/rtabmap_ros/blob/c23dbbc24969b7db440eb75c2745978895b4b73c/CMakeLists.txt#L26

zxsawqedcrfv, 2020-11-09:
> The` costmap_2d` package may have been installed by installing your new packages. When rtabmap detects you have that package, it builds some `costmap_2d` plugins. Depending on the costmap2d version, [...]

Thank you! Your answer is clear and make me easy to understand. I can pass the compilation now!
I checked my costmap_2d version is 1.16.2 and as you said it's installed by my new package. I delete it from my working space for now, after that everything works fine.

closed 2020-11-16 by zxsawqedcrfv