introlab/rtabmap_ros #487: EnvSensor’ in namespace ‘rtabmap_ros’ does not name a type

ghokulji, 2020-10-26:
I built the rtabmap_ros package from source, I am facing the error : ‘EnvSensor’ in namespace ‘rtabmap_ros’ does not name a type
 error: ‘rtabmap_ros::EnvSensor’ has not been declared.

[image]

The only error coming while catkin_make is the env_sensor erroe

matlabbe, 2020-11-13:
The message should be built: https://github.com/introlab/rtabmap_ros/blob/f748a671841be68698e3b702619e594bf62d74b0/CMakeLists.txt#L109

ghokulji, 2020-11-19:
Yes the setup.bassh is sourced.
I reinstalled the rtabmap and rtabmap_ros packaages and now it is working fine.

closed 2020-11-19 by ghokulji