introlab/rtabmap_ros #489: Localization Metric

David9696, 2020-10-31:
Hi,

In localization mode, is there a way or metric to determine how good the localization is? Possible to tell whether it is lost from a metric or topic?

Thanks.

David

matlabbe, 2020-11-13:
In localization mode, there is no mechanism to evaluate online how good the localization is with a ground truth. It is also difficult to know if the robot is lost, or just exploring new areas. You could check the [/rtabmap/info](http://docs.ros.org/en/jade/api/rtabmap_ros/html/msg/Info.html) topic, which will tell if a localization occurred with the last frame. If you didn't have a localization for a while, maybe the robot can be considered lost.

David9696, 2020-11-15:
Noted with thanks. Thanks for your great package. Good job. Will close this issue.

closed 2020-11-15 by David9696