introlab/rtabmap_ros #492: Support ORB SLAM 3?

colorDogs, 2020-11-04:
ORB SLAM3 supports stereo + IMU mode tracking, and it does not require image dedistortion processing. Rtabmap requires stereo images to be distorted, resulting in a smaller camera field. Can ORB SLAM3 be supported by Tracking?

zhuoyan28, 2024-03-24:
Can someone tell
1, RTAB-MAP commands to run KITTI image dataset alone, such as binocular images?
Where will the output be stored in the ros environment?
2. ORBSLAM+RTAB-MAP running KITTI image dataset, such as binocular image command?
Where will the output be stored in ros environment?

matlabbe, 2024-03-25:
@zhuoyan28 see https://github.com/introlab/rtabmap_ros/issues/363#issuecomment-2017035795