introlab/rtabmap_ros #493: Dual RGBD camera not working with ROS RTABMAP

Sd1010, 2020-11-05:
Hi,

I have a ROS rtabmap launch file that uses 2 Realsense Depth cameras. I recently reinstalled Rtabmap and now my launch file is not working. Its giving me the following error

`[FATAL] [1604612446.424106081]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)`

I have used `catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON` but still it's showing this error.

OS - Ubuntu 18.04.5 LTS
ROS version - Melodic

matlabbe, 2020-11-07:
Normally, `catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON` should have done the trick. You could try to see if cmake passes in those lines:
https://github.com/introlab/rtabmap_ros/blob/f748a671841be68698e3b702619e594bf62d74b0/CMakeLists.txt#L206-L215
by adding a `MESSAGE(STATUS "*****************************************************************************")`

Sd1010, 2020-11-17:
Hi @matlabbe,
Thank you for your reply. I have added the message status in rtabmap_ros/CMakeLists.txt
```

IF(RTABMAP_SYNC_MULTI_RGBD)
  MESSAGE(STATUS "1111111111111111111111111111111111111111111")
  SET(rtabmap_sync_lib_src
    ${rtabmap_sync_lib_src}
    src/impl/CommonDataSubscriberRGBD2.cpp
    src/impl/CommonDataSubscriberRGBD3.cpp
    src/impl/CommonDataSubscriberRGBD4.cpp
  )
  MESSAGE(STATUS "2222222222222222222222222222222222222222222")
ENDIF(RTABMAP_SYNC_MULTI_RGBD)
  MESSAGE(STATUS "3333333333333333333333333333333333333333333")
```
This is the output i am getting:

```
-- The imported target "vtk" references the file
   "/usr/bin/vtk"
but this file does not exist.  Possible reasons include:
* The file was deleted, renamed, or moved to another location.
* An install or uninstall procedure did not complete successfully.
* The installation package was faulty and contained
   "/usr/lib/cmake/vtk-6.3/VTKTargets.cmake"
but not all the files it references.

-- RTABMAP_SYNC_MULTI_RGBD = ON
-- RTABMAP_SYNC_USER_DATA  = ON
-- rtabmap_ros: 16 messages, 9 services
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- 1111111111111111111111111111111111111111111
-- 2222222222222222222222222222222222222222222
-- 3333333333333333333333333333333333333333333
-- WITH octomap_msgs
-- WITH find_object_2d
-- WITH rviz
-- WITH costmap_2d
-- Configuring done
-- Generating done
-- Build files have been written to: /home/adaviv/catkin_ws/build
```

matlabbe, 2020-11-18:
It seems built with it. Not sure why you are seeing the error. When RTABMAP_SYNC_MULTI_RGBD is not defined, the code is passing here:
https://github.com/introlab/rtabmap_ros/blob/c20e38b11e264f4615c1d60e3fa059282982947b/src/CommonDataSubscriber.cpp#L573

You may add some ROS_WARN("test1") in the code around there:
https://github.com/introlab/rtabmap_ros/blob/c20e38b11e264f4615c1d60e3fa059282982947b/src/CommonDataSubscriber.cpp#L499

That define is added here:
https://github.com/introlab/rtabmap_ros/blob/646ad6ee7bfbae7e3a526abe0db7ae80dde372dc/CMakeLists.txt#L53-L55

Sd1010, 2020-11-19:
Hi @matlabbe,

I have added the following in the rtabmap_ros/src/CommonDataSubscriber.cpp

```
#ifdef RTABMAP_SYNC_MULTI_RGBD
		ROS_WARN("test1111111111111111111111111111111111111111");
		if(rgbdCameras == 4)
		{
			setupRGBD4Callbacks(
					nh,
					pnh,
					subscribedToOdom_,
					subscribeUserData,
					subscribeScan2d,
					subscribeScan3d,
					subscribeScanDesc,
					subscribeOdomInfo,
					queueSize_,
					approxSync_);
		}
		else if(rgbdCameras == 3)
		{
			setupRGBD3Callbacks(
					nh,
					pnh,
					subscribedToOdom_,
					subscribeUserData,
					subscribeScan2d,
					subscribeScan3d,
					subscribeScanDesc,
					subscribeOdomInfo,
					queueSize_,
					approxSync_);
		}
		else if(rgbdCameras == 2)
		{
			ROS_WARN("test222222222222222222222222222222222222222222");
			setupRGBD2Callbacks(
					nh,
					pnh,
					subscribedToOdom_,
					subscribeUserData,
					subscribeScan2d,
					subscribeScan3d,
					subscribeScanDesc,
					subscribeOdomInfo,
					queueSize_,
					approxSync_);
		}
#else
		ROS_WARN("test333333333333333333333333333333333333333333");
		if(rgbdCameras>1)
		{
			ROS_FATAL("Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)");
		}
```

This is the output 

```
[ INFO] [1605744334.237512267]: Reading parameters from the ROS server...
[ INFO] [1605744334.271939245]: rtabmap: Database version = "0.20.0".
[ WARN] [1605744334.290262431]: rtabmap: Parameters subscribe_rgb and subscribe_rgbd cannot be true at the same time. Parameter subscribe_rgb is set to false.
[ INFO] [1605744334.292490722]: /rtabmap/rtabmap: subscribe_depth = false
[ INFO] [1605744334.292512084]: /rtabmap/rtabmap: subscribe_rgb = false
[ INFO] [1605744334.292523928]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1605744334.292534935]: /rtabmap/rtabmap: subscribe_rgbd = true (rgbd_cameras=2)
[ INFO] [1605744334.292542629]: /rtabmap/rtabmap: subscribe_odom_info = false
[ INFO] [1605744334.292553553]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1605744334.292563480]: /rtabmap/rtabmap: subscribe_scan = false
[ INFO] [1605744334.292573687]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1605744334.292584389]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1605744334.292596357]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1605744334.292606603]: /rtabmap/rtabmap: approx_sync   = true
[FATAL] [1605744334.292617306]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)
[ INFO] [1605744334.292659314]: 
[ INFO] [1605744334.300803301]: Parameters read = 0
[ INFO] [1605744334.386915520]: rtabmap 0.20.0 started...
[ INFO] [1605744334.562209087]: /rtabmap/rtabmapviz: subscribe_depth = false
[ INFO] [1605744334.562241494]: /rtabmap/rtabmapviz: subscribe_rgb = false
[ INFO] [1605744334.562254374]: /rtabmap/rtabmapviz: subscribe_stereo = false
[ INFO] [1605744334.562269108]: /rtabmap/rtabmapviz: subscribe_rgbd = true (rgbd_cameras=2)
[ INFO] [1605744334.562279581]: /rtabmap/rtabmapviz: subscribe_odom_info = true
[ INFO] [1605744334.562287812]: /rtabmap/rtabmapviz: subscribe_user_data = false
[ INFO] [1605744334.562298018]: /rtabmap/rtabmapviz: subscribe_scan = false
[ INFO] [1605744334.562307762]: /rtabmap/rtabmapviz: subscribe_scan_cloud = false
[ INFO] [1605744334.562317275]: /rtabmap/rtabmapviz: subscribe_scan_descriptor = false
[ INFO] [1605744334.562326618]: /rtabmap/rtabmapviz: queue_size    = 10
[ INFO] [1605744334.562336446]: /rtabmap/rtabmapviz: approx_sync   = true
[FATAL] [1605744334.562346779]: Cannot synchronize more than 1 rgbd topic (rtabmap_ros has been built without RTABMAP_SYNC_MULTI_RGBD option)
[ INFO] [1605744334.562426816]: 
[ INFO] [1605744334.562495297]: rtabmapviz started.

```

matlabbe, 2020-11-19:
It looks like the wrong rtabmap version is started:
```
[ INFO] [1605744334.386915520]: rtabmap 0.20.0 started...
```
Uninstall the binaries to make sure the on built from source is used:
```
sudo apt remove ros-melodic-rtabmap
```

Sd1010, 2020-11-19:
@matlabbe Thanks for the quick reply. I have uninstalled the binaries but now i am getting this error:

```
Resource not found: rtabmap_ros
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/opt/ros/melodic/share
The traceback for the exception was written to the log file
```

matlabbe, 2020-11-30:
Is rtabmap_ros git in your `catkin_ws/src` ?Your ROS path doesn't have your catkin_ws, make sure to source `~/catkin_ws/devel/setup.bash`