introlab/rtabmap_ros #494: On noetic I get "Did not receive data since 5 seconds" but it worked on melodic

chfritz, 2020-11-06:
I'm migrating from melodic to noetic and when I start the same rtabmap launch file (further below) I get this warning and rtabmap doesn't start regular operation. This is in simulation, using gazebo with a simulated Realsense D435.
```
[ WARN] [1604620365.113034984, 269.160000000]: /rtabmap/rtabmap: Did not receive data 
since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") a
nd the timestamps in their header are set. If topics are coming from different compute
rs, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are 
not published at the same rate, you could increase "queue_size" parameter (current=10)
.                                                                                     
/rtabmap/rtabmap subscribed to (approx sync):                                         
   /exact_pose,                                                                       
   /rtabmap/rgbd_image                                                                
[ WARN] [1604620370.469484424, 274.160000000]: /rtabmap/rtabmap: Did not receive data 
since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") a
nd the timestamps in their header are set. If topics are coming from different compute
rs, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are 
not published at the same rate, you could increase "queue_size" parameter (current=10)
.                                                                                     
/rtabmap/rtabmap subscribed to (approx sync):                                         
   /exact_pose,                                                                       
   /rtabmap/rgbd_image                                                                
[ WARN] [1604620375.877815127, 279.160000000]: /rtabmap/rtabmap: Did not receive data 
since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") a
nd the timestamps in their header are set. If topics are coming from different compute
rs, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are 
```

The topics are being published at a good rate:
```
> rostopic hz /exact_pose
subscribed to [/exact_pose]
WARNING: may be using simulated time
average rate: 100.000
        min: 0.000s max: 0.047s std dev: 0.01585s window: 90
average rate: 99.670
        min: 0.000s max: 0.047s std dev: 0.01661s window: 182
average rate: 99.916
        min: 0.000s max: 0.047s std dev: 0.01672s window: 238

> rostopic hz /rtabmap/rgbd_image
subscribed to [/rtabmap/rgbd_image]
WARNING: may be using simulated time
average rate: 9.447
        min: 0.066s max: 0.142s std dev: 0.03096s window: 8
average rate: 9.969
        min: 0.053s max: 0.143s std dev: 0.03443s window: 17
average rate: 9.946
        min: 0.053s max: 0.147s std dev: 0.03518s window: 27
average rate: 9.990
        min: 0.050s max: 0.147s std dev: 0.03524s window: 31
```

Launch file:
```
<launch>

  <arg name="localization" default="false"/>

  <include file="$(find rtabmap_ros)/launch/rtabmap.launch">

    <arg name="frame_id" value="fused_odom_link"/>
    <arg name="rgbd_sync" value="true"/>
    <arg name="depth_topic" value="/d435/camera/depth/image_raw"/>
    <arg name="rgb_topic" value="/d435/camera/color/image_raw"/>
    <arg name="camera_info_topic" value="/d435/camera/color/camera_info"/>
    <arg name="approx_rgbd_sync" value="true"/>
    <arg name="approx_sync" value="false"/>
    <arg name="visual_odometry" value="false"/>
    <arg name="localization" value="$(arg localization)"/>

    <arg name="rtabmapviz" value="false"/>

    <arg name="odom_topic" value="/exact_pose"/>
    <arg name="args"  value="-d --Mem/UseOdomGravity true --Optimizer/GravitySigma 0.3"/>
    <arg name="publish_tf_map" value="false"/>

  </include>
</launch>
```

And here is my TF tree:
```
└── map                                                                                                                                                                      
    ├── base_stabilized [/exact_odom]                                                                                                                                        
    │   └── base_footprint [/exact_odom]                                                                                                                                     
    │       ├── base_link [/robot_state_publisher]                                                                                                                           
    │       └── fused_odom_link [/robot_state_publisher]                                                                                                                     
    │           └── d435_camera [/robot_state_publisher]
    │               ├── d435_color [/robot_state_publisher]  
    │               ├── depth [/robot_state_publisher]
    │               ├── ired1 [/robot_state_publisher]                                                                                                                       
    │               └── ired2 [/robot_state_publisher]
    └── odom                 
```

I've already tried increasing the `queue_size` as the warning suggests but that didn't make a difference. I've also checked the headers of the topics in question and they all have proper timestamps (from gazebo's `/clock`).

This is using 
```
> dpkg -la | grep rtabmap-ros
ii  ros-noetic-rtabmap-ros                      0.20.0-3focal.20200930.013155         amd64        RTAB-Map's ros-pkg.
```
I've also tried compiling the latest rtabmap_ros from the noetic-devel branch, but that didn't make a difference.

Did anything change between melodic and noetic that might explain this? Any idea where to poke further?

chfritz, 2020-11-19:
I'm still stuck on this. Any thoughts? or is noetic not yet supported?

matlabbe, 2020-11-20:
Yes, Noetic should work (tested with [those demos](http://wiki.ros.org/rtabmap_ros#Demos)). However, they seem to have change something on how `message_filters` works between melodic and noetic, but I cannot find exactly why, causing inconsistent synchronization on Noetic. For example, I've seen synchronization not working on live topics, but if they were recorded in a rosbag and replayed, synchronization worked!? A ugly [workaround ](https://github.com/introlab/rtabmap_ros/commit/ed564c3a12293d06ca28e145ff50a974f6159cc9) that we've found is to set the `queue_size` of each individual subscribers while the [official tutorial](http://wiki.ros.org/message_filters#CA-0020b1f6c604f6ea4af7e07c8cb8576c32c5c97d_19) they don't do that (I also think we don't have to do that as it worked since ROS Hydro). As I said at the beginning, I think the problem is deeper, because depending of the size of the `librtabmap_sync.so` library, it can also influence the synchronization of the topics on Noetic. As I am developing mostly on Melodic, I could not put more time to debug in depth this problem (is that packages on Noetic are built in Debug mode by default instead of Release?).

For your problem though, the poses are published 10 times more often than the images. A `queue_size` of 10 is indeed too limited. I would increase it to 20 at least. The approximate time synchronization is explained here: http://wiki.ros.org/message_filters/ApproximateTime

Can you record a small rosbag to test your launch file above?
```
rosbag record /tf /tf_static /exact_pose /d435/camera/depth/image_raw /d435/camera/color/image_raw /d435/camera/color/camera_info
```

Sunshengjin, 2021-01-06:
Hi, all.  Is anyone making any progress? I am also stuck in this problem...If there is any progress, please reply, thanks very very much