introlab/rtabmap_ros #495: The mapping works for rotation but during translation the point cloud is duplicated

mirellameelo, 2020-11-06:
Hi!
I'm using a stereo camera and ROS2 (eloquent). I already debugged my input data as much as I could, but during the robot's translation, the point cloud has been duplicating (when rotating, images are matched as expected). The impression is that Rtabmap is not compensating the movement for the translation info and images are redraw instead of matched.

The image below: translation VERSUS rotation and grid map (notice how the wall "walk" against the movement):
[image]

Everything said is much better described by this [video](https://drive.google.com/drive/folders/1jf8mCKOxw0ZR11FvWvPaDin_7OjQU5lq?usp=sharing) - it also contains the .db file. 

Some observations of my case:
- The images are rectified and there are no distortions (D) (virtual camera). K, R, P are also informed (It was debugged and parameters match). 

- Odometry info was converted from ORB to ROS coordinate system (x+ to forward, y+ to left, z+ to up) and broadcasted to Rtabmap (```odom -> base_link```). Using (1) RVIZ2 and (2) ```tf2_echo``` I verified the coherence regarding the ROS coordinate system.

- While running, Rtabmap does not show warning or errors at all.

- Inputs consist of: right image, left image, odometry, right camera config, left camera config, and TF. [Here](https://github.com/mirellameelo/rtabmap_openvslam/blob/master/ros/2/src/openvslam/src/run_slam.cc) my node.

On my side, I exhausted debugged as I could. Since during rotation the mapping seems to work fine, my last shot is this is not by my side, but maybe a possible bug...? Given that I could not find any example for ROS2 and stereo, it has been really hard, and at the moment I feel completely blind... Any help is more than welcome.

Thanks in advance.

matlabbe, 2020-11-07:
The problem is here:
https://github.com/mirellameelo/rtabmap_openvslam/blob/593ca43de27e98a4cbe1dce9136eadb4a6ef166e/ros/2/src/openvslam/src/run_slam.cc#L166

You should use:
https://github.com/xdspacelab/openvslam/blob/bfd542ea8397954a37db3de51e8ff817e712e53c/src/openvslam/system.h#L108-L110

With mono approach, the scale is unknown, so in odometry world it thinks it move 2 cm, but in reality it should move 20 cm for example.

mirellameelo, 2020-11-07:
!!!! Thank you so much!

closed 2020-11-07 by mirellameelo

mirellameelo, 2020-11-08:
@matlabbe is great now!! =D

[image]

To use the ray-tracing function (in order to fill these empty spaces) I set "map_empty_ray_tracing" to true, but the map keeps the same. Could you please confirm the proper way to use this functionality? 
Thanks!!

matlabbe, 2020-11-08:
Try with `Grid/RayTracing` set to true.