introlab/rtabmap_ros #496: Save/Extract RGBD Frames

hosseinshirali, 2020-11-11:
Hello
I am new to this field and now i am able to get 3D point cloud of the environment by using Realsense D455 but I also want to save/extract the RGB and depth frames,  how is it possible to do that?
besides, I want to link each point clouds to its own frame, I mean I want to get a portion of point clouds then find that frames correspond to that region of point clouds.
What is a possible way to achieve that?

Thanks

matlabbe, 2020-11-16:
Open the the database in `rtabmap-databaseViewer`, then do File -> Extract images...

You can also export poses in some formats, show up the View->GraphView and do File -> Export poses...

For 

>  I mean I want to get a portion of point clouds then find that frames correspond to that region of point clouds.

There is no really way to do this, unless you use the exported poses so you know the position of each camera frame.

hosseinshirali, 2020-12-22:
thanks for your answer

closed 2020-12-22 by hosseinshirali