introlab/rtabmap_ros #498: RGBD/LocalImmunizationRatio What is this param for?

NicoNorena, 2020-11-13:
When mapping I am seen a warming,  I have increased to  0.50, but am not sure how this affects 

[WARM] Rtabmap.cpp:2029::process() Could not immunize the whole local path (163) between 1 and 1291 (max location immunized=96). You may want to increase RGBD/LocalImmunizationRatio (current=0.500000 (96 of WM=192)) to be able to immunize longer paths.

When I run the params description :
`rtabmap --params | grep LocalImmunizationRatio`
I see the description as : "Ratio of working memory for which local nodes are immunized from transfer."

matlabbe, 2020-11-30:
This is used to avoid transferring close nodes when memory management is enabled. The warning says that it requires more than 69 nodes to reach the nearest node (not in STM) inside the `RGBD/LocalRadius`. Are there a lot of nodes on the same spot? Normally is it should not be very high. This could affect the number of proximity detection and loop closures that can be detected.

NicoNorena, 2021-03-04:
I has this issue on pause.
There might be a lot of nodes on the same spot.
Should I increase the STM size? I did not change  `RGBD/LocalRadius` , the default value is 10 m.
I have the STM size at 20

matlabbe, 2021-03-05:
What is the value of RGBD/LinearUpdate and RGBD/AngularUpdate? this would help to not accumulate nodes one over the other.

NicoNorena, 2021-03-08:
I probably have very small value.  RGBD/LinearUpdate=0.01 RGBD/AngularUpdate=0.01
what values do you recommend? I think by default is set to 0.1

matlabbe, 2021-03-12:
Yes, they are set at 0.1. At 0.01, if there is a noise of 1 cm on position, the robot may add nodes even when not moving.

closed 2021-08-18 by NicoNorena