introlab/rtabmap_ros #499: rtabmap on Tiago Robot

thohemp, 2020-11-16:
Hello,

I want to test rtabmap on a real tiago Robot with visual odometry. I experience following shifting of the pointcloud. Any idea where this is coming from? Frames overview: [frames.pdf](https://github.com/introlab/rtabmap_ros/files/5545985/frames.pdf)

[image]

 My launch file is the following:
```

<launch>
  <!-- 
       Bringup Turtlebot:
       $ roslaunch turtlebot_bringup minimal.launch
       
       Mapping:
       $ roslaunch rtabmap_ros demo_turtlebot_mapping.launch
       
       Visualization:
       $ roslaunch rtabmap_ros demo_turtlebot_rviz.launch
       
       This launch file is a one to one replacement of the gmapping_demo.launch in the 
       "SLAM Map Building with TurtleBot" tutorial:
       http://wiki.ros.org/turtlebot_navigation/Tutorials/indigo/Build%20a%20map%20with%20SLAM
       
       For localization-only after a mapping session, add argument "localization:=true" to
       demo_turtlebot_mapping.launch line above. Move the robot around until it croslaunch rtabmap_ros demo_turtlebot_mapping.launch simulation:=truean relocalize in 
       the previous map, then the 2D map should re-appear again. You can then follow the same steps 
       from 3.3.2 of the "Autonomous Navigation of a Known Map with TurtleBot" tutorial:
       http://wiki.ros.org/turtlebot_navigation/Tutorials/Autonomously%20navigate%20in%20a%20known%20map
 
       For turtlebot in simulation (Gazebo):
         $ roslaunch turtlebot_gazebo turtlebot_world.launch
         $ roslaunch rtabmap_ros demo_turtlebot_mapping.launch simulation:=true
         $ roslaunch rtabmap_ros demo_turtlebot_rviz.launch
  -->

  <arg name="database_path"     default="rtabmap.db"/>
  <arg name="rgbd_odometry"     default="true"/>
  <arg name="rtabmapviz"        default="false"/>
  <arg name="localization"      default="false"/>
  <arg name="simulation"        default="false"/>
  <arg name="sw_registered"     default="false"/>
  <arg     if="$(arg localization)" name="args"  default=""/>
  <arg unless="$(arg localization)" name="args"  default="--delete_db_on_start"/>

  <arg     if="$(arg simulation)" name="rgb_topic"   default="/xtion/rgb/image_raw"/>
  <arg unless="$(arg simulation)" name="rgb_topic"   default="/xtion/rgb/image_rect_color"/>
  <arg     if="$(arg simulation)" name="depth_topic" default="/xtion/depth_registered/image_raw"/>
  <arg unless="$(arg simulation)" name="depth_topic" default="/xtion/depth_registered/image_raw"/>
  <arg name="camera_info_topic" default="/xtion/rgb/camera_info"/>
  <arg name="wait_for_transform"  default="0.2"/> 

  <arg name="tag_topic"                default="/tag_detections" />  <!-- apriltags async subscription -->

  <!-- 
      robot_state_publisher's publishing frequency in "turtlebot_bringup/launch/includes/robot.launch.xml" 
      can be increase from 5 to 10 Hz to avoid some TF warnings.
  -->
  
  <!-- Navigation stuff (move_base) -->
 <!-- <include unless="$(arg simulation)" file="$(find turtlebot_bringup)/launch/3dsensor.launch">
     <arg     if="$(arg sw_registered)" name="depth_registration" value="false"/>
     <arg unless="$(arg sw_registered)" name="depth_registration" value="true"/>
  </include>
  <include file="$(find turtlebot_navigation)/launch/includes/move_base.launch.xml"/>-->

  <!-- Mapping -->
  <group ns="rtabmap"> 

   <node pkg="nodelet" type="nodelet" name="rgbd_sync" args="standalone rtabmap_ros/rgbd_sync" output="screen">
[... 73 lines cut ...]

	  
	  <!-- localization mode -->
	  <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
	  <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
	  <param name="Mem/InitWMWithAllNodes" type="string" value="$(arg localization)"/> 
    </node>
   

    
    <!-- Visualisation RTAB-Map -->
    <node if="$(arg rtabmapviz)" pkg="rtabmap_ros" type="rtabmapviz" name="rtabmapviz" args="-d $(find rtabmap_ros)/launch/config/rgbd_gui.ini" output="screen">
      <param name="subscribe_rgbd"      type="bool" value="true"/>
      <param name="subscribe_laserScan" type="bool" value="true"/>
      <param name="frame_id"            type="string" value="base_footprint"/>
    
      <remap from="scan"            to="scan"/>

    </node>
    
  
    
  </group>
</launch>
```

matlabbe, 2020-11-16:
You should find a way to disable rrt_deployer publishing `odom->base_footprint`. What you are seeing it seems that two nodes (rrqt_deployer and rgbd_odometry) are publishing both `odom->base_footprint`. 

Without disabling rrt_deployer, it is also possible to use rrt_deployer's odometry to help rgbd_odometry, with `guess_frame_id` parameter:
```xml
    <!-- Odometry -->
    <node pkg="rtabmap_ros" type="rgbd_odometry" name="rgbd_odometry" output="screen">
      <param name="subscribe_rgbd" type="bool"   value="true"/>
      <param name="frame_id"       type="string" value="base_footprint"/>
      <param name="guess_frame_id" type="string" value="odom"/>
      <param name="odom_frame_id"  type="string" value="visual_odom"/>
      <remap from="rgbd_image" to="rgbd_image"/>

      <param name="Odom/Strategy"		value="5"/>
      <param name="OdomORBSLAM2/VocPath"	type="string" value="/home/pal/ORB_SLAM2/Vocabulary/ORBvoc.txt"/>
    </node>
```
You should then see a Tf tree like that: `visual_odom->odom->base_footprint->...`. For rtabmap node, `odom_frame_id` should be then set to `visual_odom` instead of `odom`, to get `map->visual_odom->odom->base_footprint->...`

thohemp, 2020-11-24:
Thanks,! Worked like you predicted.

closed 2020-11-24 by thohemp