introlab/rtabmap_ros #502: Match pointcloud and RGB frames

hosseinshirali, 2020-11-18:
Hi
I am new to this field, now I can create 3D pointcloud of the environment by using depth camera D455 but I want to know that is it possible to match pointcloud to related RGB frames?
I want to do an operation on my global pointcloud then do the same thing on RGB frames that are connected to that points
thanks for your help

matlabbe, 2020-11-30:
There is no ros interface to do that. However, if doing it offline is possible, there is an example [here](https://github.com/introlab/rtabmap/blob/96c1c81e22959009b236fc9bd2bbb3cd06daa65c/tools/Export/main.cpp#L416-L484) where we reconstruct the point cloud, for each camera, we create a local cloud before assembling it to global point cloud, so you could know which point has been created by which camera.

hosseinshirali, 2020-12-22:
thanks for your help

closed 2020-12-22 by hosseinshirali