introlab/rtabmap_ros #507: The plugin for class '(no class name found)' failed to load!

liuzhenboo, 2020-11-25:
Hi, there are some wrong with my rviz. Can you help me?

`[ERROR] [1606272882.586894893]: PluginlibFactory: The plugin for class '(no class name found)' failed to load.  Error: According to the loaded plugin descriptions the class (no class name found) with base class type rviz::Display does not exist. Declared types are  octomap_rviz_plugin/ColorOccupancyGrid octomap_rviz_plugin/OccupancyGrid octomap_rviz_plugin/OccupancyGridStamped octomap_rviz_plugin/OccupancyMap octomap_rviz_plugin/OccupancyMapStamped rviz/Axes rviz/Camera rviz/DepthCloud rviz/Effort rviz/FluidPressure rviz/Grid rviz/GridCells rviz/Illuminance rviz/Image rviz/InteractiveMarkers rviz/LaserScan rviz/Map rviz/Marker rviz/MarkerArray rviz/Odometry rviz/Path rviz/PointCloud rviz/PointCloud2 rviz/PointStamped rviz/Polygon rviz/Pose rviz/PoseArray rviz/PoseWithCovariance rviz/Range rviz/RelativeHumidity rviz/RobotModel rviz/TF rviz/Temperature rviz/WrenchStamped rviz_plugin_tutorials/Imu`

matlabbe, 2020-11-26:
Is this after installing or building rtabmap_ros? Otherwise you would be better to ask on ros answers.

liuzhenboo, 2020-11-26:
Thank you very much for your reply！

I
I have installed rtabmap SDK and I also compile rtabmap_ros, octomap_rviz_plugins ,octomap_ros,octomap_msgs In the same ROS workspace. 

I only use a realsense l515 camera， this is my launch file [rgbd_mapping_navi.txt](https://github.com/introlab/rtabmap_ros/files/5600064/rgbd_mapping_navi.txt). I also use your azimut3_stereo_nav.rviz .
  
when I launch this file, I can get grid_map, costmap and I can also send 2d navigation target point(/planner_goal) ,TF(map->camera_link), grid_map(/rtabmap/grid_map) and local obstacles pointcloud map(/planner_cloud)to move_base, I could get a correct path planning results,as shown below:

[image]

[image]
[image]

[image]

**But I find some tiny error in my terminal：**

[image]
Excuse me, do you know why? why '(no class name found)'? I want to know why this error happens?

liuzhenboo, 2020-11-26:
Can rtabmap_ros only use move_base for path planning? Did you write a package with path planning?

matlabbe, 2020-11-29:
The PluginlibFactory error may not be related to rtabmap. One of the RVIZ plugins cannot be properly parsed, but from with package?

rtabmap_ros can do graph planning inside its graph, but no grid-based planning. For grid-based planning, rtabmap_ros relies on move_base (or any other planning approaches taking a goal and an occupancy grid as inputs).

liuzhenboo, 2020-12-02:
> The PluginlibFactory error may not be related to rtabmap. One of the RVIZ plugins cannot be properly parsed, but from with package?
>
> rtabmap_ros can do graph planning inside its graph, but no [...]

thanks!!

closed 2020-12-02 by liuzhenboo