introlab/rtabmap_ros #510: rgbd_odometry input type error

aled96, 2020-12-07:
I am running rtabmap with a simulated Intel Realsense D435i but I have the following error:

`[RGBDOdometry::commonCallback]: Input type must be image=mono8,mono16,rgb8,bgr8,bgra8,rgba8 and image_depth=32FC1,16UC1,mono16. Current rgb=rgb8 and depth=rgb8`

Even changing the `format` of the depth in the xacro file, I receive the same error.

matlabbe, 2020-12-10:
rgb8 for depth is strange, which topic are you using?

aled96, 2020-12-11:
I am using a simulated realsense D435i. The depth comes out with rgb8 specifying a format "RGB_INT8", but if I change to mono16 or others, I have bgr8...

The topic is "camera_aligned_depth_to_color/image_raw"

aled96, 2020-12-11:
Now I managed to convert it into mono16 but rtabmap results in no cloud..
I can see Voxel_cloud but with a different scale and orientation, the others are empty

Rutika1212, 2025-02-23:
Hello, I am also getting same error what is work around for this. Below is my scenario.

I am using Jetson Orin nano with jetpack 6.1 following installations:
ubuntu 22.04 
kernel 5.15.148 tegra
librealsense sdk version 2.54.1
using intelrealsense camera D435
ros wrapper intelrealsense version 4.54.1
ros2 humble

===================================
I have also build rtabmap using  following commands

sudo apt remove ros-humble-rtabmap*

cd ~/ros2_ws
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j2" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release

==================================
Below is rtabmap version

rutika@rutika-desktop:~$ rtabmap --version
RTAB-Map:               0.21.9
PCL:                    1.12.1
With VTK:                9.1.0
OpenCV:                  4.8.0
With OpenCV xfeatures2d: false
With OpenCV nonfree:     false
With ORB OcTree:          true
With SuperPoint Torch:   false
With Python3:            false
With FastCV:             false
With OpenGV:             false
With Madgwick:            true
With PDAL:               false
With libLAS:             false
With CudaSift:           false
With TORO:                true
With g2o:                 true
With GTSAM:               true
With Vertigo:             true
With CVSBA:              false
With Ceres:              false
With OpenNI:              true
With OpenNI2:             true
With Freenect:           false
With Freenect2:          false
With K4W2:               false
With K4A:                false
With DC1394:             false
With FlyCapture2:        false
With ZED:                false
With ZED Open Capture:   false
With RealSense:          false
With RealSense SLAM:     false
With RealSense2:          true
With MYNT EYE S:         false
With DepthAI:            false
With XVisio SDK:         false
With libpointmatcher:     true
With CCCoreLib:          false
With Open3D:             false
With OctoMap:             true
With GridMap:             true
With cpu-tsdf:           false
With open chisel:        false
[... 169 lines cut ...]
[rtabmap-2]    /camera/color/camera_info \
[rtabmap-2]    /rtabmap/odom_info
[rtabmap_viz-3] [WARN] [1740298877.184015315] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Ajusting topic_queue_size (10) and sync_queue_size (10) can also help for better synchronization if framerates and/or delays are different. If topics are not published at the same rate, you could increase "sync_queue_size" and/or "topic_queue_size" parameters (current=10 and 10 respectively).
[rtabmap_viz-3] rtabmap_viz subscribed to (approx sync):
[rtabmap_viz-3]    /rtabmap/odom \
[rtabmap_viz-3]    /camera/color/image_raw \
[rtabmap_viz-3]    /camera/depth/image_rect_raw \
[rtabmap_viz-3]    /camera/color/camera_info \
[rtabmap_viz-3]    /rtabmap/odom_info
[rgbd_odometry-1] [WARN] [1740298879.381832103] [rtabmap.rgbd_odometry]: rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set.
[rgbd_odometry-1] rgbd_odometry subscribed to (approx sync, topic_queue_size=10, sync_queue_size=10):
[rgbd_odometry-1]    /camera/color/image_raw,
[rgbd_odometry-1]    /camera/depth/image_rect_raw,
[rgbd_odometry-1]    /camera/color/camera_info
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[rgbd_odometry-1] [INFO] [1740298879.431760796] [rclcpp]: signal_handler(signum=2)
[rtabmap_viz-3] [INFO] [1740298879.431906208] [rtabmap.rtabmap_viz]: rtabmap_viz stopping spinner...
[rtabmap_viz-3] [INFO] [1740298879.435124642] [rtabmap.rtabmap_viz]: rtabmap_viz: All done! Closing...
[rtabmap-2] [INFO] [1740298879.435422827] [rclcpp]: signal_handler(signum=2)
[rtabmap-2] [INFO] [1740298879.486212251] [rtabmap.rtabmap]: Parameters are not saved (No configuration file provided...)
[rtabmap-2] rtabmap: Saving database/long-term memory... (located at /home/rutika/.ros/rtabmap.db)
[rtabmap-2] rtabmap: Saving database/long-term memory...done! (located at /home/rutika/.ros/rtabmap.db, 0 MB)
[INFO] [rgbd_odometry-1]: process has finished cleanly [pid 7714]
[INFO] [rtabmap-2]: process has finished cleanly [pid 7716]
[ERROR] [rtabmap_viz-3]: process has died [pid 7718, exit code 255, cmd '/home/rutika/ros2_ws/install/rtabmap_viz/lib/rtabmap_viz/rtabmap_viz ~/.ros/rtabmap_gui.ini --ros-args --log-level rtabmap.rtabmap_viz:=info --log-level rtabmap_viz:=info --ros-args -r __node:=rtabmap_viz -r __ns:=/rtabmap -p use_sim_time:=False --params-file /tmp/launch_params_y4ir6tjw -r rgb/image:=/camera/color/image_raw -r depth/image:=/camera/depth/image_rect_raw -r rgb/camera_info:=/camera/color/camera_info -r rgbd_image:=rgbd_image_relay -r left/image_rect:=/stereo_camera/left/image_rect_color -r right/image_rect:=/stereo_camera/right/image_rect -r left/camera_info:=/stereo_camera/left/camera_info -r right/camera_info:=/stereo_camera/right/camera_info -r scan:=/scan -r scan_cloud:=/scan_cloud -r odom:=odom'].

matlabbe, 2025-03-04:
What means `colorizer`? The depth image received has rgb8 encoding, which is wrong. We expect one of `32FC1` (meters), `16UC1` (mm), `mono16`(mm) encodings for depth.