introlab/rtabmap_ros #512: `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)' collect2: error: ld returned 1 exit status

z760, 2020-12-11:
usr/bin/ld: warning: libopencv_core.so.3.4, needed by /home/jcfh/volans/devel/lib/librtabmap_core.so, may conflict with libopencv_core.so.3.2
/home/jcfh/volans/devel/.private/rtabmap_ros/lib/librtabmap_ros.so: undefined reference to `cv::error(int, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*, char const*, int)'
collect2: error: ld returned 1 exit status
CMakeFiles/rtabmap_save_objects_example.dir/build.make:1341: recipe for target '/home/jcfh/volans/devel/.private/rtabmap_ros/lib/rtabmap_ros/save_objects_example' failed
make[2]: *** [/home/jcfh/volans/devel/.private/rtabmap_ros/lib/rtabmap_ros/save_objects_example] Error 1
CMakeFiles/Makefile2:295: recipe for target 'CMakeFiles/rtabmap_save_objects_example.dir/all' failed
make[1]: *** [CMakeFiles/rtabmap_save_objects_example.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2

matlabbe, 2020-12-11:
You may have 2 opencv versions installed, rtabmap is linking on 3.4, and rtabmap_ros is linking on 3.2 (which is maybe a dependency coming from cv_bridge). Which ros version are you using?

z760, 2020-12-11:
melodic
thanks!