introlab/rtabmap_ros #516: tf setting between lidar and RGBD

bill4u, 2020-12-15:
[image]
The launch file is:  roslaunch realsense2_camera rs_d400_and_t265.launch

In order to run the following command:
roslaunch rtabmap_ros rtabmap.launch \
   args:="-d --Mem/UseOdomGravity true --Optimizer/GravitySigma 0.3" \
   odom_topic:=/t265/odom/sample \
   frame_id:=t265_link \
   rgbd_sync:=true \
   depth_topic:=/d400/aligned_depth_to_color/image_raw \
   rgb_topic:=/d400/color/image_raw \
   camera_info_topic:=/d400/color/camera_info \
   approx_rgbd_sync:=false \
   visual_odometry:=false

How to set the TF between the lidar and the t265_link
Any ideas?
Thanks.

bill4u, 2020-12-15:
Do I need to transform the odom topic of t265 with tilt angle of 45deg?

bill4u, 2020-12-24:
@matlabbe Can you give me some advices? THANKS a lot.

matlabbe, 2020-12-24:
If you take t265_link as your base_link, you could add a static transform t265_link -> lidar_link (containing the 45 deg pitch rotation and x1,z1 offsets).

You could do also t265_link -> base_link -> lidar_link. In [this example with Tango](http://wiki.ros.org/rtabmap_ros/Tutorials/Tango%20ROS%20Streamer#Turtlebot), as Tango has its own odometry frame, we had to offset the odometry frame to match the base frame of the robot. You may do something similar for the 45 deg.

bill4u, 2020-12-29:
> If you take t265_link as your base_link, you could add a static transform t265_link -> lidar_link (containing the 45 deg pitch rotation and x1,z1 offsets).
>
> You could do also t265_link -> base_link [...]

Thanks for your reply. It works!!! I will close this issue.

closed 2020-12-29 by bill4u