introlab/rtabmap_ros #517: The frame convention of the latest realsense SDK is not the same as the config in vins-fusion reposirory.

TouchDeeper, 2020-12-16:
Hi, Mathieu

I used the `test_d435i_vio.launch` to run vins-fusion supported rtabmap. An error occured:
`[ERROR] [1608127730.914892966]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.`

I find the `realsense_stereo_imu_config.yaml` in VINS-FUSION has a different frame convention with my realsense d435i,  the default frame convention of  `realsense_stereo_imu_config.yaml` is:
[image]
The frame convention of my realsense d435i is below: 
[image]
[image]
you can note that the body frame convention is different with the config in VINS-FUSION. I can run VINS-FUSION well with my config calibrated by Kalibr. Then I use my config to run `test_d435i_vio.launch`, but errors about stereo baseline occur.
My config `realsense_stereo_imu_config_me.yaml` is (omit the same content with the default config):
```
body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 0.999917,   -0.0127019,   0.0020851,  -0.00182752,
           0.0126912,   0.999907,    0.00506861,  0.0093345,
          -0.00214928, -0.00504173,  0.999985,    0.0200701,
             0,          0,          0,           1 ]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [ 0.999921 ,  -0.0123262,   0.0025727,    0.048313,
           0.0123166,   0.999917,    0.00371045,   0.0101984,
          -0.00261822, -0.00367847,  0.99999,      0.0210006,
             0,          -0,          -0,           1 ]

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.1         # accelerometer measurement noise standard deviation. #0.2   0.04
gyr_n: 0.05        # gyroscope measurement noise standard deviation.     #0.05  0.004
acc_w: 7.1765713730075628e-04         # accelerometer bias random work noise standard deviation.  #0.002
gyr_w: 4.0e-05       # gyroscope bias random work noise standard deviation.     #4.0e-5
g_norm: 9.805         # gravity magnitude
```
The `test_d435i_vio.launch` is the same as yours except I use my own config `realsense_stereo_imu_config_me.yaml`

The SDK information is :
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | 05.12.09.00 | 
| Operating System & Version |   Ubuntu 16 | 
| Kernel Version (Linux Only)    |  4.15.0-128-generic             | 
| ros Version |   kinetic | 
| Platform                                 | PC |
| SDK Version                            |  2.40.0                      | 
| Language                            | C                         | 
| Segment			|  Robot                  | 
| installation Method			|  backend installation refer to https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md                  | 

Could you please provide some advice? Thank you a lot! The output information of  `test_d435i_vio.launch` is below:
```
PARAMETERS
 * /camera/realsense2_camera/accel_fps: 250
 * /camera/realsense2_camera/accel_frame_id: camera_accel_frame
 * /camera/realsense2_camera/accel_optical_frame_id: camera_accel_opti...
 * /camera/realsense2_camera/align_depth: True
 * /camera/realsense2_camera/aligned_depth_to_color_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_fisheye_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra1_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/aligned_depth_to_infra2_frame_id: camera_aligned_de...
 * /camera/realsense2_camera/allow_no_texture_points: False
 * /camera/realsense2_camera/base_frame_id: camera_link
 * /camera/realsense2_camera/calib_odom_file: 
[... 250 lines cut ...]
[ERROR] [1608127730.914892966]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ INFO] [1608127730.921717093]: Setting RTAB-Map parameter "Mem/IncrementalMemory"="true"
[ INFO] [1608127730.922213178]: Setting RTAB-Map parameter "Mem/InitWMWithAllNodes"="false"
 16/12 22:08:50,932 WARNING [140340849788672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ERROR] [1608127730.946915072]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1608127730.980049023]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ INFO] [1608127731.007483194]: Setting RTAB-Map parameter "RGBD/OptimizeMaxError"="0"
[ERROR] [1608127731.013762099]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1608127731.047054873]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1608127731.080353738]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
 16/12 22:08:51,083 WARNING [140340849788672] (messenger-libusb.cpp:42) control_transfer returned error, index: 768, error: No data available, number: 61
[ERROR] [1608127731.113618788]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ INFO] [1608127731.189205273]: Update RTAB-Map parameter "Optimizer/GravitySigma"="0.3" from arguments
[ERROR] [1608127731.196835159]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1608127731.305269935]: hwmon command 0x75( 0 1df 0 27f ) failed (response -6= Invalid parameter)
[ERROR] [1608127731.329564498]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ WARN] [1608127731.334585573]: Rtabmap: Parameter name changed: "RGBD/OptimizeRobust" -> "Optimizer/Robust". Please update your launch file accordingly. Value "true" is still set to the new parameter name.
[ WARN] [1608127731.337611182]: Rtabmap: Parameter name changed: "RGBD/OptimizeStrategy" -> "Optimizer/Strategy". Please update your launch file accordingly. Value "1" is still set to the new parameter name.
[ERROR] [1608127731.362870489]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ INFO] [1608127731.381390969]: RTAB-Map detection rate = 1.000000 Hz
[ INFO] [1608127731.381527394]: rtabmap: Deleted database "/home/td/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[ INFO] [1608127731.381553667]: rtabmap: Using database from "/home/td/.ros/rtabmap.db" (0 MB).
[ERROR] [1608127731.396163788]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
[ERROR] [1608127731.563181463]: The stereo baseline (-0.049985) should be positive (baseline=-Tx/fx). We assume a horizontal left/right stereo setup where the Tx (or P(0,3)) is negative in the right camera info msg.
```

matlabbe, 2020-12-25:
This error is independent of TF, it is coming from the camera _info topics. The camera projection matrix for the right camera_info should have P(0,3) negative. Can you show the camera_info of the left and right cameras?

TouchDeeper, 2020-12-26:
@matlabbe Thank you for your reply. 

The camera_info of left_infra is:
```
header: 
  seq: 867
  stamp: 
    secs: 1608946479
    nsecs: 473160331
  frame_id: "camera_infra1_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [383.15728759765625, 0.0, 319.8735046386719, 0.0, 383.15728759765625, 241.6318359375, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [383.15728759765625, 0.0, 319.8735046386719, 0.0, 0.0, 383.15728759765625, 241.6318359375, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

the camera_info of right_infra is:
```
header: 
  seq: 1473
  stamp: 
    secs: 1608946499
    nsecs: 644384161
  frame_id: "camera_infra2_optical_frame"
height: 480
width: 640
distortion_model: "plumb_bob"
D: [0.0, 0.0, 0.0, 0.0, 0.0]
K: [383.15728759765625, 0.0, 319.8735046386719, 0.0, 383.15728759765625, 241.6318359375, 0.0, 0.0, 1.0]
R: [1.0, 0.0, 0.0, 0.0, 1.0, 0.0, 0.0, 0.0, 1.0]
P: [383.15728759765625, 0.0, 319.8735046386719, 19.151988983154297, 0.0, 383.15728759765625, 241.6318359375, 0.0, 0.0, 0.0, 1.0, 0.0]
binning_x: 0
binning_y: 0
roi: 
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: False
```

matlabbe, 2020-12-27:
19.151988983154297 should be -19.151988983154297

TouchDeeper, 2020-12-28:
@matlabbe Thank you for your reply.
But it is not convenient to modify the camera_info, this should be caused by realsense's firmware, and other people will also encounter this problem.  What's more, I think the positive 19.151988983154297 makes sense in the frame convention of my realsense d435i. Could you please provide some advice to modify rtabmap to adapt to this situation?

TouchDeeper, 2021-01-04:
@matlabbe please help~