introlab/rtabmap_ros #518: Any way of choosing the frame_id of CloudXYZ?

jorgemia, 2020-12-22:
From what I can tell, the point cloud nodelet publishes a point cloud in the same frame as the depth image (as it just takes the same header), in my case `left_camera_frame`. Is there any way of changing this frame to eg. `base_link` easily within rtabmap?

Or should I lookup the transform between the camera and base_link myself and use that to transform the point cloud into the new frame in a new node?

matlabbe, 2020-12-23:
I assume you are referring to CloudXYZ output from rtabmap_ros/point_cloud_xyz nodelet. There is indeed to no option to transform the cloud in another frame in that node. You would have to subscribe to TF on your side and transform the cloud yourself. Note sure why you would want it to be in another frame though.

If you use it with move_base local costmap obstacle layers, it will automatically transform the cloud in the base frame of the local costmap.

jorgemia, 2020-12-24:
Yes that's what I was referring to. I wanted to transform it to use the cloud in the base_link frame (to get data relative to the robot center) for a custom wall-following node I'm working on. I was using rtabmap to transform my depth image into a pointcloud in this case, but yeah I should probably keep it in the original frame within rtabmap for the times that I use it with move_base and just transform the cloud message to base_link in my custom node to do the operations I want. Thanks!

closed 2020-12-24 by jorgemia