introlab/rtabmap_ros #519: Rtabmap_ROS Errors

ManChrys, 2020-12-30:
Hello again guys!!

As you can see im trying all of your projects and im very excited you did great job :D

I'm using ZED2. Zed_rtabmap_example from the other repo works fine i dont know if that helps .

When im using all launch files i have this error : 

SUMMARY
========

PARAMETERS
 * /data_recorder/DbSqlite3/InMemory: false
 * /data_recorder/Kp/MaxFeatures: -1
 * /data_recorder/Mem/MapLabelsAdded: false
 * /data_recorder/Mem/RehearsalSimilarity: 1.0
 * /data_recorder/Mem/STMSize: 1
 * /data_recorder/RGBD/AngularUpdate: 0
 * /data_recorder/RGBD/CreateOccupancyGrid: false
 * /data_recorder/RGBD/Enabled: false
 * /data_recorder/RGBD/LinearUpdate: 0
 * /data_recorder/RGBD/MaxLocalRetrieved: 0
 * /data_recorder/RGBD/ProximityBySpace: false
 * /data_recorder/Rtabmap/DetectionRate: 0
 * /data_recorder/Rtabmap/MaxRetrieved: 0
 * /data_recorder/Rtabmap/MemoryThr: 2
 * /data_recorder/approx_sync: True
 * /data_recorder/database_path: output.db
 * /data_recorder/depth/image_transport: raw
 * /data_recorder/frame_id: camera_link
 * /data_recorder/ground_truth_base_frame_id: 
 * /data_recorder/ground_truth_frame_id: 
 * /data_recorder/left/image_transport: raw
 * /data_recorder/odom_frame_id: camera_link
 * /data_recorder/publish_tf: False
 * /data_recorder/queue_size: 10
 * /data_recorder/rgb/image_transport: raw
 * /data_recorder/right/image_transport: raw
 * /data_recorder/subscribe_rgbd: False
 * /data_recorder/subscribe_scan: False
 * /data_recorder/subscribe_scan_cloud: False
 * /data_recorder/subscribe_stereo: False
 * /rosdistro: melodic
 * /rosversion: 1.14.10

NODES
  /
    data_recorder (rtabmap_ros/rtabmap)

auto-starting new master
process[master]: started with pid [22563]

setting /run_id to 9abce1b0-4a93-11eb-97e8-3c6aa7d21fc7
process[rosout-1]: started with pid [22574]
started core service [/rosout]
process[data_recorder-2]: started with pid [22581]
[ INFO] [1609328339.099298945]: Starting node...
[ INFO] [1609328339.133459902]: Initializing nodelet with 8 worker threads.
[ INFO] [1609328339.224079067]: /data_recorder(maps): map_filter_radius          = 0.000000
[ INFO] [1609328339.224102660]: /data_recorder(maps): map_filter_angle           = 30.000000
[ INFO] [1609328339.224126351]: /data_recorder(maps): map_cleanup                = true
[ INFO] [1609328339.224138948]: /data_recorder(maps): map_always_update          = false
[ INFO] [1609328339.224146264]: /data_recorder(maps): map_empty_ray_tracing      = true
[ INFO] [1609328339.224153291]: /data_recorder(maps): cloud_output_voxelized     = true
[ INFO] [1609328339.224160367]: /data_recorder(maps): cloud_subtract_filtering   = false
[ INFO] [1609328339.224168308]: /data_recorder(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1609328339.224642926]: /data_recorder(maps): octomap_tree_depth         = 16
[ INFO] [1609328339.238962718]: rtabmap: frame_id      = camera_link
[ INFO] [1609328339.238983218]: rtabmap: odom_frame_id = camera_link
[ INFO] [1609328339.238992851]: rtabmap: map_frame_id  = map
[ INFO] [1609328339.239000894]: rtabmap: use_action_for_goal  = false
[ INFO] [1609328339.239010506]: rtabmap: tf_delay      = 0.050000
[ INFO] [1609328339.239019012]: rtabmap: tf_tolerance  = 0.100000
[ INFO] [1609328339.239026542]: rtabmap: odom_sensor_sync   = false
[ INFO] [1609328339.257778845]: Setting RTAB-Map parameter "DbSqlite3/InMemory"="false"
[ INFO] [1609328339.333544904]: Setting RTAB-Map parameter "Kp/MaxFeatures"="-1"
[ INFO] [1609328339.360288142]: Setting RTAB-Map parameter "Mem/MapLabelsAdded"="false"
[ INFO] [1609328339.363943739]: Setting RTAB-Map parameter "Mem/RehearsalSimilarity"="1.0"
[ INFO] [1609328339.364967500]: Setting RTAB-Map parameter "Mem/STMSize"="1"
[ INFO] [1609328339.384212220]: Setting RTAB-Map parameter "RGBD/AngularUpdate"="0"
[ INFO] [1609328339.384485525]: Setting RTAB-Map parameter "RGBD/CreateOccupancyGrid"="false"
[ INFO] [1609328339.384747864]: Setting RTAB-Map parameter "RGBD/Enabled"="false"
[ INFO] [1609328339.387082485]: Setting RTAB-Map parameter "RGBD/LinearUpdate"="0"
[ INFO] [1609328339.391750980]: Setting RTAB-Map parameter "RGBD/MaxLocalRetrieved"="0"
[ INFO] [1609328339.399043859]: Setting RTAB-Map parameter "RGBD/ProximityBySpace"="false"
[ INFO] [1609328339.409238038]: Setting RTAB-Map parameter "Rtabmap/DetectionRate"="0"
[ INFO] [1609328339.413177992]: Setting RTAB-Map parameter "Rtabmap/MaxRetrieved"="0"
[ INFO] [1609328339.413438313]: Setting RTAB-Map parameter "Rtabmap/MemoryThr"="2"
[ INFO] [1609328339.586146188]: RTAB-Map detection rate = 0.000000 Hz
[ INFO] [1609328339.586267862]: rtabmap: Using database from "/home/geo/.ros/output.db" (0 MB).
[ INFO] [1609328339.617371623]: rtabmap: Database version = "0.20.7".
[ INFO] [1609328339.637447080]: /data_recorder: subscribe_depth = true
[ INFO] [1609328339.637491895]: /data_recorder: subscribe_rgb = true
[ INFO] [1609328339.637512911]: /data_recorder: subscribe_stereo = false
[ INFO] [1609328339.637528664]: /data_recorder: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1609328339.637544813]: /data_recorder: subscribe_odom_info = false
[ INFO] [1609328339.637560217]: /data_recorder: subscribe_user_data = false
[ INFO] [1609328339.637578758]: /data_recorder: subscribe_scan = false
[ INFO] [1609328339.637593583]: /data_recorder: subscribe_scan_cloud = false
[ INFO] [1609328339.637609970]: /data_recorder: subscribe_scan_descriptor = false
[ INFO] [1609328339.637625719]: /data_recorder: queue_size    = 10
[ INFO] [1609328339.637641920]: /data_recorder: approx_sync   = true
[ INFO] [1609328339.637677960]: Setup depth callback
[ INFO] [1609328339.650792194]: 
/data_recorder subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info
[ INFO] [1609328339.746673772]: rtabmap 0.20.7 started...
**[ WARN] [1609328344.651272034]: /data_recorder: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/data_recorder subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info
[ WARN] [1609328349.651731294]: /data_recorder: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=10).
/data_recorder subscribed to (approx sync):
   /camera/rgb/image_rect_color \
   /camera/depth_registered/image_raw \
   /camera/rgb/camera_info**

**geo@geo:~$  rostopic hz my_topic**
WARNING: topic [/my_topic] does not appear to be published yet

**geo@geo:~$ rostopic list**
/camera/depth_registered/image_raw
/camera/rgb/camera_info
/camera/rgb/image_rect_color
/gps/fix
/imu/data
/rosout
/rosout_agg
/rtabmap/cloud_ground
/rtabmap/cloud_map
/rtabmap/cloud_obstacles
/rtabmap/global_path
/rtabmap/global_path_nodes
/rtabmap/global_pose
/rtabmap/goal
/rtabmap/goal_node
/rtabmap/goal_out
/rtabmap/goal_reached
/rtabmap/grid_map
/rtabmap/grid_prob_map
/rtabmap/info
/rtabmap/initialpose
/rtabmap/labels
/rtabmap/landmarks
/rtabmap/local_grid_empty
/rtabmap/local_grid_ground
/rtabmap/local_grid_obstacle
/rtabmap/local_path
/rtabmap/local_path_nodes
/rtabmap/localization_pose
/rtabmap/mapData
/rtabmap/mapGraph
/rtabmap/mapPath
/rtabmap/octomap_binary
/rtabmap/octomap_empty_space
/rtabmap/octomap_full
/rtabmap/octomap_global_frontier_space
/rtabmap/octomap_grid
/rtabmap/octomap_ground
/rtabmap/octomap_obstacles
/rtabmap/octomap_occupied_space
/rtabmap/odom
/rtabmap/odom_info
/rtabmap/odom_last_frame
/rtabmap/odom_local_map
/rtabmap/odom_local_scan_map
/rtabmap/odom_rgbd_image
/rtabmap/proj_map
/rtabmap/scan_map
/tf
/tf_static
/user_data_async

Thank you again guys.

matlabbe, 2020-12-30:
How did you launch [data_recorder.launch](https://github.com/introlab/rtabmap_ros/blob/master/launch/data_recorder.launch)? You would have to start zed2 wrapper launch file too and remap the correct topic you want to record.

Note also that data_recorder.launch is used to record raw frames without any odometry/slam, it just fills a database with camera data. That database can be then used as source in RTAB-Map standalone.

ManChrys, 2020-12-31:
Hi @matlabbe,

that would be my second question. When im running roslaunch zed_wrapper zed2.launch this happens

SUMMARY
========

PARAMETERS
 * /rosdistro: melodic
 * /rosversion: 1.14.10
 * /zed2/zed2_description: <?xml version="1....
 * /zed2/zed_node/auto_exposure_gain: True
 * /zed2/zed_node/auto_whitebalance: True
 * /zed2/zed_node/brightness: 4
 * /zed2/zed_node/contrast: 4
 * /zed2/zed_node/depth/depth_downsample_factor: 1.0
 * /zed2/zed_node/depth/depth_stabilization: 1
 * /zed2/zed_node/depth/max_depth: 20.0
 * /zed2/zed_node/depth/min_depth: 0.7
 * /zed2/zed_node/depth/openni_depth_mode: False
 * /zed2/zed_node/depth/quality: 1
 * /zed2/zed_node/depth/sensing_mode: 0
 * /zed2/zed_node/depth_confidence: 50
 * /zed2/zed_node/depth_texture_conf: 100
 * /zed2/zed_node/exposure: 100
 * /zed2/zed_node/gain: 100
 * /zed2/zed_node/gamma: 8
 * /zed2/zed_node/general/base_frame: base_link
 * /zed2/zed_node/general/camera_flip: False
 * /zed2/zed_node/general/camera_model: zed2
 * /zed2/zed_node/general/camera_name: zed2
 * /zed2/zed_node/general/gpu_id: -1
 * /zed2/zed_node/general/grab_frame_rate: 30
 * /zed2/zed_node/general/resolution: 2
 * /zed2/zed_node/general/self_calib: True
 * /zed2/zed_node/general/serial_number: 0
 * /zed2/zed_node/general/svo_compression: 2
 * /zed2/zed_node/general/verbose: False
 * /zed2/zed_node/general/zed_id: 0
 * /zed2/zed_node/hue: 0
 * /zed2/zed_node/mapping/fused_pointcloud_freq: 1.0
 * /zed2/zed_node/mapping/mapping_enabled: False
 * /zed2/zed_node/mapping/max_mapping_range: -1
 * /zed2/zed_node/mapping/resolution: 0.05
 * /zed2/zed_node/object_detection/confidence_threshold: 50
 * /zed2/zed_node/object_detection/object_tracking_enabled: True
 * /zed2/zed_node/object_detection/od_enabled: False
 * /zed2/zed_node/object_detection/people_detection: True
 * /zed2/zed_node/object_detection/vehicle_detection: False
 * /zed2/zed_node/point_cloud_freq: 15.0
 * /zed2/zed_node/pos_tracking/area_memory: True
 * /zed2/zed_node/pos_tracking/area_memory_db_path: 
 * /zed2/zed_node/pos_tracking/fixed_z_value: 0.0
 * /zed2/zed_node/pos_tracking/floor_alignment: False
 * /zed2/zed_node/pos_tracking/imu_fusion: True
 * /zed2/zed_node/pos_tracking/init_odom_with_first_valid_pose: True
 * /zed2/zed_node/pos_tracking/initial_base_pose: [0.0, 0.0, 0.0, 0...
 * /zed2/zed_node/pos_tracking/map_frame: map
 * /zed2/zed_node/pos_tracking/odometry_frame: odom
 * /zed2/zed_node/pos_tracking/path_max_count: -1
 * /zed2/zed_node/pos_tracking/path_pub_rate: 2.0
 * /zed2/zed_node/pos_tracking/publish_map_tf: True
 * /zed2/zed_node/pos_tracking/publish_tf: True
 * /zed2/zed_node/pos_tracking/two_d_mode: False
 * /zed2/zed_node/pub_frame_rate: 15.0
 * /zed2/zed_node/saturation: 4
 * /zed2/zed_node/sensors/publish_imu_tf: True
 * /zed2/zed_node/sensors/sensors_timestamp_sync: False
 * /zed2/zed_node/sharpness: 4
 * /zed2/zed_node/stream: 
[... 128 lines cut ...]
[ INFO] [1609398973.813045292]: Advertised on topic /zed2/zed_node/confidence/confidence_map
[ INFO] [1609398973.813367070]: Advertised on topic /zed2/zed_node/disparity/disparity_image
[ INFO] [1609398973.813722164]: Advertised on topic /zed2/zed_node/point_cloud/cloud_registered
[ INFO] [1609398973.814059196]: Advertised on topic /zed2/zed_node/pose
[ INFO] [1609398973.814377577]: Advertised on topic /zed2/zed_node/pose_with_covariance
[ INFO] [1609398973.814704363]: Advertised on topic /zed2/zed_node/odom
[ INFO] [1609398973.815044589]: Advertised on topic /zed2/zed_node/path_odom
[ INFO] [1609398973.815397595]: Advertised on topic /zed2/zed_node/path_map
[ INFO] [1609398973.815792700]: Advertised on topic /zed2/zed_node/imu/data
[ INFO] [1609398973.816101305]: Advertised on topic /zed2/zed_node/imu/data_raw
[ INFO] [1609398973.816436387]: Advertised on topic /zed2/zed_node/imu/mag
[ INFO] [1609398973.816745634]: Advertised on topic /zed2/zed_node/temperature/imu
[ INFO] [1609398973.817059725]: Advertised on topic /zed2/zed_node/atm_press
[ INFO] [1609398973.817383920]: Advertised on topic /zed2/zed_node/temperature/left
[ INFO] [1609398973.817677939]: Advertised on topic /zed2/zed_node/temperature/right
[ INFO] [1609398973.818050090]: Advertised on topic /zed2/zed_node/left_cam_imu_transform [LATCHED]

What does that means? :/  As i can it says

[ INFO] [1609398971.461610142]: ZED connection -> SUCCESS
[ INFO] [1609398973.462024243]:  ...  ZED ready

so whats going on here ? 

Thank you again guys. And i wish to all of you guys happiness, love and health  in the new year.

ManChrys, 2021-01-07:
**UPDATE**
i fixed this.

Zed-ros-wrapper --> zed-wrapper--> params --> zed2.yaml

i just make objection detection/od_enabled true instead of false.

[image]

closed 2021-01-07 by ManChrys