introlab/rtabmap_ros #520: What does this number(2.1981) mean?

colorDogs, 2021-01-04:
I would like to use the VO from Rtabmap and Odometry from the car for fusion using EKF. But I don't know how the VO of Rtabmap generates the covariance matrix. What does this magic number mean when I look at the code?
source code:
file: util3d_motion_estimation.cpp
line: 168

screenshot
[image]

By the way, what are the dimensions of the covariance matrix? Is xyz in terms of m or px?

Looking forward to your reply. Thank you

matlabbe, 2021-01-11:
For the magic 2.1981, see http://www.pcl-users.org/RANSAC-and-models-with-normals-tp4033082p4039008.html

For the units, they are in m^2 for the covariance unless your camera calibration is not in meters.

colorDogs, 2021-01-14:
> For the magic 2.1981, see http://www.pcl-users.org/RANSAC-and-models-with-normals-tp4033082p4039008.html
> 
> For the units, they are in m^2 for the covariance unless your camera calibration is not in meters.

Thank you very much. I understand

closed 2021-01-14 by colorDogs

Byson-source, 2022-10-23:
Hi, matlabbe! I have checked your link, but the forum has been closed, unfortunately. Can you explain its principle? Thank you very much!

matlabbe, 2022-10-23:
That number has been taken from here:
https://github.com/PointCloudLibrary/pcl/blob/fa3da0b10198199d913e4231db996e7570209c2a/sample_consensus/include/pcl/sample_consensus/sac_model.h#L432-L443

The original source seems coming from:
Paul W. Holland & Roy E. Welsch (1977) Robust regression using iteratively reweighted least-squares, Communications in Statistics - Theory and Methods, 6:9, 813-827, DOI: [10.1080/03610927708827533](https://doi.org/10.1080/03610927708827533)

which cited by this other [paper](https://arxiv.org/pdf/2111.00102.pdf) mentioning 2.1981.