introlab/rtabmap_ros #522: RTABMAP_ROS ERRORS

ManChrys, 2021-01-11:
Hello guys,

im running the launch file and worked fine but while im exporting the pointcloud or when im doing <<edit-->view point cloud>>  then crashes. Im trying it with regenerate cloud filter and cloud filtering with the defaults params.

ERROR NUMBER 1

[image]

ERROR NUMBER 2

[image]

im following the path but the log file doesnt appear to exist.

[image]

[roslaunch-geo-305.log](https://github.com/stereolabs/zed-ros-examples/files/5793130/roslaunch-geo-305.log)
[roslaunch-geo-1835.log](https://github.com/stereolabs/zed-ros-examples/files/5793131/roslaunch-geo-1835.log)
[roslaunch-geo-30865.log](https://github.com/stereolabs/zed-ros-examples/files/5793137/roslaunch-geo-30865.log)

i think these are the logs because i cant find 
log file: /home/geo/.ros/log/88e098ea-5369-11eb-8217-3c6aa7d21fc7/zed-rtabmapviz-3*.log
log file: /home/geo/.ros/log/88e098ea-5369-11eb-8217-3c6aa7d21fc7/zed-zed_node-2*.log

I dont know what im doing wrong.Any suggestion? :P 

Thanks in advance.

matlabbe, 2021-01-11:
For the first error, it is ZED wrapper related.

For the crash from rtabmapviz, what are the parameters in Export dialog you are using? It seems related to voxel size. You may increase it or set it to 0 to skip that part, what is the size of the map?

ManChrys, 2021-01-22:
Hello @matlabbe 

do you mean this  ? In rtabmap_ros where are the params? 

[Untitled Document 1.txt](https://github.com/introlab/rtabmap_ros/files/5855470/Untitled.Document.1.txt)

Thanks in advance.

matlabbe, 2021-01-23:
In the export dialog, looks like the assembled voxel size is 1 cm: `ExportCloudsDialog\assemble_voxel=0.01`. Set it larger or set to 0 to disable.

ManChrys, 2021-01-25:
this parameter which are you talking about is in the rtabmap.launch file ?

because inside the launch file there is a parameter where voxes size is already 0

<param name="voxel_size"  type="double" value="0.0"/>

[rtabmap.txt](https://github.com/introlab/rtabmap_ros/files/5864833/rtabmap.txt)

matlabbe, 2021-01-25:
This parameter is only used in export/view clouds dialog (it is a UI parameter). We cannot set it like RTAB-Map's parameters in launch file.
[image]

ManChrys, 2021-01-25:
Ok I see . Thank you my friend.

closed 2021-01-25 by ManChrys