introlab/rtabmap_ros #523: Finer Map

ManChrys, 2021-01-11:
Hello guys,

when im using zed_rtabmap_example i have this result

[image]

Post processing makes it a little better but still have that "noise". Which parameters improve the pointcloud?

Thanks again guys.

matlabbe, 2021-01-15:
You could tune zed parameters to avoid too much interpolation in textureless areas. You may tune those parameters: https://github.com/stereolabs/zed-ros-wrapper/blob/f5a691c39b01c565b7d3897c1f36f413c21d5782/zed_wrapper/params/common.yaml#L17-L18
In rtabmap standalone, we set them by default to 100 and 90 respectively.

ManChrys, 2021-01-16:
ill try that my friend and ill inform you soon.

Thank you one more time for your help and advices. Have a nice weekend.

ManChrys, 2021-01-19:
Hello @matlabbe, yeah it worked you were right. Thank you my friend.

closed 2021-01-19 by ManChrys