introlab/rtabmap_ros #524: Questions about day and night RGB changes

SeungRyeol, 2021-01-14:
I have a question while using the package.

We will implement visual inertial odometry through **RGBD+IMU** of D435i.

If I finish mapping with features during the **daytime**, can rtabmap match features well when I bring the map at **night**?
(or vice versa: mapping-**night**, loading_map-**daytime** )

I think it'll be okay to use it with dense through Depth, but I'm not sure.

I'm curious whether rtabmap's performance is affected by changes in RGB values during the day and night.

matlabbe, 2021-01-15:
It depends on which feature is used for loop closure detection. Ignoring licensing issues, SuperPoint feature is quite good against illumination changes. Otherwise, float descriptors seem more illumination invariant than binary descriptors.

Another option is to do ICP-based localization when using lidar or tof camera, but it won't work with stereo clouds produced with D435i.

SeungRyeol, 2021-01-20:
Thank you for your answer.

In fact, I am thinking about <RGBD + IMU> or <Stereo + IMU (VINS Fusion with RTAB)>.

Do you have any additional advice?

I don't know what would be a better solution under the conditions of changing environment and lighting.

matlabbe, 2021-02-12:
For visual odometry, unless  you turn off suddenly the lights, if should be okay with lighting changing slowly. For localization, you may set rtabmpa to use superpoint for example. This setting can be independent of odometry settings (use fast binary features for odometry, and slow superpoint features for localization).

SeungRyeol, 2021-03-04:
Thank you for your kind answer.

closed 2021-03-04 by SeungRyeol