introlab/rtabmap_ros #525: ROS2 Foxy parameters

soldierofhell, 2021-01-14:
Hi @matlabbe,
In the current construction of ROS2 nodes I'm not able to input any rtabmap parameters into nodes in launch files either through yaml or launch files. Here, for example, in the Odometry node there's a code that should read and override parameters, but for me it doeasn't work
https://github.com/introlab/rtabmap_ros/blob/d21b03ae7a6843307f948f9a9b5c558240d84d63/src/OdometryROS.cpp#L280-L301
In particular I tried the following way (the only example I found currently in ROS2 branch)
https://github.com/introlab/rtabmap_ros/blob/8ef6cff8777aa43818afe4744fe5942916f6fe40/launch/ros2/turtlebot3_scan.launch.py#L23-L24
That could be related with eloquent -> foxy interface change

Config through INI files works and for me is the only way to change rtabmap parameters

Could you please confirm that parameters input through yaml or dict should work (and e.g. work for eloquent)

Regards,

SBarzz, 2021-01-18:
I think that the problems that the parameters are not declared
so you should add this line in OdometryROS.cpp:
`declare_parameter(iter->first);`

Lines 235-254:

```
for(rtabmap::ParametersMap::iterator iter=parameters_.begin(); iter!=parameters_.end(); ++iter)
	{
		rclcpp::Parameter parameter;

		declare_parameter(iter->first);
		if(get_parameter(iter->first, parameter))
		{
			std::string vStr = parameter.as_string();
			RCLCPP_INFO(this->get_logger(), "Setting odometry parameter \"%s\"=\"%s\"", iter->first.c_str(), vStr.c_str());
			iter->second = vStr;
		}

		if(iter->first.compare(Parameters::kVisMinInliers()) == 0 && atoi(iter->second.c_str()) < 8)
		{
			RCLCPP_WARN(this->get_logger(), "Parameter min_inliers must be >= 8, setting to 8...");
			iter->second = uNumber2Str(8);
		}
	}

```

matlabbe, 2021-03-02:
It has been done this way for rtabmap node during the ros1->ros2 refactoring, but I forgot for odometry. Here is how it is done for rtabmap node:
https://github.com/introlab/rtabmap_ros/blob/688584db65b6e95b176680ae765642979252d760/src/CoreWrapper.cpp#L262-L279

Here what is used in odometry (wrong):
https://github.com/introlab/rtabmap_ros/blob/688584db65b6e95b176680ae765642979252d760/src/OdometryROS.cpp#L227-L242 

I think it would have to be:
```cpp
 	rclcpp::Parameter parameter; 
 	std::string vStr = this->declare_parameter(iter->first, iter->second); 
 	if(vStr.compare(iter->second)!=0) 
 	{ 
 		RCLCPP_INFO(this->get_logger(), "Setting odometry parameter \"%s\"=\"%s\"", iter->first.c_str(), vStr.c_str()); 
 		iter->second = vStr; 
 	} 
```

matlabbe, 2021-03-02:
The commit above should fix this issue.

commit 931e2ac4 (2021-03-02): Fixed #525
closed 2021-03-02 by matlabbe